dsmrreader/dsmr-reader #715: Discontinue MySQL support

dennissiemensma, 2019-09-19:
For DSMR-reader v3.x.

GieltjE, 2019-11-22:
Welk probleem is er precies met de tijdzones in MySQL?
Vind het zelf erg zonde als deze ondersteuning zou verdwijnen, heb een net wat zwaardere pc die naast dsmr reader nog een aantal andere services host die op MariaDB draaien waarbij er dus geforceerd twee databases simultaan moeten draaien.

Misschien kan ik helpen om een oplossing te maken.

dennissiemensma, 2019-11-24:
@GieltjE het issue met MySQL was vooral dat de DB expliciet tijdzone-ondersteuning moet hebben, omdat er anders kleine verschillen in tijden voorkomen met DST. Hoewel de tijdzones wel handmatig te installeren waren, ging de voorkeur uit naar Postgres, waar dit gewoon native in zat.

Een ander ding dat speelde is dat ik liever 1 DB-backend ondersteun, zodat ik niet op sommige plekken moet checken wat voor DB het is. Bijvoorbeeld bij backups en in bijbehorende documentatie.

Nou ga ik in DSMR-reader 3.x waarschijnlijk het DB-backup-gedeelte lostrekken buiten de applicatie en dan is het ondersteunen van andere DB's weer minder een issue. Ik laat het voor nu even open en kom er later op terug.

GieltjE, 2019-11-25:
@dennissiemensma is er voor python geen abstractielaag voor de database?
Als je in de toekomst iemand nodig hebt om te testen op b.v. MariaDB of wat uitzoekwerk gedaan wilt hebben hoor ik het graag.

dennissiemensma, 2019-11-25:
Django heeft wel een abstractie-laag, maar dat is alleen voor CRUD en niet backups, voor zover ik weet. Zoals gezegd kom ik er later op terug.

dennissiemensma, 2019-12-31:
Voor nu laat ik dit voor wat het is, maar officieel ondersteun ik alleen Postgres. In de praktijk zal het ook werken op MySQL door het ORM, maar ik kan geen garanties geven tegen issues zoals #722.

MySQL zie ik dan als een 'feature' waarbij de gebruikers zelf verantwoordelijk zijn voor backup en alles.

closed 2019-12-31 by dennissiemensma

dennissiemensma, 2019-12-31:
Ironisch genoeg kreeg ik lokaal de mysqlclient niet aan de praat. Ter logging voor mijzelf:

```
    /usr/bin/ld: cannot find -lssl
    /usr/bin/ld: cannot find -lcrypto
    collect2: error: ld returned 1 exit status
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

Opgelost met:
```
sudo apt install libssl-dev
```

GieltjE, 2021-03-02:
Klein scriptje om naar MySQL/MariaDB te migreren (ik weet dat het officieel niet meer ondersteund word, maar ik heb voor zolang het kan geen zin om 2 databases te draaien).
https://gist.github.com/GieltjE/6ae098448ab524a45916df53b3db4554
Hij berekend correct de tijdzones, zorgen dat MySQL/MariaDB op CET timezone staat, dan zou alles goed moeten gaan.