dsmrreader/dsmr-reader #716: Update van 2.0.2 naar 2.4: database bijwerken mislukt

Freez, 2019-09-19:
### Wat gebruik je?
* DSMR-reader versie 2.0.2 naar 2.4
* Hardware RaspberryPi 3
* DSMR-protocol slimme meter V5

Net geprobeerd om DSMR bij te werken naar de laatste versie maar ik krijg een foutmelding bij het uitvoeren van deploh.sh

` --- Checking & synchronizing database changes/migrations.
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying auth.0010_alter_group_name_max_length...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.ProgrammingError: must be owner of relation auth_group

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 364, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 83, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/commands/migrate.py", line 234, in handle
    fake_initial=fake_initial,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/executor.py", line 245, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/operations/fields.py", line 249, in database_forwards
    schema_editor.alter_field(from_model, from_field, to_field)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/schema.py", line 535, in alter_field
    old_db_params, new_db_params, strict)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/schema.py", line 124, in _alter_field
    new_db_params, strict,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/schema.py", line 685, in _alter_field
    params,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/schema.py", line 137, in execute
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 67, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.ProgrammingError: must be owner of relation auth_group
`
De database migraties kunnen dus niet worden uitgevoerd, als ik nu in DSMR naar statistieken ga ik de fout: column dsmr_datalogger_meterstatistics.latest_telegram does not exist

Enig idee hoe ik dit kan oplossen? Het opnieuw draaien van het script geeft dezelfde melding.

dennissiemensma, 2019-09-19:
Het lijkt een rechten issue in de database, al kan ik niet verklaren hoe dat zomaar komt. Ik zoek even hoe die rechten weer goedgezet kunnen worden.

Wat krijg je te zien bij:
```
sudo su - postgres
psql dsmrreader
\dt
```

dennissiemensma, 2019-09-19:
Ondertussen kun je nog proberen of je de vorige versie kan draaien:
```
sudo su - dsmr
git checkout tags/v2.0.2
./post-deploy.sh
```

Freez, 2019-09-20:
Output psql

dsmrreader=# \dt
                              List of relations
 Schema |                      Name                       | Type  |  Owner
--------+-------------------------------------------------+-------+----------
 public | auth_group                                      | table | postgres
 public | auth_group_permissions                          | table | postgres
 public | auth_permission                                 | table | postgres
 public | auth_user                                       | table | postgres
 public | auth_user_groups                                | table | postgres
 public | auth_user_user_permissions                      | table | postgres
 public | django_admin_log                                | table | postgres
 public | django_content_type                             | table | postgres
 public | django_migrations                               | table | postgres
 public | django_session                                  | table | postgres
 public | dsmr_api_apisettings                            | table | postgres
 public | dsmr_backup_backupsettings                      | table | postgres
 public | dsmr_backup_dropboxsettings                     | table | postgres
 public | dsmr_consumption_consumptionsettings            | table | postgres
 public | dsmr_consumption_electricityconsumption         | table | postgres
 public | dsmr_consumption_energysupplierprice            | table | postgres
 public | dsmr_consumption_gasconsumption                 | table | postgres
 public | dsmr_datalogger_dataloggersettings              | table | postgres
 public | dsmr_datalogger_dsmrreading                     | table | postgres
 public | dsmr_datalogger_meterstatistics                 | table | postgres
 public | dsmr_datalogger_retentionsettings               | table | postgres
 public | dsmr_frontend_frontendsettings                  | table | postgres
 public | dsmr_frontend_notification                      | table | postgres
 public | dsmr_mindergas_mindergassettings                | table | postgres
 public | dsmr_mqtt_jsondaytotalsmqttsettings             | table | postgres
 public | dsmr_mqtt_jsontelegrammqttsettings              | table | postgres
 public | dsmr_mqtt_message                               | table | postgres
 public | dsmr_mqtt_mqttbrokersettings                    | table | postgres
 public | dsmr_mqtt_rawtelegrammqttsettings               | table | postgres
 public | dsmr_mqtt_splittopicdaytotalsmqttsettings       | table | postgres
 public | dsmr_mqtt_splittopicmeterstatisticsmqttsettings | table | postgres
 public | dsmr_mqtt_splittopictelegrammqttsettings        | table | postgres
 public | dsmr_notification_notificationsetting           | table | postgres
 public | dsmr_notification_statusnotificationsetting     | table | postgres
 public | dsmr_pvoutput_pvoutputaddstatussettings         | table | postgres
 public | dsmr_pvoutput_pvoutputapisettings               | table | postgres
 public | dsmr_stats_daystatistics                        | table | postgres
 public | dsmr_stats_electricitystatistics                | table | postgres
 public | dsmr_stats_hourstatistics                       | table | postgres
 public | dsmr_stats_note                                 | table | postgres
 public | dsmr_weather_temperaturereading                 | table | postgres
 public | dsmr_weather_weathersettings                    | table | postgres
(42 rows)

Terug naar de vorige versie geeft dezelfde fout: django.db.utils.ProgrammingError: must be owner of relation auth_group

Freez, 2019-09-20:
Inmiddels versie 2.0.2 weer werkend gekregen door de eigenaar van de tabellen aan te passen in dsmrreader. Vanavond probeer ik de laatste versie te installeren.

dennissiemensma, 2019-09-20:
Top! Fijn om te horen dat je het zelf al gevonden had. Mocht je alsnog meldingen krijgen vanavond, dan hoor ik het graag!

closed 2019-09-20 by dennissiemensma