dsmrreader/dsmr-reader #717: archief

style2k, 2019-09-20:
* DSMR-reader versie *: `v2.4
ik heb een pi 3 en had daar in januarie dsrm opgezet . waarom weet ik niet maar 6 dagen geleden is het hele handeltje blijkbaar gestopt want de laatste backup is van 6 dagen geleden . 
nu heb ik nieuw sd kaart gehaald en alles opnieuw geinstalleerd .
heb het nu werkend alleen heb ik dus wel me backup terug gezet en dat heeft blijkbaar gewerkt want mijn settings etc zijn terug gezet .
alleen mijn archief is weg ik zie januarie en februarie maar verder nix . en ook zie ik alleen wat ik verbruikt heb maar de totaal bedragen staan er niet bij. 

hoe kan dit kreeg ook geen foutmelding bij het restoren van de backup .
en komt er ook ondersteuning voor one drive want heb wel dropbox maar zie het liever terug op onedrive :)

gr. rene .

dennissiemensma, 2019-09-21:
Wat krijg je te zien als je het volgende doet:

```
sudo su - postgres
psql dsmrreader

# Dagen in archief
select day from dsmr_stats_daystatistics order by day desc;

# Dagen met verbruik
select date(read_at) as day from dsmr_consumption_electricityconsumption group by day order by day desc;
```

Zie je bij een van beiden recente dagen staan?

style2k, 2019-09-21:
net thuis en nu lijken de verbruiks waarden er wel allemaal ingevuld te zijn. alleen de prijzen is nog niet helemaal goed . heb in mei een nieuw abo gehad en de nieuwe prijs ingevoerd en de nieuwe prijzen vanaf mei staan er allemaal in 
en wat ik zie als ik die commandos invoer is dit 
pi@raspberrypi:~ $ sudo su - postgres
postgres@raspberrypi:~$ psql dsmrreader
psql (11.5 (Raspbian 11.5-1+deb10u1))
Type "help" for help.

dsmrreader=#

dennissiemensma, 2019-09-21:
Ik mis een stukje output in je bericht hierboven.

Hoe dan ook, als je prijzen hebt aangepast voor historische gegevens, kun je ze opnieuw laten uitrekene met: https://dsmr-reader.readthedocs.io/nl/latest/faq.html#recalculate-prices-retroactively

style2k, 2019-09-21:
heb ff in de instellingen gekeken . stonden 2 tarieven in maar de oude tarieven stonden op gestart 1 mei dat heb ik aangepast naar januarie . 
moet ik dan evengoed dat invoeren om te hertellen of gaat dat ook automatisch weer ingevuld worden ?

dennissiemensma, 2019-09-21:
Ik denk dat je het beste alle tarieven kunt invullen zoals ze waren/goed zijn en dan opnieuw laat berekenen met bovenstaande. 
Je kunt dan de eerstvolgende backup even apart opslaan, dan heb je altijd een goede kopie van alles.

style2k, 2019-09-21:
ik word hier echt niet goed van nu vraagt die weer het dsrm ww en ik heb de instalatie handleiding gewoon precies gevolgd maar heb daar geen ww in hoeven voeren steeds en nu kom ik daar dus niet in

dennissiemensma, 2019-09-21:
Ergens in de installatie-stappen staat dat je deze twee regels in je bashrc horen te staan:
```
source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader
```

Ik vermoed dat die missen. Je kunt ze toevoegen **vanuit `pi` of `root` gebruiker** met:
```
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'
```

Als je daarna dit doet (inlogt als `dsmr`), krijg je die foutmelding niet meer:
```
sudo su - dsmr
```

style2k, 2019-09-21:
ok top werkt . alleen wat is er nu berekend ? bedragen staan ook vanaf januarie weer bij alleen de bedragen zoals bedrag van september stond op 13 nogwat en staat nu op 12 nogwat dus heeft die daar ook berekeningen toegepast ? want dat is als het goed is vanav mei al met het goeie tarief verekend ??

dennissiemensma, 2019-09-21:
Het enige wat je hoeft te doen is ervoor zorgen dat je alle prijzen van het afgelopen jaar goed hebt staan. Als die contracten goed ingevuld zijn kun je het hergenereren. De applicatie herberekent dan alle dagen.

Kloppen de uitkomsten niet?

style2k, 2019-09-22:
heb ff de rekenmachine erbij gepakt . verbruikt 31,63 aan stroom kosten 0.21094 is dus 6,6720322 euro.
hij geeft aan in dsrm 6,68 euro. verschil is niet heel groot maar hij heeft em nu naar boven afgerond terwijl je zou zeggen naar beneden afronden .scheelt nu toch ong een cent en dat is in 1 geval als je alles doorrekent kan het bedrag dus heel anders uit komen ..dit gaat over de huidige prijs de prijzen van januarie tot mei klopt het wel en zijn de prijzen wel goed afgerond .

dennissiemensma, 2019-09-22:
Bedankt voor je melding. Dat was me nog niet eerder opgevallen. Ik heb het kunnen achterhalen en in de volgende versie is het gefixt.

closed 2019-09-22 by dennissiemensma

dennissiemensma, 2019-10-01:
Uitgebracht in [v2.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.5.0)