dsmrreader/dsmr-reader #719: Fout tijdens installatie: Stap 8 > ./manage.py migrate

dimitrizetzsche, 2019-09-26:
Hoi ik was dit project aan het installeren via de toegelichte installatie procedure. Tot stap 8 ging alles vlot. Ik loop vast op volgend commando: 
`./manage.py migrate`

dat resulteert in deze foutmelding: 
`Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/__init__.py", line 260
    subcommands = [*get_commands(), 'help']
                          ^
SyntaxError: can use starred expression only as assignment target`

Iemand een idee waarom dit fout loopt? Alvast bedankt.

dennissiemensma, 2019-09-26:
Ik zie dat je `python3.4` gebruikt. Deze wordt helaas niet meer ondersteund. Je kunt alleen Python 3.5+ draaien voor dit project. Ik raad je aan om zelfs minimaal 3.6 te installeren.

Welk OS gebruik je? 
```
lsb_release -a
```

dimitrizetzsche, 2019-09-27:
@dennissiemensma  bedankt voor de snelle reactie. 

Ik hebt python 3.6 geïnstalleerd en als mijn default versie gezet om te gebruiken: 
`$ python -V
Python 3.6.0`

Ik blijf helaas op hetzelfde probleem stoten. Mogelijk 

`$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie`

dennissiemensma, 2019-09-27:
Je mist nog wat informatie, ik had niet gedacht dat je het direct al werkend had:

* Virtualenvs bevatten een **kopie** van Python, dus als je 3.4 hebt gebruikt om hem eerder aan te maken, staat die er nog hardcoded in.

De oplossing is waarschijnlijk om de virtualenv opnieuw aan te maken, zodat die 3.6 gebruikt.

```
sudo su - dsmr

# Mogelijk doet dit niets, maar is niet erg:
deactivate

rm -rf ~/.virtualenvs/dsmrreader/

virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3

source ~/.virtualenvs/dsmrreader/bin/activate
```

Controleer nu de Python-versie in de virtualenv:
```
python --version
```

Als dat nu goed is, doe onderstaande en ga dan verder waar je gebleven was:
```
pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt

./manage.py migrate

...
```

dimitrizetzsche, 2019-09-27:
Ik vermoed dat ik een loop van errors zit :-/.

Het uitvoeren van commando
`virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3`
 
gaf me dit
``` 
Running virtualenv with interpreter /usr/local/bin/python3
Using base prefix '/usr/local'
New python executable in /home/dsmr/.virtualenvs/dsmrreader/bin/python3
Also creating executable in /home/dsmr/.virtualenvs/dsmrreader/bin/python
Installing setuptools, pip...
  Complete output from command /home/dsmr/.virtuale...mrreader/bin/python3 -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip:
  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 646, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 616, in _load_backward_compatible
  File "/usr/share/python-wheels/pip-1.5.6-py2.py3-none-any.whl/pip/__init__.py", line 72, in <module>
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 646, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 616, in _load_backward_compatible
  File "/usr/share/python-wheels/pip-1.5.6-py2.py3-none-any.whl/pip/log.py", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 646, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 616, in _load_backward_compatible
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/pkg_resources.py", line 1520, in <module>
AttributeError: module 'importlib._bootstrap' has no attribute 'SourceFileLoader'
----------------------------------------
...Installing setuptools, pip...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2378, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 830, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 1032, in create_environment
    install_wheel(to_install, py_executable, search_dirs)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 975, in install_wheel
    'PIP_NO_INDEX': '1'
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 917, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/dsmr/.virtuale...mrreader/bin/python3 -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip failed with error code 1
``` 

Ik heb dit even genegeerd en dan volgende commando uitgevoerd
`source ~/.virtualenvs/dsmrreader/bin/activate`

wat me dit gaf: 
`-su: /home/dsmr/.virtualenvs/dsmrreader/bin/activate: No such file or directory`

uiteindelijk krijg ik nu wel de juiste python versie: 
```
$ python --version
Python 3.6.0
```

Uitvoeren van 
```pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt``` 

eindigde met volgende error:
```
Command "/usr/local/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-wy7d0zbu/crcmod/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-qtg7g8oo-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-wy7d0zbu/crcmod/
```

dennissiemensma, 2019-09-27:
Je hoeft de rest niet meer te proberen als het aanmaken van de virtualenv al faalt. Het is wat lastig om van afstand te debuggen, welke versie krijg je hierbij te zien:

```
sudo su - dsmr
/home/dsmr/.virtualenvs/dsmrreader/bin/python3 --version
```

dimitrizetzsche, 2019-09-30:
```
sudo su - dsmr
```
geeft me 
```-su: /home/dsmr/.virtualenvs/dsmrreader/bin/activate: No such file or directory```

``` 
/home/dsmr/.virtualenvs/dsmrreader/bin/python3 --version
```
 geeft me
```
Python 3.6.0
```

dimitrizetzsche, 2019-09-30:
ik heb het opgelost gekregen door 
```
sudo su - dsmr
source env/bin/activate
virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3
source ~/.virtualenvs/dsmrreader/bin/activate
```

En vanaf hier verder met 
```
pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt
./manage.py migrate
```

Alles gaf output 'OK'

closed 2019-09-30 by dimitrizetzsche

dennissiemensma, 2019-09-30:
Fijn, bedankt voor je terugkoppeling!