dsmrreader/dsmr-reader #720: vast gelopen

sjaakveenendaal, 2019-09-27:
Dag,

De software op de sd kaart werkt niet meer nu wil ik de data veilig stellen voordat ik dsmr reader opnieuw ga installeren maar waar kan ik de data vinden ?

dennissiemensma, 2019-09-27:
De data staat standaard in `/var/lib/postgres/`, maar als je SD_kaart niet meer werkt vraag ik me af of je er nog bij kan. Heb je geen backup's buiten de kaart om?

sjaakveenendaal, 2019-09-28:
Hoi Dennis,

Nee geen data meer gewoon dom dat ik geen backup heb gemaakt.

Vandaag nieuw SD kaartje gekocht en geprobeerd om DSMR opnieuw te
installeren maar weet niet wat ik fout doe misschien kan jij zien wat er
misgaat ?

pi@raspberrypi:~ $ # Packages
pi@raspberrypi:~ $ sudo apt-get install -y postgresql
postgresql-server-dev-all nginx supervisor git python3 python3-pip
python3-virtualenv virtualenvwrapper
Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd
De statusinformatie wordt gelezen... Klaar
git is reeds de nieuwste versie (1:2.20.1-2).
python3 is reeds de nieuwste versie (3.7.3-1).
python3-pip is reeds de nieuwste versie (18.1-5+rpt1).
De volgende extra pakketten zullen geïnstalleerd worden:
  binfmt-support clang-7 dctrl-tools libclang-common-7-dev libclang1-7
  libffi-dev libgc1c2 libllvm7 libncurses-dev libnginx-mod-http-auth-pam
  libnginx-mod-http-dav-ext libnginx-mod-http-echo libnginx-mod-http-geoip
  libnginx-mod-http-image-filter libnginx-mod-http-subs-filter
  libnginx-mod-http-upstream-fair libnginx-mod-http-xslt-filter
  libnginx-mod-mail libnginx-mod-stream libobjc-8-dev libobjc4 libomp-7-dev
  libomp5-7 libpq-dev libtinfo-dev llvm-7 llvm-7-dev llvm-7-runtime
  nginx-common nginx-full postgresql-11 postgresql-client-11
  postgresql-client-common postgresql-common postgresql-server-dev-11
  python-meld3 python-stevedore python-virtualenv python-virtualenv-clone
  ssl-cert sysstat virtualenv virtualenv-clone
Voorgestelde pakketten:
  clang-7-doc debtags ncurses-doc libomp-7-doc postgresql-doc-11 llvm-7-doc
  fcgiwrap nginx-doc postgresql-doc libjson-perl openssl-blacklist
  supervisor-doc isag
De volgende NIEUWE pakketten zullen geïnstalleerd worden:
  binfmt-support clang-7 dctrl-tools libclang-common-7-dev libclang1-7
  libffi-dev libgc1c2 libllvm7 libncurses-dev libnginx-mod-http-auth-pam
  libnginx-mod-http-dav-ext libnginx-mod-http-echo libnginx-mod-http-geoip
  libnginx-mod-http-image-filter libnginx-mod-http-subs-filter
  libnginx-mod-http-upstream-fair libnginx-mod-http-xslt-filter
  libnginx-mod-mail libnginx-mod-stream libobjc-8-dev libobjc4 libomp-7-dev
  libomp5-7 libpq-dev libtinfo-dev llvm-7 llvm-7-dev llvm-7-runtime nginx
  nginx-common nginx-full postgresql postgresql-11 postgresql-client-11
  postgresql-client-common postgresql-common postgresql-server-dev-11
  postgresql-server-dev-all python-meld3 python-stevedore python-virtualenv
  python-virtualenv-clone python3-virtualenv ssl-cert supervisor sysstat
  virtualenv virtualenv-clone virtualenvwrapper
0 opgewaardeerd, 49 nieuw geïnstalleerd, 0 te verwijderen en 0 niet
opgewaardeerd.
Er moeten 68,9 MB aan archieven opgehaald worden.
Na deze bewerking zal er 351 MB extra schijfruimte gebruikt worden.
Ophalen:1 http://raspbian.mirror.liteserver.nl/raspbian buster/main armhf
libgc1c2 armhf 1:7.6.4-0.4 [212 kB]
Fout:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf libllvm7
armhf 1:7.0.1-8+rpi1
  404  Not Found [IP: 93.93.128.193 80]
Ophalen:2 http://raspbian.mirror.liteserver.nl/raspbian buster/main armhf
binfmt-support armhf 2.2.0-2 [56,7 kB]
Fout:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf
libclang-common-7-dev armhf 1:7.0.1-8+rpi1
  404  Not Found [IP: 93.93.128.193 80]
Ophalen:4 http://raspbian.mirror.liteserver.nl/raspbian buster/main armhf
libobjc4 armhf 8.3.0-6+rpi1 [43,0 kB]
Ophalen:5 http://raspbian.mirror.liteserver.nl/raspbian buster/main armhf
libobjc-8-dev armhf 8.3.0-6+rpi1 [207 kB]
Fout:7 http://raspbian.raspberrypi.org/raspbian buster/main armhf
libclang1-7 armhf 1:7.0.1-8+rpi1
  404  Not Found [IP: 93.93.128.193 80]
Fout:8 http://raspbian.raspberrypi.org/raspbian buster/main armhf clang-7
armhf 1:7.0.1-8+rpi1
[... 196 lines cut ...]
is mislukt  404  Not Found [IP: 93.93.128.193 80]
E: Ophalen van
http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-11/postgresql-11_11.4-1_armhf.deb
is mislukt  404  Not Found [IP: 93.93.128.193 80]
E: Ophalen van
http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-common/postgresql_11+200+deb10u1_all.deb
is mislukt  404  Not Found [IP: 93.93.128.193 80]
E: Ophalen van
http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-11/postgresql-server-dev-11_11.4-1_armhf.deb
is mislukt  404  Not Found [IP: 93.93.128.193 80]
E: Ophalen van
http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-common/postgresql-server-dev-all_200+deb10u1_all.deb
is mislukt  404  Not Found [IP: 93.93.128.193 80]
E: Kon sommige archieven niet ophalen. Misschien kunt u 'apt-get update'
uitvoeren of het met '--fix-missing' proberen?

met vriendelijke groet,

Sjaak van Veenendaal

Dubceksingel 162
6716 RE Ede

0318 - 220091
06 - 83176418

dennissiemensma, 2019-09-28:
Je kunt de suggestie proberen, meestal fixt dat het:

```
sudo apt-get update
```

sjaakveenendaal, 2019-09-28:
Hoi Dennis,

 

Dat had ik al geprobeerd maar na nog 3x is het gelukt.

 

Wel alle gegevens kwijt van bijna een jaar maar goed dat is mijn eigen fout..

 

Bedankt voor je hulp top..

 

Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

jeroenpeters1986, 2019-09-28:
@sjaakveenendaal Bij mij heeft de SD kaart het ook begeven in de zin dat de Pi niet meer wilde booten. Heb je de kaart nog wel geprobeerd in te lezen via een desktop pc? Soms kun je dan toch nog bij de mappen. Dit werkt het beste vanuit een Linux omgeving

sjaakveenendaal, 2019-09-28:
bij was hij echt helemaal stuk dus jammer.

met vriendelijke groet,

Sjaak van Veenendaal

Dubceksingel 162
6716 RE Ede

0318 - 220091
06 - 83176418

jeroenpeters1986, 2019-09-28:
Helaas! Ik dacht ik geef het toch nog een keer aan, ik kon het niet met
zekerheid teruglezen in je ticket. Het zou zonde zijn als dat toch een
optie was geweest voor je.

sjaakveenendaal, 2019-09-28:
Doe je goed

closed 2019-09-29 by dennissiemensma