dsmrreader/dsmr-reader #721: replaced smartmeter, no readings

RezzZ, 2019-09-27:
*Bedankt voor het aanmaken van een issue!*
### Wat gebruik je?
* DSMR-reader versie: `v2.4`
* Hardware RaspberryPi 3B+`
* DSMR-protocol slimme meter: `v5`

### Bij fouten
Gisteren een nieuwe slimme meter gekregen omdat de oude niet met de netwerkbeheerder communiceerde (nu Iskea SMR5.0, was Landis+Gyr SMR5.0) maar ik krijg nu geen nieuwe standen meer door in DSMR-Reader. Ik las al oudere issues dat je een dag moet wachten maar de dag is voorbij en krijg nog steeds geen nieuwe data binnen.

Vreemd genoeg zegt dsmr reader dat het 2 maanden, 3 weken geleden is sinds de laatste meting (ik had de kabel er vanaf het begin inzitten, vrees dat er een tijdsynchronisatie nog moest plaatsvinden vandaar deze melding) maar er komt nu niets meer binnen.

Is er iets wat ik nog moet instellen/resetten?

dennissiemensma, 2019-09-27:
Kun je een telegram delen? Deze staat onderaan deze pagina: `/admin/dsmr_datalogger/meterstatistics/1/change/`

RezzZ, 2019-09-27:
```
/ISK5\2M550T-1011

1-3:0.2.8(50)
0-0:1.0.0(190927230959S)
0-0:96.1.1(4530303334303037343430383239303139)
1-0:1.8.1(000014.049*kWh)
1-0:1.8.2(000004.855*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.351*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00005)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(190703012111S)(0000000333*s)
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(236.5*V)
1-0:52.7.0(234.7*V)
1-0:72.7.0(236.8*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.126*kW)
1-0:41.7.0(00.093*kW)
1-0:61.7.0(00.131*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353635353136383531313137)
0-1:24.2.1(190927230509S)(03990.009*m3)
!E878
```

RezzZ, 2019-09-27:
en ik zie dat ik sinds 16 uur vandaag weer goede statistieken krijg.. misschien toch een datum/tijd dingetje dat nog even via gprs moest worden geüpdatet?

dennissiemensma, 2019-09-27:
De tijden van je meter en gasmeter lijken wel te kloppen:

```
190927230959S
190927230509S
```

Het kan inderdaad zijn dat het net vandaag is gefixt.

dennissiemensma, 2019-09-27:
Mocht je later alsnog weer afwijkende waarden zien, dan hoor ik het graag!

closed 2019-09-27 by dennissiemensma

RezzZ, 2019-09-27:
dank iig voor je hulp!