dsmrreader/dsmr-reader #722: Minimale backup (sinds v2.3.0) laat processen stoppen bij MySQL gebruikers

jeroenpeters1986, 2019-09-28:
### Mijn Setup
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.4.0`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v4`
* Database type: `mysql`

### Melding
Ik had DSMR-Reader v2.1.0 geïnstalleerd. Helaas ging mijn SD kaartje was recent helaas stuk. Geen probleem, met de backup was het zo hersteld. Toen had ik wel v2.4.0. De load op mijn Pi ging aardig omhoog en ik kreeg geen notificaties (afgezien van de melding dat er al lang geen reading was binnengekomen, vanwege het terugplaatsen backup). 

Na een aantal dagen zonder verbruiksnotificatie op mijn telefoon (terwijl het testen wel goed werkte) ben ik pas op onderzoek uitgegaan. Wat blijkt, doordat ik mysql gebruik komt er een melding: "Unsupported backend". Daarna gaan de backendprocessen niet verder (zoals een notificatie). Ik kwam met die melding eerst uit op de normale backup, maar die werd gewoon gemaakt. Het blijkt de minimal backup te zijn.

Ik zag in #715 dat MySQL backup er vanaf v3 uit gaat. Ik geloof dat  ik de enige MySQL gebruiker ben en ik waardeer het dat het al zo lang op beide backends werkt. Daarom heb ik er nu toch dit issue voor gemaakt. Ik zal zelf de codewijzigngen maken en een Pull Request opsturen.

jeroenpeters1986, 2019-09-29:
Ik zie ook dat bij de geplande taken de mailbackup staat. Als ik me niet vergis is de mailbackup de "partial" backup. Als je die dus verzet naar bijv 2029 zou het misschien al niet meer voorkomen.

dennissiemensma, 2019-09-29:
Ik denk dat ik het wel kan fixen door de mysql-backup toe te voegen. Het is een kleine moeite en bijna een copy paste van de volledige back-up variant

jeroenpeters1986, 2019-09-29:
Ja, goed idee! Zo ga ik dat doen idd

dennissiemensma, 2019-09-29:
Ik heb inmiddels al een opzet voor je werkend. Ik zal het meenemen in de dev versie straks

jeroenpeters1986, 2019-09-29:
Oh tof! Ik was nog niet begonnen omdat ik nog fysieke dingen te klussen had ipv digitaal :-P
Ik zal de branch uitchecken wanneer het erin staat

dennissiemensma, 2019-09-29:
Het saat nu in `development`

jeroenpeters1986, 2019-09-29:
Top! Ik heb het gecheckt hier en het werkt goed! Het migraten duurde even
maar ik had de backup al vrij snel in de mail.

notifications@github.com>

dennissiemensma, 2019-09-29:
Top, bedankt voor proberen!

closed 2019-09-29 by dennissiemensma

dennissiemensma, 2019-10-01:
Uitgebracht in [v2.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.5.0)