dsmrreader/dsmr-reader #724: MQTT - delete messages

remcoongit, 2019-10-01:
*Bedankt voor het aanmaken van een issue!*
### Wat gebruik je?
v2.4.0 - 2019-09-19
VM debian
protocol v2/v4 maakt niet zoveel uit

### Bij fouten
http://ip/status
MQTT messages | Queued messages: 14,712,528
Ik was vergeten mqtt berichten uit te zetten. Heb zelf in homeassistant een script gemaakt die via http de status ophaalt.

Hoe kan ik die berichten weggooien?

dennissiemensma, 2019-10-01:
```
sudo su - postgres
psql dsmrreader

# Query
truncate table dsmr_mqtt_message;
```

remcoongit, 2019-10-01:
Worked like a charm!

closed 2019-10-01 by remcoongit

dennissiemensma, 2019-10-01:
Ik zal er sowieso voor zorgen dat dit in een volgende release niet meer gebeurt. Het heeft geen zin om 14M berichten in te plannen als die toch niet verstuurd worden namelijk.