dsmrreader/dsmr-reader #725: overwerkte telegrammen

style2k, 2019-10-01:
[image]

ik kijk in het status scherm en zie dit , heeft dit invloed op de metingen ??

dennissiemensma, 2019-10-01:
Je kunt kijken of je wat ziet in de log van `dsmr_backend`: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

style2k, 2019-10-01:
[dsmr_backend.log](https://github.com/dennissiemensma/dsmr-reader/files/3677815/dsmr_backend.log)

dit is de log file ik zie iets van een netwerk maar de pi heeft gewoon verbinding met het netwerk

dennissiemensma, 2019-10-01:
Hij probeert verbinding te maken met de mailserver die je insteld hebt vermoed ik. Kan het zijn dat daar wat mee is?

Desnietemin zou dat niet de rest moeten ophouden, dus ik zal daar wat dieper naar kijken.

dennissiemensma, 2019-10-01:
Ik heb het even geprobeerd en ondanks de foutmelding, zou die wel telegrammen moeten verwerken, zoals bedoeld. Je kunt de logging op `DEBUG` zetten (zie eerdere link) voor meer informatie over wat het proces exact doet.

style2k, 2019-10-01:
o ok naja heb email config ff uitgezet en heb nog precies hetzelfde/
kan de file niet wegschrijven  na ik die aangepast heb
is dat niet in dsmr  te zetten dat je een logfile kan zien ofzo . dat command line gebeuren heb ik echt nix mee

style2k, 2019-10-01:
ik zie overigens wel dat de onverwerkte telegrams afnemen heel langzaam nadat ik dus de email backup heb uitgezet

jeroenpeters1986, 2019-10-01:
@style2k, deed de backups via een eigen server of via Gmail? Ik heb voor
gmail andere instellingen gebruikt namelijk omdat ik bij de instellingen
uit de suggestie je een foutmelding krijgt achter een IPv6 ipadres (wat ik
thuis uitgaand heb).

style2k, 2019-10-01:
via een gmail adres en daar heb ik de gegevens ingevuld die er bij stonden 

Heb je GMail? Voer dan als host aspmx.l.google.com in, als poort 25 en gebruik TLS. Voer vervolgens niet je gebruikersnaam of wachtwoord in. Vul je GMail-adres hierboven in als ontvanger

jeroenpeters1986, 2019-10-01:
Ah okay. Ik kreeg daarmee een foutmelding (niet dezelfde als jij).

Ik gebruik nu als host `smtp.gmail.com`, met poort `587` en met mijn eigen username en wachtwoord van het (speciaal voor dit soort systemen gemaakte) e-mailaccount.

style2k, 2019-10-01:
die settings werken wel idd ff een app wachtwoord aangemaakt bij google die ingevuld en die test komt ook goed door . dus de backup ook weer aangezet . 
er zijn nog steeds onverwerkte telegrams maar de aantallen lopen nog steeds naar beneden straks nog is kijken of dat het uiteindelijk weer oplost dan :)

style2k, 2019-10-01:
de errors zijn weg nu . dus het probleem was idd de email instelling.

closed 2019-10-02 by dennissiemensma