dsmrreader/dsmr-reader #730: Tijdsrange graphs dashboard instelbaar en onderling koppelen

adebree, 2019-10-07:
*Bedankt voor het aanmaken van een issue!*
### Wat gebruik je?
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.5`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Intel NUC met docker xirixiz/dsmr-reader-docker`
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v5`

### Overige
Twee suggesties: 

1) Wat ik erg fijn zou vinden is om de graphs op het dashboard met een geselecteerde range van 24 uur (max dus) te openenen. Ik heb nu de indruk dat de tijdsrange bepaald wordt door een aantal datapunten oid, wat als gevolg heeft dat de graphs die er staan allemaal een andere tijdsrange hebben. Standaard openen met een gelijke range geeft mij sneller een overview over wat er nu en in het recente verleden gebeurd is.

2) De selected range van de graphs op het dashboard zijn aanpasbaar. Wat ik erg fijn zou vinden is een manier waarop de tijdsrange van alle grafieken tegelijk zou wijzigen. Daarbij maakt het me niet uit of dat via een losse control of bij het aanpassen van één van de grafieken gecontroleerd kan worden.

Minor improvements wat mij betreft!

PS: complimenten voor je software, en de manier hoe je met feedback van je gebruikers omgaat. Keep that up!

dennissiemensma, 2019-10-07:
Bedankt voor je suggesties en vriendelijke woorden. Op dit moment toont het dashboard al de afgelopen 24 uur (als je die zelf op max zet met de sliders). De standaard-rage bij openen wordt inderdaad bepaald door het aantal datapunten, zodat de grafieken standaard alleen recente informatie tonen.

Ik kan helaas de ranges niet gelijktrekken, want het is erg afhankelijk of iemand een DSMR v4 of v5 meter heeft. En daarnaast speelt de groepering van data ook een rol. 
Met een v4 meter komt er slechts 1x per uur een nieuwe gasstand binnen en dat zou betekenen dat de grafiek vertekend raakt t.o.v. elektriciteit, die zowel per seconden als per minuut ingesteld kan worden.

Daar heb ik op dit moment geen goede oplossing voor denk ik.

adebree, 2019-10-08:
Ai, dat is spijtig. Wellicht dat ik je verkeerd begrijp, maar zou het wel mogelijk zijn om de volledige 'tijds' range van alle graphs programmatisch te selecteren? Ongeacht hoeveel data punten dat zijn? Dat is namelijk precies wat ik nu met de hand aan het doen ben per graph. Na het openen van het dashboard sleep ik dan bij alle graphs de linker slider helemaal naar links.

dennissiemensma, 2019-10-08:
Daar is opzich wel wat voor te maken. Nu toont die standaard alleen het laatste deel.

dennissiemensma, 2019-12-18:
In de admin-instellingen bij de Frontend-configuratie is een instelling waarbij je nu al de standaard 'breedte' van de grafieken kan instellen. 

De max is echter nu 120 items. In de volgende release wordt dat 86400 items (60 x 60 x 24), zodat je het zelf kan instellen naar een standaard groter bereik.

closed 2019-12-18 by dennissiemensma

dennissiemensma, 2020-01-05:
Uitgebracht in v2.13.0