dsmrreader/dsmr-reader #731: weer ee nfout met telegrams

style2k, 2019-10-08:
dsmr 2.6

### Bij fouten
ik had tijdje terug een probleem met de telegrams . nu is het nog erger en doet die volgens mij nix meer .
ik heb de mail configuratie getest en daar die het vorige keer niet deed komt die nu wel door de test .
maar krijg weer telegram fouten en dsmr is al 7 uur plus niet meer geupdate ..
[image]

ook klopt de tijdlijn niet meer ,
[image]

ik heb inmiddels de update gedaan en stata nu dus op 2.6 maar geen idee wat er nu weer aan de hand is . dit is de log file dsmr backend die ik er net heb afgehaald

[dsmr_backend.log](https://github.com/dennissiemensma/dsmr-reader/files/3703622/dsmr_backend.log)

wie weet zie jij wat . ik zie wel weer een can not reacht netwerk ding zoals vorige keer maar de mail test in Django werkt gewoon en de mail komt gewoon binnen dus dat werkt gewoon.

gr. rene

dennissiemensma, 2019-10-08:
Aan het einde zie ik wat fouten met de verbinding naar de database, maar dat is vreemd want die lijkt gewoon op dezelfde Pi te draaien.

Heb je toevallig nog andere zaken op de Pi draaien? In een eerder ticket gaf je aan dat je een Pi 3 hebt en die is meer dan snel genoeg om DSMR-reader te draaien.

Een andere oorzaak zou nog een trage SD-kaart kunnen zijn, maar dat komt zelden voor.

style2k, 2019-10-08:
nee draait nix anders dan dsrm en de database

dennissiemensma, 2019-10-08:
Je kunt nogmaals proberen om DEBUg-logging in te schakelen voor meer informatie: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

En dan na het herstarten van de processen inhoudelijk kijken wat `/var/log/supervisor/dsmr_backend.log` nog meer meldt en hoelang die over iets doet.

style2k, 2019-10-08:
ff gekeken maar die stond nog op debuging

dennissiemensma, 2019-10-08:
Als debugging goed aan staat, zou je dit soort informatie in de logfile moeten terug zien:
```

[2019-10-08 20:43:29,991] DEBUG    Sleeping 1 sec(s)
[2019-10-08 20:43:31,014] DEBUG    Calling 0 backend service(s)
[2019-10-08 20:43:31,044] DEBUG    Sleeping 1 sec(s)
[2019-10-08 20:43:32,063] DEBUG    Calling 0 backend service(s)
[2019-10-08 20:43:32,095] DEBUG    Sleeping 1 sec(s)
[2019-10-08 20:43:33,115] DEBUG    Calling 0 backend service(s)
[2019-10-08 20:43:33,159] DEBUG     - Processed reading: 11757441 @ 2019-10-08 20:42:07+02:00 (1.416 kW)
[2019-10-08 20:43:33,168] DEBUG     - Processed reading: 11757442 @ 2019-10-08 20:42:17+02:00 (1.642 kW)
[2019-10-08 20:43:33,178] DEBUG     - Processed reading: 11757443 @ 2019-10-08 20:42:27+02:00 (0.265 kW)
[2019-10-08 20:43:33,193] DEBUG     - Processed reading: 11757444 @ 2019-10-08 20:42:37+02:00 (1.395 kW)
[2019-10-08 20:43:33,203] DEBUG     - Processed reading: 11757445 @ 2019-10-08 20:42:47+02:00 (1.636 kW)
[2019-10-08 20:43:33,212] DEBUG     - Processed reading: 11757446 @ 2019-10-08 20:42:57+02:00 (0.251 kW)
[2019-10-08 20:43:33,228] DEBUG    Sleeping 1 sec(s)
[2019-10-08 20:43:34,252] DEBUG    Calling 0 backend service(s)
[2019-10-08 20:43:34,273] DEBUG    Sleeping 1 sec(s)
[2019-10-08 20:43:35,288] DEBUG    Calling 0 backend service(s)
[2019-10-08 20:43:35,307] DEBUG    Sleeping 1 sec(s)
[2019-10-08 20:43:36,332] DEBUG    Calling 0 backend service(s)
```

Je moet daarvoor het backend-proces herstarten.

Je zou er altijd bij moeten kunnen als `dsmr` gebruiker op de Pi:
```
sudo su - dsmr

# Bestand aanpassen

./post-deploy.sh
```

style2k, 2019-10-08:
ik pas het bestand aan via de pi 3 desktop maar kan em niet overschrijven .

dennissiemensma, 2019-10-08:
https://www.raspberrypi.org/magpi/edit-text/

style2k, 2019-10-08:
die had ik al gevonden maar goed veranderd nu krijg ik steeds bad gateway
[image]

dennissiemensma, 2019-10-08:
Wat krijg je als je dit doet?

```
sudo su - dsmr
python dsmrreader/settings.py
```

Toevallig iets als `SyntaxError: invalid syntax`?

style2k, 2019-10-08:
File "dsmrreader/settings.py", line 42
    DEBUG LOGGING['loggers']['commands']['level'] = 'DEBUG'
                ^
SyntaxError: invalid syntax

dennissiemensma, 2019-10-08:
Er staat een typefout in de regel. Je kunt hem overnemen zoals die in de documentatie staat:

```
LOGGING['loggers']['commands']['level'] = 'DEBUG'
```

style2k, 2019-10-08:
ja ben er beetje klaar mee krijg nog steeds hetzelfde . bad gateway

dennissiemensma, 2019-10-08:
Heb je de processen herstart na je laatste wijziging?

style2k, 2019-10-08:
ja en gaf geen errors zoals de vorige keer . geen syntax errors te zien

style2k, 2019-10-08:
[dsmr_backend.log](https://github.com/dennissiemensma/dsmr-reader/files/3704335/dsmr_backend.log)

dennissiemensma, 2019-10-08:
Ok hij is duidelijk bezig om de achterstand in te halen, maar dat gaat niet heel snel. Hij verwerkt ongeveer 1 meting per twee seconden, maar elke seconde komt er weer een meting bij.

Hoewel ik niet kan zien waarom die traag is, kun je meerdere dingen doen:
* Of groeperen per minuut aanzetten en kijken of dat sneller gaat.
* Of minder metingen per minuut binnen laten komen. Bijvoorbeeld 20 of 30 ipv 60.

Nog een voorkeur voor iets?

style2k, 2019-10-08:
ja want in dashboard gis die bij 20 over 11 ofzo en gaat wel vooruit maarkomt niet vooruit ja mss 30 metingen per minuut ofzo vind ik ook prima zolang het eind resultaat maar klopt .

dennissiemensma, 2019-10-08:
Je kunt de volgende code onderaan hetzelfde bestand zetten die je eerder heb aangepast: `dsmrreader/settings.py`

```
# Max 1 meting elke 5 seconden
DSMRREADER_DATALOGGER_SLEEP = 5
```

Daarnaast kun je weer een `#` zetten voor deze regel in datzelfde bestand:

```
LOGGING['loggers']['commands']['level'] = 'DEBUG'
```

Zoiets:
[image]

Daarna kun je weer alle processen herstarten. Als het goed is staat dan debugging uit en krijg je minder telegrammen binnen en haalt die uiteindelijk de achterstand in.

style2k, 2019-10-08:
ff afwachten dus of het bijgewerkt word de onverwachte telegrammen lijken af te nemen . maar wat kan nu de oorzaak zijn dan want vind het erg vreemd dat zich dat ineens voordoet dat die ineens 10 uur achter loopt .

dennissiemensma, 2019-10-08:
Het is moeilijk te zeggen op afstand, dan moet je echt gaan profilen en kijken welke deel van de applicatie traag is op je Pi.

Als het voor nu opgelost lijkt, doordat er geen nieuwe achterstand bij komt, zou ik het laten voor wat het is en kijken hoe het gaat.

style2k, 2019-10-08:
ja ik kijk het idd ff aan . maar die minder metingen hebben geen invloed op het resultaat toch?

dennissiemensma, 2019-10-09:
Correct.

dennissiemensma, 2019-10-09:
Is de achterstand inmiddels weggewerkt en loopt het weer door?

style2k, 2019-10-09:
yep het loopt in principe weer door al komen de onverwerkte telegrams nooit onder de 5 dus hij blijft groen en schommelt dan tussen 5 en 15 .
maar toch is er iets wat flink vertraagt toch zou alleen niet weten wat .

dennissiemensma, 2019-10-09:
Ik zou het dan laten voor wat het is als het verder goed werkt. Als het goed is zie je volledige gegevens in de dashboard en archief en mis je nu verder niets, ondanks dat er minder en vertraagd telegrammen binnenkomen.

closed 2019-10-09 by dennissiemensma