dsmrreader/dsmr-reader #732: DSMR Reader stop na x tijd met meten

eterpstra, 2019-10-08:
Hoi Dennis,

Ik heb sinds een paar dagen DSMR reader op een ODroid N2 boardje geinstalleerd.

Het probleem is dat DSMR reader soms geen verbinding meer heeft met de slimme meter. Ik zie dan zie ik in het dashboard ''laatste meting 1 uur gelden of ''laatste meting 22 min geleden''

Als ik vervolgens supervisor services herstart ( sudo supervisorctl restart all
)  dan werkt het weer.  Een bepaalde periode van tijden dat het niet werkt  is er niet echt.

Heb je enig idee wat dit kan zijn? EN waar zou ik eventueel de logs kunnen vinden?

Gr,
Elroy

dennissiemensma, 2019-10-08:
Je kunt sowieso deze logfiles bekijken: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Ik zou beginnen bij de datalogger.

dennissiemensma, 2019-10-10:
@eterpstra ben je nog wat wijzer geworden?

eterpstra, 2019-10-10:
Ik moet nog even afwachten of het is gelukt.. het script voor dataloggers stopte elke keer.. dus heb nu een aanpassing gedaan om te zien of dat werkt.

Ik laat het je einde week weten.

dennissiemensma, 2019-10-18:
@eterpstra werkt het nu wel?

dennissiemensma, 2019-10-24:
Mocht ik je alsnog verder kunnen helpen en hoor ik het graag!

closed 2019-10-24 by dennissiemensma