dsmrreader/dsmr-reader #735: Drop support for Python 3.5

dennissiemensma, 2019-10-09:
Django 3.x ondersteunt straks Python 3.5 niet meer. Dus daarom bij DSMR-reader 3.x ook meteen Python 3.5 niet meer ondersteunen

dajappie, 2019-11-15:
Hi Dennis, aangezien Raspberry Stretch geen Python 3.6 of hoger package heeft moet deze dan dus handmatig geinstalleerd worden. Misschien iets voor in de installguide?

jeroenpeters1986, 2019-11-15:
Ter referentie: Een aardig gedeelte hiervan was al gecovered in https://github.com/dennissiemensma/dsmr-reader/issues/624#issuecomment-485377652

dennissiemensma, 2019-11-15:
Dank voor de heads up. Ik heb mijn lessen geleerd van de vorige Python-upgrade van het project.

De vorige keer heb ik de `master` branch aangehouden bij DSMR-reader 2.0 (waarmee een nieuwe Python-versie nodig was). Daardoor kregen gebruikers zomaar de nieuwste versie die niet backwards-compatible was met oudere Python. Dat ga ik niet nodig een keer doen. 

Ik ga (en heb nu al) aparte branches gemaakt voor DSMR-reader v3.x en v2.x. Nadat 3.x uit is, update ik de `master` branch niet meer en moet men zelf wisselen van branch, samen met wat handmatige wijzigingen die nodig zijn. Daarvoor zal ik inderdaad een soort upgrade-guide voor maken.

Overigens wacht ik nog op [Django 3.0](https://docs.djangoproject.com/en/3.0/releases/3.0/) die medio December wordt verwacht. En daarna moet ik eerst zelf nog aan de slag voor alle backwards-incompatible changes, voordat het in DSMR-reader 3.x kan

dennissiemensma, 2019-12-18:
Alvast een eerste opzet van een upgrade guide gemaakt. Ik kon het niet op een Pi testen, maar wel op een VPS en dat ging goed.

https://dsmr-reader.readthedocs.io/en/v3/faq/v3_upgrade.html

closed 2020-01-04 by dennissiemensma