dsmrreader/dsmr-reader #737: When Upgrading Key (id)=(xx) already exists.

megapearl, 2019-10-10:
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.6`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Ubuntu Linux`
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v5`

### Overige
Hoi Dennis,

Er lijkt mij iets fout te staan in mijn database, bij sommige updates krijg ik duplicate key errors,
De laatste 2 updates gingen goed (van 2.4 -> 2.5 en van 2.5 -> 2.6)
Maar van 2.6 naar 2.7 gaat het mis.

Hoe kan ik de hele database checken/repareren?

(dsmrreader) dsmr@ubuntu-server:~/dsmr-reader$ ./deploy.sh

 --- You are currently running version:
2.6

 Please make sure you've read EACH release note BEFORE deploying, since some
 changes might not always be backwards compatible or depend on your database!

     >>>   https://github.com/dennissiemensma/dsmr-reader/releases   <<<

 ~~~ Still want to deploy? (type Y + ENTER to continue or press CTRL + C to abort)
y

 --- Checking Python version.
Running Python 3.6.8 (default, Apr  9 2019, 04:59:38)
[GCC 8.3.0]

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 142, done.
remote: Counting objects: 100% (142/142), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 142 (delta 91), reused 120 (delta 76), pack-reused 0
Receiving objects: 100% (142/142), 114.70 KiB | 1.27 MiB/s, done.
Resolving deltas: 100% (91/91), completed with 40 local objects.
From https://github.com/dennissiemensma/dsmr-reader
   1da2ccd3..fd686547  master      -> origin/master
   1da2ccd3..fd686547  development -> origin/development
   b1c999de..73aee228  v2          -> origin/v2
 * [new branch]        wip/2.8     -> origin/wip/2.8
 * [new tag]           v2.7.0      -> v2.7.0

 --- Merging/updating checkout.
Updating 1da2ccd3..fd686547
Fast-forward
 .circleci/config.yml                                                                                            |    3 +-
 docs/changelog.rst                                                                                              |   13 ++
 docs/faq.rst                                                                                                    |    8 +-
 docs/index.rst                                                                                                  |    2 +-
 docs/locale/nl/LC_MESSAGES/changelog.mo                                                                         |  Bin 91166 -> 91954 bytes
 docs/locale/nl/LC_MESSAGES/changelog.po                                                                         |  909 +++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------------------------
 docs/locale/nl/LC_MESSAGES/faq.mo                                                                               |  Bin 3805 -> 4398 bytes
 docs/locale/nl/LC_MESSAGES/faq.po                                                                               |  456 ++++++++++++++++++++-------------------------------
 docs/locale/nl/LC_MESSAGES/troubleshooting.mo                                                                   |  Bin 4974 -> 5398 bytes
 docs/locale/nl/LC_MESSAGES/troubleshooting.po                                                                   |  156 +++++++++++-------
 docs/troubleshooting.rst                                                                                        |   35 +++-
 dsmr_api/tests/v2/test_consumption.py                                                                           |    3 +-
 dsmr_backend/mixins.py                                                                                          |    8 +-
 dsmr_consumption/services.py                                                                                    |    2 +-
 dsmr_datalogger/models/reading.py                                                                               |    4 +-
 dsmr_dropbox/tests/test_services.py                                                                             |    6 +-
 dsmr_frontend/migrations/0022_v270_release.py                                                                   |   37 +++++
 dsmr_frontend/static/dsmr_frontend/bootstrap-datepicker-1.9.0-dist/css/bootstrap-datepicker.css                 |  477 ++++++++++++++++++++++++++++++++++++++++++++++++++++++
 dsmr_frontend/static/dsmr_frontend/bootstrap-datepicker-1.9.0-dist/css/bootstrap-datepicker.css.map             |    1 +
 dsmr_frontend/static/dsmr_frontend/bootstrap-datepicker-1.9.0-dist/css/bootstrap-datepicker.min.css             |    7 +
 dsmr_frontend/static/dsmr_frontend/bootstrap-datepicker-1.9.0-dist/css/bootstrap-datepicker.standalone.css      |  510 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 dsmr_frontend/static/dsmr_frontend/bootstrap-datepicker-1.9.0-dist/css/bootstrap-datepicker.standalone.css.map  |    1 +
 dsmr_frontend/static/dsmr_frontend/bootstrap-datepicker-1.9.0-dist/css/bootstrap-datepicker.standalone.min.css  |    7 +
[... 328 lines cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "django_content_type_pkey"
DETAIL:  Key (id)=(39) already exists.**

 --- Checking & synchronizing static file changes.

99 static files copied to '/var/www/dsmrreader/static', 438 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_mqtt                      [OK]

 --- Clearing cache...
[2019-10-10 21:40:47,921] INFO     Clearing cache: default
[2019-10-10 21:40:47,922] INFO     Clearing cache: mqtt
[2019-10-10 21:40:47,922] INFO     Done

 --- Deployed version:
2.7

 >>> Deployment complete. <<<
(dsmrreader) dsmr@ubuntu-server:~/dsmr-reader$

dennissiemensma, 2019-10-10:
Welke output krijg je hiermee:

```
sudo su - dsmr
./manage.py showmigrations | grep -v '[X]'
```

dennissiemensma, 2019-10-10:
Je kunt sowieso dit nog proberen:
```
sudo su - postgres
psql dsmrreader

BEGIN;
SELECT setval(pg_get_serial_sequence('"django_content_type"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_content_type";
COMMIT;
```

En daarna:
```
sudo su - dsmr
./post-deploy.sh
```

Wellicht krijg je nu de melding niet meer.

megapearl, 2019-10-11:
> Welke output krijg je hiermee:
> 
> ```
> sudo su - dsmr
> ./manage.py showmigrations | grep -v '[X]'
> ```

```
(dsmrreader) dsmr@ubuntu-server:~/dsmr-reader$ ./manage.py showmigrations | grep -v '[X]'
admin
auth
contenttypes
dsmr_api
dsmr_backend
dsmr_backup
dsmr_consumption
dsmr_datalogger
dsmr_frontend
dsmr_mindergas
dsmr_mqtt
dsmr_notification
dsmr_pvoutput
dsmr_stats
dsmr_weather
sessions
(dsmrreader) dsmr@ubuntu-server:~/dsmr-reader$

```

megapearl, 2019-10-11:
> Je kunt sowieso dit nog proberen:
>
> ```
> sudo su - postgres
> psql dsmrreader
>
> BEGIN;
> SELECT setval(pg_get_serial_sequence('"django_content_type"','id'), coalesce(max("id"), 1), max("id") IS NOT null) [...]

Inderdaad geen error meer:
```
postgres@ubuntu-server:~$ psql dsmrreader
psql (10.9 (Ubuntu 10.9-0ubuntu0.18.10.1))
Type "help" for help.

dsmrreader=# BEGIN;
BEGIN
dsmrreader=# SELECT setval(pg_get_serial_sequence('"django_content_type"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_content_type";
 setval
--------
     42
(1 row)

dsmrreader=# COMMIT;
COMMIT
dsmrreader=# \q

```

deploy lijkt nu goed te gaan:
```
(dsmrreader) dsmr@ubuntu-server:~/dsmr-reader$ ./post-deploy.sh

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.6.8 (default, Apr  9 2019, 04:59:38)
[GCC 8.3.0]

 --- Checking & synchronizing base requirements for changes.
Requirement already satisfied: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.7)
Requirement already satisfied: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.3.3)
Requirement already satisfied: django==2.2.6 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (2.2.6)
Requirement already satisfied: django-colorfield==0.1.15 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.1.15)
Requirement already satisfied: django-debug-toolbar==2.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (2.0)
Requirement already satisfied: django-filter==2.2.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2.0)
Requirement already satisfied: djangorestframework==3.10.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (3.10.3)
Requirement already satisfied: django-solo==1.1.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (1.1.3)
Requirement already satisfied: dropbox==9.4.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (9.4.0)
Requirement already satisfied: gunicorn==19.9.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (19.9.0)
Requirement already satisfied: paho-mqtt==1.4.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (1.4.0)
Requirement already satisfied: pyserial==3.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (3.4)
Requirement already satisfied: python-dateutil==2.8.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (2.8.0)
Requirement already satisfied: pytz==2019.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (2019.3)
Requirement already satisfied: requests==2.22.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (2.22.0)
Requirement already satisfied: urllib3==1.25.6 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (1.25.6)
Requirement already satisfied: coreschema in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (0.0.4)
Requirement already satisfied: uritemplate in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (3.0.0)
Requirement already satisfied: itypes in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.1.0)
Requirement already satisfied: sqlparse in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from django==2.2.6->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (0.3.0)
Requirement already satisfied: six>=1.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from dropbox==9.4.0->-r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.12.0)
Requirement already satisfied: idna<2.9,>=2.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from requests==2.22.0->-r dsmrreader/provisioning/requirements/base.txt (line 15)) (2.8)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (from requests==2.22.0->-r dsmrreader/provisioning/requirements/base.txt (line 15)) (3.0.4)
[... cut ...]
0 static files copied to '/var/www/dsmrreader/static', 537 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_mqtt                      [OK]

 --- Clearing cache...
[2019-10-11 08:16:13,924] INFO     Clearing cache: default
[2019-10-11 08:16:13,925] INFO     Clearing cache: mqtt
[2019-10-11 08:16:13,925] INFO     Done
(dsmrreader) dsmr@ubuntu-server:~/dsmr-reader$

```

closed 2019-10-11 by dennissiemensma