dsmrreader/dsmr-reader #738: Feature Request: Temperatuur bij vergelijkingen

dataegg, 2019-10-11:
* v2.4
* ESX 6.5, debian 10
* DSMR-protocol slimme meter v5

Kan bij de vergelijking ook de gemiddelde temperatuur zichtbaar worden gemaakt per periode. Met deze informatie is het makkelijker om stook kosten (gas) te vergelijken.

dataegg, 2019-10-11:
Nog een aanvulling. Een totaal van Elektriciteit (dal + piek) zou ook handig zijn in het vergelijkingsoverzicht

dennissiemensma, 2019-10-11:
Bedankt voor je verzoek. Ik zal het overwegen.

Betreffende je aanvulling, je kunt er ook voor kiezen om de tarieven in de applicatie te laten samenvoegen. Dat is een optie in de admin bij de frontend/weergave instellingen.

dataegg, 2019-10-12:
Dank de reactie en tip.

dennissiemensma, 2020-09-28:
Ik heb het opnieuw overwogen, maar ik denk dat je in dat geval beter kunt kijken naar MinderGas.nl. 

Daar wordt ook gerekend met [graaddagen](https://mindergas.nl/degree_days_calculation/explanation) dus je kunt kijken of je daar de informatie heen kunt sturen (kan automatisch) of een paar meterstanden handmatig invullen. Vooral omdat temperatuur alleen niet voldoende zegt over het gasverbruik.

closed 2020-09-28 by dennissiemensma