dsmrreader/dsmr-reader #739: Feature: Voltages inzichtelijk in Archief

jorkzijlstra, 2019-10-15:
*Bedankt voor het aanmaken van een issue!*
### Wat gebruik je?
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.5`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Rock Pi 4B`
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v5`

### Overige
Met de huidige versie zijn de voltages inzichtelijk gemaakt op het dashboard. Mij lijkt het nuttig om deze ook op het Archief te tonen.

Dan kan je bijvoorbeeld inzien of de stroom is uitgevallen tijdens je vakantie. Dit scheelt dan weer het leggen van een stuiver bovenop een ijsblokje in de vriezer.

dennissiemensma, 2019-10-15:
Bedankt voor je suggestie. Ik denk dat het een goede aanvulling is voor #587. 

Het Archief is namelijk alleen bedoeld voor historische dagtotalen en voltages zijn niet op te tellen. In #587 is het de bedoeling om de data uit het dashboard te ontsluiten over een te selecteerbare periode of dag. Mits de data dan nog aanwezig is.

closed 2019-10-15 by dennissiemensma

jorkzijlstra, 2019-10-15:
Hoi Dennis,

#587 Lijkt idd de juiste oplossing hiervoor te zijn. Alvast bedankt.