dsmrreader/dsmr-reader #740: Update issue (1.28)

djvanas, 2019-10-16:
*Bedankt voor het aanmaken van een issue!*
### Wat gebruik je?
* DSMR v1.28
* Pi3
* DSMR-protocol slimme meter  v4

Na sudo su - dsmr, ./deploy.sh en bevstigen van update gebeurt er van alles maar uiteindelijk is versie 1.28 were deployed. Zie screenshot

[image]

djvanas, 2019-10-16:
> Wat krijg je als je dit doet:
> 
> ```
> sudo su - dsmr
> git branch
> ```

[image]

dennissiemensma, 2019-10-16:
Je zou dan dit kunnen proberen:
```
sudo su - dsmr
git fetch
git pull
./post-deploy.sh
```

djvanas, 2019-10-17:
Dan krijg ik dit met resultaat dat 1.28 gedeployed blijft;

[image]

dennissiemensma, 2019-10-17:
Ik denk dat het belangrijk is dat je eventuele fouten ook doorgeeft. Op de rand van je screenshot zie ik nog net bovenin een foutmelding staan. Er lijken lokaal wijzigingen gedaan in de code die onder versiebeheer staat. 

Ik weet niet wat je hebt aangepast en of dat bewust is geweest, maar je zult dat ongedaan moeten maken:

```
sudo su - dsmr
git reset --hard
./deploy.sh
```

dennissiemensma, 2019-10-18:
Mocht je nog meer hulp nodig hebben, dan hoor ik het graag.

closed 2019-10-18 by dennissiemensma

djvanas, 2019-10-21:
Tnx - fixed