dsmrreader/dsmr-reader #741: Vraag: data forwarding (Postgres 12 test)

jorkzijlstra, 2019-10-16:
### Wat gebruik je?
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.7`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Rock Pi 4b`
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v5`

Postgres 12 is sinds kort uit en deze zou beter performen. Ik wil dus kijken of dsmr hierop goed performed. Ik heb hiervoor 2 keuzes kwa setups.

Inmiddels heb ik 2 dsmr instanties draaien met beide op een eigen postgres database. De ene op Postgres 10.5 en de andere op Postgres 12.

Ik kan alleen de p1 poort niet laten uitlezen door beide applicatie en zal dus de data van de 10.5 naar de 12 moeten zien te krijgen. Ik heb de documentatie bekeken en ik vermoed dat het antwoord nee is maar wil het toch even vragen. 

Is het mogelijk om data te forwarden van de ene dsmrreader naar een slave dsmrreader?

Indien dit niet mogelijk is zal ik even moeten kijken hoe ik een master slave setup voor postgres kan inrichten al vermoed ik dat dat niet zo moeilijk zal zijn.

jeroenpeters1986, 2019-10-16:
Heb je ze dicht bij elkaar staan? Ik kwam zelf namelijk toevallig laatst het volgende tegen op Marktplaats: http://www.slimmermetenergie.com/product/p1-splitter-actief/
Althans, de splitter, niet exact dit product. Ik heb er geen ervaring mee, maar er worden er nog een aantal aangeboden. Misschien is dat wat voor je als je het softwarematig niet gemakkelijk voor elkaar krijgt.

jorkzijlstra, 2019-10-16:
Ik gebruik docker containers en heb dus 4 containers (2x postgres + 2x dsmr) op dezelfde host draaien. Een splitter is dus geen optie. Al eerder heb ik geprobeerd de `/dev/ttyUSB0` op 2 containers te mappen maar de laatste container die opkomt is degene die deze kan uit lezen. De andere container kan de `/dev/ttyUSB0` niet gebruiken.

Bedankt voor de suggestie

dennissiemensma, 2019-10-17:
Je kunt hier meerdere dingen doen:
* Als aanvulling op de suggestie van Jeroen kun je ook softwarematig splitten op de host zelf met `socat`. In o.a. #377 staat [wat meer informatie](https://github.com/dennissiemensma/dsmr-reader/issues/377#issuecomment-442237344) hier over. Je creeert op je host dan een tweede seriele poort.

* Master-slave is een optie die je zelf aangeeft. Hoewel dat makkelijk te realiseren zou moeten zijn, kan het wat bijeffecten hebben omdat je vermoedelijk niet op de slave kan schrijven. Inloggen op de webinterface zou mogelijk niet werken (sessie komt in database).

* De beste optie is denk ik om de oplossing in #557. In je tweede installatie zet je de API aan en in je eerste installatie (degene die de poort uitleest) voeg je de plugin toe. Deze stuurt dan telegrammen door via de API. Meer informatie: https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-3-forwarding-raw-telegram-data-to-another-instance-by-api

jorkzijlstra, 2019-10-17:
Hoi Dennis,

Die laatste is degene die ik zocht. Blijkbaar heb ik er toch overheen gelezen.
Het zal wel komend weekend worden voordat ik deze kan testen.

Laat wel weten wat de uitkomst is.

Grt, Jork

dennissiemensma, 2019-10-18:
Prima. Mocht je alsnog tegen issues aanlopen, comment gerust weer hier.

closed 2019-10-18 by dennissiemensma

jorkzijlstra, 2019-10-20:
De software zelf lijkt gewoon normaal te werken en kwa performance moet ik nog kijken of ik wat meer inzichtelijk kan krijgen.

Ik heb wel een issue aangemaakt voor het docker gedeelte omdat de backups niet meer lukken.
https://github.com/xirixiz/dsmr-reader-docker/issues/75

jorkzijlstra, 2019-11-02:
Ik heb de software nu bijna 2 weken naast elkaar gedraaid waarbij de 10.5 versie de data forwarden naare de 12 instantie via de api.

Postgres 10.5
[image]

Postgres 12
[image]

Conclusie:
Er zijn hele kleine verschillen:
- postgres 12 lijkt iets minder cpu te gebruiken
- postgres 12 lijkt iets minder last te hebben van database locks

Dit zijn alleen hele kleine verschillen, Het grootste issue is dat de postgres 10.5 client gebruikt wordt om met postgres 12 te communiceren. De postgres 12 client is namelijk nog niet uit. Vanwege het gebruik van de 10.5 client lijken de backups niet meer te werken al heb ik hier niet diep naar gekeken.

Python Postgres client kan hier gemonitord worden:
https://pkgs.alpinelinux.org/packages?name=postgresql-client&branch=v3.10&repo=main&arch=x86_64

Als met al draait de software dus goed en we zullen dus moeten wachten op de postgres 12 client release.

dennissiemensma, 2019-11-02:
Bedankt voor het delen van je bevindingen. Was het puur een experiment of zijn er plannen om Postgresl 12 binnenkort al in Raspbian te integreren? Ik zie dat ik zelf 11.x gebruik op een vrij recente installatie.

jorkzijlstra, 2019-11-02:
Was puur een test omdat Postgres 12 beter zou moeten performen met minder resources. En iets voor een pi dat minder resources zou gebruikt is altijd mooi. Ik heb begrepen dat 11 daarentegen wat minder performed, al weet ik niet meer op welk vlak precies.

Mijn host machine is een Rock Pi 4B (4Gb mem) met een Ubuntu stretch installatie. Hierdoor draai ik de dsmr software binnen Docker. Het is dus de vraag wanneer de 12 client gereleased wordt. Ik zag dat de r0 inmiddels is gereleased en verwacht dat het binnen een paar maanden wel zal gebeuren.

Ik ben niet op de hoogte van raspbian en hoe het updaten van de client daar in zou gaan. Ik heb zelf geen raspbian draaien en kan het dus ook niet nagaan om het installeren van een nieuwere client mogelijk is.

dennissiemensma, 2019-11-02:
Mocht je sowieso nog rare dingen zien qua performance dan hoor ik het graag. Ik heb al lange tijd niet meer gekeken naar bottlenecks qua queries of locks.