dsmrreader/dsmr-reader #742: DSMR reader werkt niet meer na vervanging slimme meter

cranberryNL, 2019-10-17:
Vanmorgen is mijn slimme meter vervangen omdat deze geen meterstanden meer verzond. De meter is identiek op de communicatiemodule na. De oude had een CDMA-module en de nieuwe een GPRS-module met SIM-kaart.

Na de vervangen van de meter werkt DSMR reader niet meer. Er lijken geen meterstanden ontvangen te worden. Via Mosquitto zie ik geen MQTT-messages meer binnenkomen.
Als de interface (/dev/ttyUSB0) in Domoticz koppel, zie ik dat er (correcte) data binnenkomt. De slimme meter en P1-kabel werken dus goed.
In de instellingen heb ik gegekeken wat er verder nog gedaan kan worden aan communicatie-instellingen, maar dat leverde niets op.

In dit forum ben ik alleen berichten tegengekomen van absurde waarden. Zover kom ik echter nog niet. Alle data die ik zie is van het moment dat mijn Linux-machine een shutdown kreeg omdat de meter moest worden vervangen en alles in huis dus spanningsloos zou worden. Zelfs het actuele verbruik rechtsbovenin blijft hangen op de oude waarde (215W)

Ik draai v2.7.

cranberryNL, 2019-10-17:
Update: na spitten in de diverse logs, stuitte ik op dit bericht in django.log:

[2019-10-17 10:50:45,875] ERROR    @ services | Discarded telegram with future timestamp: /ISK5\2M550T-1011

1-3:0.2.8(50)
0-0:1.0.0(191017105045S)
0-0:96.1.1(012345678912341234123534643554354542)  //meter id ff aangepast
1-0:1.8.1(000000.295*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.314*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.291*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(234.7*V)
1-0:52.7.0(234.8*V)
1-0:72.7.0(236.3*V)
1-0:31.7.0(000*A)
1-0:51.7.0(002*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.145*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.021*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.470*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0()
0-1:24.2.1(700101010000W)(00000000)
!D373

De timestamp (bovenin) klopt en ook op mijn Linux-machine zie ik geen afwijk groot tijdverschil (weet zo snel niet ik hoe moet controleren of de secondes ook gelijk lopen).

Het valt me sowieso ook op dat er geen gasmeterstand wordt meegestuurd. Daarvoor ga ik nog in conclaaf met de netbeheerder.

dennissiemensma, 2019-10-17:
Het tijdstip van de gasmeter lijkt niet te kloppen (jaar 2070) en daarom faalt die. Er zit overigens een aardige marge in wat toegestaan is hoor, dat komt niet aan op secondes, minuten of uren. Het systeem wijst het telegram af als er meer dan 24 uur verschil is, in de toekomst. De netbeheerder kan dit voor je oplossen, of je kunt een dagje wachten of de meter vanzelf goed gaat.

cranberryNL, 2019-10-17:
Dank voor je snelle reactie Dennis.

De netbeheerder geeft aan dat, het juist doorsturen van de gasmeterstand naar de elektriciteitsmeter (waarin de communicatiemodules zijn ingebouwd) afhankelijk is van het bij de netbeheerder volledig verwerkt zijn van de meterwissel. Dit kan 10 werkdagen duren. 
Het lijk mij wat onwaarschijnlijk, maar vooralsnog kan ik er niets mee. Ik wacht af of het één dezer dagen wel lukt.
Ik zal de voorgang hier melden, zodat in de toekomst anderen er wellicht iets aan hebben.

dennissiemensma, 2019-10-17:
Ik zal binnenkort eens kijken of ik in deze situaties de gasmetingen optioneel kan maken. Tot nu toe was het nooit echt een issue voor nieuwe gebruikers, maar voor bestaande gebruikers is het wat onnodig dat een wachtende gasmeter ook de elektra-metingen tegenhoudt.

cranberryNL, 2019-10-17:
Even een update: alles werk tnu goed. De gasmeterstand is zichtbaar op de meter en in de telegrammen is deze ook weer aanwezig, incl. juiste datum/tijd

Ik heb nog even gezocht naar een mogelijkheid om de (dag)statistieken aan te passen, maar dat lukt niet zo.
Hier staat het beschreven, maar ik kan geen mogelijkheid vinden om e.e.a. aan te passen.

dennissiemensma, 2019-10-18:
Als het goed is staat het bijna onderaan alle links in het admin-overzicht.

* `/admin/dsmr_stats/daystatistics/`
* `/admin/dsmr_stats/hourstatistics/`

dennissiemensma, 2019-10-24:
@cranberryNL heb je het kunnen vinden?

closed 2019-11-02 by dennissiemensma