dsmrreader/dsmr-reader #743: Kleine verbetering aan installatie - nginx

TweakM, 2019-10-19:
Wat een geweldige software! Eenvoudig en doeltreffend! @dennissiemensma: heel erg bedankt!

### Wat gebruik ik?
* DSMR-reader versie: `v2.7`
* Hardware: `RaspberryPi  4`

* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v5`

### Kleine verbetering
Bij de installatie wordt het config bestand vanuit de provisioning directory naar /etc/nginx/sites-enabled gekopieerd. Om de sites-enabled folder te gebruiken zoals het is bedoeld kan beter een kopie worden gemaakt van het config bestand naar de sites-available en daarna een symlink naar sites-enabled.
Hiervoor kan je deze commando's gebruiken:
```
sudo cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/nginx/dsmr-webinterface /etc/nginx/sites-available/
sudo ln -s /etc/nginx/sites-available/dsmr-webinterface /etc/nginx/sites-enabled/
```
Alternatief kan je ook een symlink in sites-available maken naar de provisioning folder en dan doorlinken naar sites-enabled. Voor dit laatste scenario gebruik je deze commando's:
```
sudo ln -s /home/dsmr/dsmr-reader/dsmrreader/provisioning/nginx/dsmr-webinterface /etc/nginx/sites-available/dsmr-webinterface
sudo ln -s /etc/nginx/sites-available/dsmr-webinterface /etc/nginx/sites-enabled/
```
Misschien is het een optie om de documentatie aan te passen naar één van beide opties?

dennissiemensma, 2019-10-19:
Bedankt voor je toevoegingen. Ik zal het in een van de toekomstige versies voor je doorvoeren.

TweakM, 2019-10-19:
Bedankt voor je snelle reactie!

dennissiemensma, 2019-10-24:
Gaat mee in docs build voor volgende release.

closed 2019-10-24 by dennissiemensma

dennissiemensma, 2019-10-25:
Uitgebracht in [v2.9.0](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.9.0).