dsmrreader/dsmr-reader #745: 502 Bad Gateway

Schaduwval, 2019-10-20:
*Bedankt voor het aanmaken van een issue!*
### Wat gebruik je?
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.7`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3.`
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v5`

### Bij fouten
* Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.
* Beschrijf de situatie waarin het mis gaat.

### Overige
*Ik heb sinds vandaag een probleem met het benaderen van de webinterface van DSMR-reader, dat lukt helaas niet meer* De onderstaande informatie krijg ik als ik DSMR-reader probeer te updaten. Enig idee waar het aan kan liggen?

`You are currently running version: 
2.7

 --- Checking Python version.
Running Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 8 (delta 0), reused 2 (delta 0), pack-reused 0
error: unable to create temporary file: Geen ruimte meer over op apparaat
fatal: failed to write object
fatal: unpack-objects failed

 --- Merging/updating checkout.
fatal: update_ref failed for ref 'ORIG_HEAD': could not write to '.git/ORIG_HEAD'

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]

 --- Checking & synchronizing base requirements for changes.
ERROR: Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 188, in main
    status = self.run(options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/pip/_internal/commands/install.py", line 295, in run
    wheel_cache = WheelCache(options.cache_dir, options.format_control)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/pip/_internal/cache.py", line 204, in __init__
    self._ephem_cache = EphemWheelCache(format_control)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/pip/_internal/cache.py", line 180, in __init__
    self._temp_dir.create()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/pip/_internal/utils/temp_dir.py", line 75, in create
    tempfile.mkdtemp(prefix="pip-{}-".format(self.kind))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/tempfile.py", line 497, in mkdtemp
    prefix, suffix, dir, output_type = _sanitize_params(prefix, suffix, dir)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/tempfile.py", line 269, in _sanitize_params
    dir = gettempdir()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/tempfile.py", line 435, in gettempdir
    tempdir = _get_default_tempdir()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/tempfile.py", line 370, in _get_default_tempdir
    dirlist)
FileNotFoundError: [Errno 2] No usable temporary directory found in ['/tmp', '/var/tmp', '/usr/tmp', '/home/dsmr/dsmr-reader']
WARNING: You are using pip version 19.2.3, however version 19.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

 --- Checking & synchronizing database changes/migrations.
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 195, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 178, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 364, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 83, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/commands/migrate.py", line 87, in handle
    executor = MigrationExecutor(connection, self.migration_progress_callback)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/executor.py", line 18, in __init__
    self.loader = MigrationLoader(self.connection)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/loader.py", line 49, in __init__
    self.build_graph()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/loader.py", line 212, in build_graph
    self.applied_migrations = recorder.applied_migrations()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/recorder.py", line 73, in applied_migrations
    if self.has_table():
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/migrations/recorder.py", line 56, in has_table
    return self.Migration._meta.db_table in self.connection.introspection.table_names(self.connection.cursor())
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 256, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 233, in _cursor
    self.ensure_connection()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 195, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 178, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 537 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)kill: (605): Proces bestaat niet
   [OK]
 * Reloading process: dsmr_backend                   [??] PID file does not exist
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

 --- Clearing cache...

 --- Deployed version: 
2.7`

dennissiemensma, 2019-10-20:
Een van de foutmeldingen geeft aan dat je sd-kaart geen vrije ruimte meer heeft. Je kunt dat zien met "df -h". Ben je in de gelegenheid om iets op te schonen of draait er niets anders op de Pi?

Schaduwval, 2019-10-20:
Er draait alleen DSMR-reader op de Pi.

jeroenpeters1986, 2019-10-20:
Wat ook nog kan is dat je de SD card niet hebt uitgebreid na installatie
van het Operating System op de SDcard.

Als je Raspbian draait kun je inloggen op de Pi met SSH en dan ‘sudo
raspi-config’ intikken. Dan is er de optie Expand Filesystem, die neemt dan
je volledige SD card in gebruik, waar dat niet altijd het geval is na
installatie. Ik hoop dat dat het is, want dan kun je even weer vooruit!

dennissiemensma, 2019-10-20:
Je kunt daarnaast ook nog even kijken hoeveel ruimte DSMR-reader gebruikt. De grootte van de database (in MB):
```
sudo du -ms /var/lib/postgresql/
```

Afhankelijk van hoe groot je SD-kaart is, en de resultaten van `df -h`, kun je de suggestie van Jeroen proberen.

Wanneer je al aan de max van je SD-kaart zit, kun je retentie aanzetten en/of het aantal telegrammen per minuut beperken. Mocht dat het geval zijn, laat het even weten. Ik wacht eerst je eigen bevindingen af.

Schaduwval, 2019-10-20:
> Wat ook nog kan is dat je de SD card niet hebt uitgebreid na installatie van het Operating System op de SDcard. Als je Raspbian draait kun je inloggen op de Pi met SSH en dan ‘sudo raspi-config’ [...]

Bedankt voor je suggestie! ik heb het bovenstaande gedaan en de Pi opnieuw opgestart maar helaas nog steeds het zelfde.

Schaduwval, 2019-10-20:
> Je kunt daarnaast ook nog even kijken hoeveel ruimte DSMR-reader gebruikt. De grootte van de database (in MB):
>
> ```
> sudo du -ms /var/lib/postgresql/
> ```
>
> Afhankelijk van hoe groot je SD-kaart is, en [...]

De grote van de database is: `3992	/var/lib/postgresql/` 

Er zit een SanDisk Ultra 16 GB kaartje in.

Het resultaat van `df -h`= 

/dev/root           15G      14G     0 100% /
devtmpfs           460M        0  460M   0% /dev
tmpfs              464M        0  464M   0% /dev/shm
tmpfs              464M      12M  452M   3% /run
tmpfs              5,0M     4,0K  5,0M   1% /run/lock
tmpfs              464M        0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1      44M      22M   22M  51% /boot
tmpfs               93M        0   93M   0% /run/user/1000

dennissiemensma, 2019-10-20:
Zou je eens willen kijken wat de grootte hiervan is:
```
sudo du -ms /home/dsmr/dsmr-reader/backups/
```

Er wordt elke dag een backup gemaakt en na een week geroteerd. Het kan goed zijn dat deze backup's erg groot zijn inmiddels.

Je kunt dan een paar van de oudste verwijderen.

Schaduwval, 2019-10-20:
> Zou je eens willen kijken wat de grootte hiervan is:
> 
> ```
> sudo du -ms /home/dsmr/dsmr-reader/backups/
> ```
> 
> Er wordt elke dag een backup gemaakt en na een week geroteerd. Het kan goed zijn dat deze backup's erg groot zijn inmiddels.
> 
> Je kunt dan een paar van de oudste verwijderen.

De grote van de backups: 

`3618	/home/dsmr/dsmr-reader/backups/`

dennissiemensma, 2019-10-20:
Hmm dat is niet zoveel als dat ik had gedacht. Bij elkaar kom je dan op zo'n 8 GB, terwijl er 14 GB in gebruik is.

Het is de vraag waar nog meer gebruik zit. Wat krijg je hiermee te zien:

```
sudo du -msc /* | sort -n
```

dennissiemensma, 2019-10-20:
Als aanvulling. Je zult denk ik wat keuzes moeten maken qua dataopslag. DSMR-v5 meters zorgen voor veel data.

----

Ik gaf eerder aan dat je retentie kan aanzetten, zodat metingen na ene X periode worden verwijderd. Je kunt hier meer informatie vinden over hoe dat werkt: https://dsmr-reader.readthedocs.io/nl/latest/admin/datalogger.html

In de Statistieken-pagina van de applicatie staat ergens een knop waarje op kunt klikken om het aantal metingen te zien. Als je retentie aan hebt gezet zou het aantal metingen na een nacht verminderd moeten zijn.

----

Daarnaast kan ik je aanraden om het aantal metingen per seconden te beperken. Gezien je een v5-meter hebt, krijg je die nu elke seconde en dat zorgt voor heel veel data. Als je puur je verbruik/meterstanden wilt bijhouden, kun je ervoor kiezen om bijvoorbeeld elke 5 seconde een meting te laten binnenkomen. Hiervoor moet je een instelling aanpassen in de config.

```
sudo su - dsmr

# Open dsmrreader/settings.py, bijvoorbeeld met VI or nano
vi dsmrreader/settings.py

# Voeg deze regel toe aan het einde van het bestand en sla het bestand op.
DSMRREADER_DATALOGGER_SLEEP = 5

# Herlaadt de applicatie.
./post-deploy.sh
```

Schaduwval, 2019-10-20:
> Hmm dat is niet zoveel als dat ik had gedacht. Bij elkaar kom je dan op zo'n 8 GB, terwijl er 14 GB in gebruik is.
> 
> Het is de vraag waar nog meer gebruik zit. Wat krijg je hiermee te zien:
> 
> ```
> sudo du -msc /* | sort -n
> 

> Hmm dat is niet zoveel als dat ik had gedacht. Bij elkaar kom je dan op zo'n 8 GB, terwijl er 14 GB

dennissiemensma, 2019-10-20:
Je hebt het issue nu gesloten. Ben je er uit gekomen?

Schaduwval, 2019-10-20:
> Je hebt het issue nu gesloten. Ben je er uit gekomen?

Dat deed ik per ongeluk! Kan het ook niet zijn dat het kaartje wat er zit niet meer 100 procent meer is?

Dit is het resultaat van `sudo du -msc /* | sort -n`

pi@raspberrypi:~ $ sudo du -msc /* | sort -n
du: kan geen toegang krijgen tot '/proc/3698/task/3698/fd/3': Bestand of map bestaat niet
du: kan geen toegang krijgen tot '/proc/3698/task/3698/fdinfo/3': Bestand of map bestaat niet
du: kan geen toegang krijgen tot '/proc/3698/fd/3': Bestand of map bestaat niet
du: kan geen toegang krijgen tot '/proc/3698/fdinfo/3': Bestand of map bestaat niet
0	/dev
0	/proc
0	/sys
1	/lost+found
1	/media
1	/mnt
1	/srv
1	/tmp
7	/bin
7	/etc
8	/sbin
12	/run
22	/boot
42	/root
186	/lib
1320	/opt
3205	/usr
3907	/home
5535	/var
14247	totaal
pi@raspberrypi:~ $

dennissiemensma, 2019-10-20:
Die meldingen hoef je je niet druk om te maken. Het is wel handig als je af en toe een backup veiligstelt voor wanneer je kaartje wel kapot gaat (ze falen uiteindelijk allemaal). 

Ik denk dat je het beste de oudste backup kunt verwijderen en daarna mijn stappen uit https://github.com/dennissiemensma/dsmr-reader/issues/745#issuecomment-544282652 doet.

Oudste backup verwijderen kan met:
```
rm /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Monday.sql.gz
```

Schaduwval, 2019-10-21:
Allereerst bedankt voor de hulp! Ik heb gister de oudste backup verwijderd, toen deed alles het weer. Echter vandaag kreeg ik de onderstaande melding: 

Server Error

Sorry, something unexpected happened.
Exception:

OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 28, in get_context_data
    context_data['capabilities'] = dsmr_backend.services.backend.get_capabilities()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 33, in get_capabilities
    'electricity': ElectricityConsumption.objects.exists(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 766, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/query.py", line 522, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1070, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1098, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 256, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 233, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 195, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 178, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

dennissiemensma, 2019-10-21:
Heb je ook de andere stappen uitgevoerd? Je kunt even kijken of het weer de schijf is met `df -h`.

Schaduwval, 2019-10-22:
Ja dat heb ik gedaan. en het was weer de schijf die vol was. Ik heb nu de backup weer verwijderd waardoor het werkt. Maar ik merk dat DSMR-reader erg langzaam werkt. En nu ik zie geen grafieken meer bij het dashboard, dit zijn 3 witte vlakken.  

Is het ene probleem opgelost, komt er een andere om de hoek kijken..

Schaduwval, 2019-10-24:
Ja dat kan ik mij voorstellen. 

Ik heb de retentie aangezet. En andere stappen uitgevoerd. Helaas blijkt het nu zo niet goed te werken zoals het voor deed. Er worden geen metingen geregistreerd. Is het een optie om de raspberry pi opnieuw installeren en dsmr reader opnieuw te installeren? Dat icm met een nieuw kaartje eraan in En dan de backups op te slaan op een externe HD? (Als dat mogelijk is) 

Vind het namelijk DSMR Reader wel een fijn programma!   

Ik hoor graag je advies.

dennissiemensma, 2019-10-24:
Dat is ook een optie. Je kunt in de webinterface onder "Export" (of wat de NL-vertaling ook is) een export maken van alle dagstatistieken in CSV.

Wil je kijken of dat lukt? En de CSV ook inhoudelijk bekijken of daar nog je gegevens in staan? 

Ik kan je dan later wel helpen met terugzetten van die gegevens en dan kun je in je nieuwe installatie gewoon met een schoon systeem beginnen.

dennissiemensma, 2019-10-24:
Voor de duidelijkheid. Alle metingen en andere detail-data is 99% van de schijfruimte. De dagstatistieken is de belangrijkste 1% dus als je die kan veiligstellen ben je niet alle historie kwijt (van wat je had).

closed 2019-11-08 by dennissiemensma