dsmrreader/dsmr-reader #747: Geen totalen gegenereerd als er een periode van telegrams mist - hoe op te lossen?

Jurrie, 2019-10-21:
* DSMR-reader versie: `v2.6`
* Hardware: x86 host met Docker image https://github.com/xirixiz/dsmr-reader-docker
* DSMR-protocol slimme meter: `v4/v5`

Ik heb een DSMR al een tijd geleden geïnstalleerd, maar het heeft een tijdje uitgestaan. Inmiddels heb ik het weer aangezet. Er mist dus een hele periode aan telegrams. Ik zie wel actuele metingen, maar geen nieuwe dag- / maand- / jaartotalen op de "archive" pagina. De laatste is van 4 september; daarna is DSMR een tijd uitgezet.

De status pagina geeft bij "Day & hour statistics" de melding "Data processing is lagging behind: Sept. 4, 2019". Als ik debug logging aanzet, zie ik in de stderr log van dsmr_backend herhaaldelijk:
```
[2019-10-21 10:52:01,861] DEBUG    Stats: No consumption found for 2019-09-05
[2019-10-21 10:52:01,870] DEBUG    Sleeping 1 sec(s)
[2019-10-21 10:52:02,900] DEBUG    Calling 0 backend service(s)
```
De melding klopt - er zijn geen telegrams voor die datum.

Hoe krijg ik DSMR zo ver dat hij de periode zonder ontvangen telegrams skipt?

dennissiemensma, 2019-10-21:
Bedankt voor je melding. Dit is een bug en ik zal kijken of ik er makkelijk een fix voor kan maken.

jeroenpeters1986, 2019-10-21:
Ah top dat dit al gemeld is, ik zag het ook toen ik gisteren een Pi voor een vriend van me had geïnstalleerd en ik vandaag (na mijn eigen dsmr weer aangesloten te hebben) zag dat er neits meer gedaan was qua verwerking.

dennissiemensma, 2019-10-22:
Ik heb hiervoor een fix gemaakt en die gaat zo mee met de volgende release.

closed 2019-10-22 by dennissiemensma

jeroenpeters1986, 2019-10-22:
Ik had hem al even uitgecheckt van develop gisteren en dat leek te werken!
👍

notifications@github.com>

dennissiemensma, 2019-10-22:
Uitgebracht in [v2.8](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.8.0)