dsmrreader/dsmr-reader #748: Live verbruik doorsturen (niet via MQTT)

BramDeRuyck, 2019-10-21:
Erg mooie software en uitstekend gedocumenteerd.

Ik zou graag in real-time de _verwerkte_ telegrammen doorsturen naar de cloud (via events).
Zoals ik het begrijp zijn er 3 mogelijkheden:

1. De API gebruiken om elke seconde te laatste reading op te halen https://dsmr-reader.readthedocs.io/nl/latest/api.html#get-datalogger-dsmrreading
Output is exact wat nodig is, maar je zit pullt ipv pusht
2. Data logger installeren en aanpassen
https://dsmr-reader.readthedocs.io/nl/latest/installation/datalogger.html
Hier pas ik de code aan zodat het POST request naar de cloud gaat ipv naar dsmr-reader.
Nadelen: niet verwerkte telegrammen, wat bij fouten, downtime etc?
3. MQTT bridging
Gebruik maken van de bestaande MQTT functionaliteit naar een broker.
Nadelen: mijn cloud stack heeft geen ondersteuning voor MQTT 

Gewenste oplossing: net zoals bij MQTT een remote end point kunnen instellen, maar dan via HTTP(s) ipv MQTT als protocol.

dennissiemensma, 2019-10-21:
Bedankt voor jevraag. Ik denk dat je twee opties hebt:

*DSMR_parser*
Je kunt [deze library](https://github.com/ndokter/dsmr_parser) gebruiken en dan zelf een script in elkaar draaien die vervolgens de request naar je backend stuurt. De library valideert volgens mij ook de CRC-fouten.

*Plugin*
Het is mogelijk om een 'plugin' [te maken(https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-3-forwarding-raw-telegram-data-to-another-instance-by-api) dat triggert op basis van een event binnen DSMR-reader. Zodra er een telegram verwerkt is, kan die doorgestuurd worden via HTTP naar je backend. Mocht je deze oplossing willen, laat maar even weten, dan kijk ik of ik snel iets in elkaar kan knutselen voor je. Nadeel is wel dat je heel DSMR-reader hiervoor moet draaien, terwijl je dat misschien niet gebruikt.

MarbBram, 2019-10-22:
Dag Dennis,
bedankt voor de snelle reactie.
Optie 1 staat volledig los van dsmr-reader? Ik wil nog altijd graag gebruik maken van jouw mooie tool voor visualisatie, dus ik denk dat ik voor optie 2 via een plug-in ga zodat de data ook elders beschikbaar kan gemaakt worden.

dennissiemensma, 2019-10-22:
Het is [een leesbaar formaat zoals die in de database staat](https://github.com/dennissiemensma/dsmr-reader/blob/v2.7.0/dsmr_datalogger/models/reading.py#L14). Daar is JSON, XML of wat je dan ook wilt van te maken. Het telegram is op dat moment dus net aangemaakt en dus gevalideerd.

dennissiemensma, 2019-10-22:
Ik heb hier een voorbeeld voor je neergezet bij # 4: https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-4-forwarding-dsmr-readings-in-json-format-to-some-api

Je zult zelf denk ik wat moeten proberen om het helemaal goed te krijgen qua URL naar je API, evt authenticatie en als welk onderdeel van de request je het JSON-bericht wilt sturen.

Het moet redelijk voor zich spreken, maar als je er niet uit komt hoor ik het graag. Let op dat je het `dsmr_backend` proces telkens herstart na elke wijzging. En ook checkt of die niet crasht door een fout in de code.

dennissiemensma, 2019-10-22:
Mocht je trouwens niet `dsmr_datalogger` gebruiken, maar een remote datalogger die P1-telegrammen naar de API van DSMR-reader stuurt, dan zul je `dsmr_webinterface` telkens moeten herstarten om te testen of het werkt.

MarbBram, 2019-10-30:
Plug-in 4 werkt vlekkeloos!
Plug-in 3 werkt niet: https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-3-forwarding-raw-telegram-data-to-another-instance-by-api

Ik gebruik inderdaad een remote datalogger die via API naar DSMR-reader stuurt
Via ./reload.sh herstart ik telkens de hele boel (dsmr_webinterface, dsmr_datalogger en dsmr_backed)

Ik zie dat forward_telegram_to_api exact elke 5 minuten wordt ingeladen (via een simple GET request die ik had toegevoegd om te debuggen)
De functie _handle_forward_telegram_to_api_ wordt echter nooit uitgevoerd..

Ik zie als verschil tussen plug-in 3 en 4 dit:

- raw_telegram.connect(receiver=handle_forward_telegram_to_api)
- @receiver(django.db.models.signals.post_save)

Enige reden waarom je de receiver verschillend aanpakt?

dennissiemensma, 2019-10-31:
Bedankt voor je input. Het verschil is puur een andere/nieuwe manier van signalen koppelen aan code. In houdelijk zou het geen verschil moeten maken.

Ik kan me herinneren dat ik het goed getest heb, maar ik zal er op een later moment nog een keer extra checken. En eventueel de andere drie varianten gelijktrekken.

MarbBram, 2019-10-31:
[Plug-in 4 ](https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-4-forwarding-dsmr-readings-in-json-format-to-some-api)heeft ook nog wat fouthandling nodig
Anders heb je een fout elke keer je in de back-end een instelling wijzigt (hij reageert ook op dat signaal, maar die bevat geen timestamp, en is niet de bedoeling ook.
Of er zou een check moeten zijn in de plug-in dat het om een dsmrreading gaat.

    try:
	    instance.timestamp = timezone.localtime(instance.timestamp)
    except AttributeError as ae:
	    print('forward_json_dsmrreading_to_api:', ae)
	    return

MarbBram, 2019-10-31:
Plug-in 3 werkt goed, het lag aan mijn set-up.

dennissiemensma, 2019-10-31:
Bedankt voor je teurgkoppeling, er mist inderdaad wat code qua checks. Normaal schrijft een receiver zich ook in op een bepaald type opject, zoals een dsmrreading. Dat mist inderdaad in # 4

dennissiemensma, 2019-11-02:
Voor plugin 4 heb je deze wijziging nodig:

```
    import django.db.models.signals

    @receiver(django.db.models.signals.post_save)
```

Wordt
```
    import django.db.models.signals

    from dsmr_datalogger.models.reading import DsmrReading

    @receiver(django.db.models.signals.post_save, sender=DsmrReading)
```

Ik zal het bijwerken in de docs.

closed 2019-11-02 by dennissiemensma

dennissiemensma, 2019-11-02:
De andere plugins heb ik ook bijgewerkt in de docs voor de volgende release.