dsmrreader/dsmr-reader #750: Piek- en dalmetingen omgedraaid

braham2019, 2019-10-22:
Hallo, ik gebruik sinds vanochtend v2.7 via een Raspberry Pi 3B+. De applicatie werkt uitstekend. De RPI is gekoppeld aan een Belgische digitale meter van Fluvius.

Op mijn dashboard zie ik echter dat momenteel (overdag om 15u) de metingen worden weergegeven van het daltarief, ipv het piektarief.

In België zijn OBIS codes
1.8.1 Totale afname van energie in kWh dagtarief
1.8.2 Totale afname van energie in kWh nachttarief 
2.8.1 Totale injectie van energie in kWh dagtarief
2.8.2 Totale injectie van energie in kWh nachttarief

Wij noemen dat trouwens dag- en nachttarief ipv piek- en daltarief, maar dat terzijde. Zie ook: https://www.fluvius.be/sites/fluvius/files/2019-04/Technische-info-Displays-digitale-elektriciteitsmeter.pdf 

Zijn deze OBIS codes anders in België dan in Nederland ?

kudos trouwens voor een mooie applicatie :-)

dennissiemensma, 2019-10-22:
Bedankt voor je vraag. Als je binnen de applicatie deze URL opent: `/admin/dsmr_datalogger/meterstatistics/1/change/`, dan zie je onderaan de pagina als het goed is het telegram. Je zou even een telegram kunnen bekijken van voor 15:00 en eentje van na 15:00 om te zien wat het verschil is.

Je ziet dan waarschijnlijk een verschil in het cijfer van vlaggetje `96.14.0`, wat het huidige tarief aangeeft.

Je kunt hier de DSMR (NL) specs zien op pagina 20 en 21 van: 
* https://github.com/dennissiemensma/dsmr-reader/blob/master/docs/_static/downloads/netbeheer-nederland-dsmr-specs-v5.0.2.pdf

De huidige mapping binnen de applicatie staat hier:
* https://github.com/dennissiemensma/dsmr-reader/blob/v2.8.0/dsmr_datalogger/dsmr.py

Het lijkt overeen te komen, afgezien van de naamgeving.

dennissiemensma, 2019-10-22:
Het lijkt er inderdaad op dat het omgedraaid is. Hier is hoogtarief `2`.

braham2019, 2019-10-22:
Bedankt voor je antwoord. Enige optie om in een toekomstige versie rekening te houden met deze speciale Belgische situatie ?

dennissiemensma, 2019-10-23:
In de applicatie zelf kan ik niets voor je betekenen. 

Maar ik denk dat ik wel een instelling kan maken die de mapping omdraait bij het inlezen van de gegevens. Dus dat het systeem van dagtarief hoogtarief maakt en vice versa voor nachttarief.

braham2019, 2019-10-24:
Als je het ergens kan meenemen in de roadmap, zou dat perfect zijn voor de Belgische Fluvius klanten. Ik weet niet hoe het zit met andere energietransporteurs (intercommunales) in België.

dennissiemensma, 2019-10-24:
Ik zal kijken hoe ik dit kan realiseren op termijn.

MarbBram, 2019-10-31:
Dit is de situatie in NL
[img]https://i.imgur.com/4lDYlO1.png[/img]

dennissiemensma, 2019-11-02:
@floyson-reference hoe zit het bij jou qua tarieven? 

Ik besef me net dat ik in je ticket #697 alleen heb gehoord over gas dat niet verwerkt werd, maar niet over de naamgeving van de tarieven.

@MarbBram als ik de tarieven omdraai, dan heeft dat ook gevolgen voor iedereen en bestaande data. Dus wellicht is het achteraf niet zo handig om te doen.

floyson-reference, 2019-11-02:
Had idd ook al opgemerkt dat niet klopt (in België toch)

Mvg
Frederic Loyson

dennissiemensma, 2019-11-02:
Bedankt voor de input. Ik denk dat ik de tarieven dan niet ga omdraaien. Ze zijn namelijk qua naamgeving nog steeds incorrect en het zijn technisch gezien tarief 1 en 2. Alleen heten ze anders in Nederland.

Ik zal in ieder geval de wijzigingen voor gasmeters doorpushen. Mochten er later meer verschillen zijn met Belgische meters, kan ik dat vrij eenvoudig rechttrekken.

closed 2019-11-02 by dennissiemensma

braham2019, 2019-11-03:
Ik heb in DSMR.py deze omgedraaid. Dat werkt toch wel ? Wat bedoel je met dat je de tarieven niet gaat omdraaien ?

    '1-0:1.8.1': 'electricity_delivered_2',
    '1-0:2.8.1': 'electricity_returned_2',
    '1-0:1.8.2': 'electricity_delivered_1',
    '1-0:2.8.2': 'electricity_returned_1',

dennissiemensma, 2019-11-03:
In de eerste instantie dacht ik dat de tarieven omgedraaid waren, maar ze kloppen wel. Het is puur de naamgeving die afwijkt in België. En in bijvoorbeeld Luxemburg is het wederom anders.

Er is namelijk nog een vijfde veld die het huidige tarief aangeeft en het is onlogisch als de applicatie de meterstanden wegschrijft op tarief 2, terwijl het eigenlijk tarief 1 is. Bijkomend nadeel is dat wanneer ik de tarieven nu omdraai, het bij alle andere gebruikers ook omgedraaid wordt. 

Het kan zijn dat er ooit een instelling komt voor de naamgeving van de tarieven, maar dat is aardig wat werk om te realiseren.

dennissiemensma, 2019-11-05:
Uitgebracht in [v2.10](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.10.0)

braham2019, 2019-11-06:
Ik heb gisteren geüpgraded naar deze versie. Ik merk nu echter twee problemen die de statuspagina aangeeft: gasmeting en statistieken
- Recorded gas consumption is lagging behind: 10 hours ago
- Data processing is lagging behind: Nov. 4, 2019

Enig idee ?
Deze processen liepen daarvoor wel goed.

dennissiemensma, 2019-11-06:
Je kunt debug logging tijdelijk aanzetten en kijken in de backend log wat de reden is. https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

braham2019, 2019-11-06:
Ik had niet gezien dat je in de configuratie Belgie-Fluvius moest selecteren; dus dat heb ik nu gedaan.
Data van gisteren is ondertussen wel geprocessed. 

De gasmetingen echter komen nog steeds niet door nu.

MarbBram, 2019-11-06:
Wat is er nu eigenlijk precies geïmplementeerd in v2.10?

dennissiemensma, 2019-11-06:
@braham2019 excuus, dat is wat slordig van mij. Ik had in mijn hoofd er (automatisch)melding van te maken maar ik heb dat blijkbaar niet gedaan.

dennissiemensma, 2019-11-06:
@MarbBram er is ondersteuning voor Belgische meter en Luxemburgse meter. Echter, de tarieven zijn niet omgedraaid zoals eerder gemeld was. De tarieven hebben dezelfde tariefnummers als uit de slimme meter. De naamgeving van een tarief is namelijk het grootste verschil tussen NL en BE.

dennissiemensma, 2020-02-18:
@braham2019 @MarbBram @floyson-reference 

Ter info, in release v3.4 komt ondersteuning om zelf de namen in te stellen voor de tarieven (#858)

skank01, 2021-02-04:
Hoi,

Vraagje.
Bij mij (belgie) zijn ze ook omgedraaid.
In de instelling heb ik belgie-fluvies (gas correctie) ingesteld.
Maar de meter is idd omgedraaid.

Als ik het zo hier lees, moet ik gewoon die naamgeving veranderen?
En wat dan met de MQTT berichten, moet ik ze na de naamgeving ook daar wijzigen of gaat dat mee?

dennissiemensma, 2021-02-04:
@skank01 onderwater gaat het over "tarief 1" en "tarief 2", ook in MQTT. 

De betekenis van deze tarieven verschilt tussen NL en BE, dus je zult daar inderdaad zelf de juiste naam aan moeten geven als het niet klopt met je eigen situatie. En dus ook in MQTT een andere betekenis moeten hangen aan de ontvangende zijde van de MQTT-berichten.

skank01, 2021-02-04:
ok top
dus in het dashboard van dsmr reader aanpassen , en dan in home assistant aanpassen?
in de mqtt settings zelf niet he?

Ff offtopic, daarna zwijg ik
De berekening van kosten, best in dsmrreader? of in home assistant?
Top applicatie trouwens !! Bedankt daarvoor

dennissiemensma, 2021-02-04:
Ik weet niet hoe het in HA werkt, maar ik kan me voorstellen dat een standaard DSMR MQTT ontvanger/dashboard in HA ook een tariefnaam toont ipv "tarief 1" of "tarief 2". Als dat het geval is, zul je ze daar moeten omdraaien. Als je in HA alleen tarief 1/2 ziet, zul je zelf moeten onthouden welke tariefnummer wat is.
Als alternatief kun je in DSMR-reader ook nog de MQTT-mapping aanpassen en tarief 1 laten uitzenden op tarief 2 topic en andersom. Net wat je voorkeur heeft.

De "kosten" zoals in DSMR-reader zijn puur bedoeld voor een benadering/gevoel van wat een dag kost. Als dat afdoende is kun je het via DSMR-reader doen. Als je exact je leverancier wilt narekenen dan kun je dat het beste buiten DSMR-reader doen.