dsmrreader/dsmr-reader #754: Gas/electra verbruik nr PVoutput

ph0real, 2019-10-23:
### Wat gebruik je?
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.7`
* Hardware RaspberryPi 3b
* DSMR-protocol slimme meter *(v2, v4 of v5)*: `v...`

### Overige
Een vraag over hoe de verbruiksdata wordt geupload naar PVoutput. Ik krijg de indruk dat gas+electra wordt opgeteld en richting pvoutput gaat. Zoja, is dit te onderscheiden van elkaar dmv bijv. custom attributes in PVoutput?

Gr Dennis

dennissiemensma, 2019-10-23:
Bedankt voor je vraag. Heb je toevallig wat meer informatie over hoe je dat idee krijgt?

Voor zover ik weet [wordt er geen gas gebruikt](https://github.com/dennissiemensma/dsmr-reader/blob/v2.8.0/dsmr_pvoutput/services.py#L85) in de upload naar PVOutput.

ph0real, 2019-10-25:
Hoi Dennis. Ik geef toe dat er een aanname in zat. Wat ik in ieder geval constateer is dat de consumption cijfers in PVoutput altijd hoger liggen dan DSMR aangeeft. 
Bijv. van datum x zijn de consumption cijfers in PVoutput 5.316kwh.
Kijk ik vervolgens in DSMR dan is verbruik 3.69 kwh. Teruggeleverd die dag is 0.73.
Waar ik dus eigenlijk benieuwd naar ben is waar het verschil in zit. Als ik de gas cijfers erbij optel kom ik ook niet op dezelfde getallen uit, dus daar zit het m inderdaad niet in.

Gr. Dennis

dennissiemensma, 2019-10-25:
Je kunt debug logging aanzet en dan in de log van dsmr_backend kijken wat die exact uploadt.

* Zet debugging log level aan: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html
* Zoek (na een tijdje) in de log `/var/log/supervisor/dsmr_backend.log` naar ` - PVOutput | Uploading data:`.

ph0real, 2019-10-25:
Ik duik er dit weekend ff in, dankje!

Sent from ProtonMail Mobile

ph0real, 2019-10-30:
Ik Sluit m Dennis. Ik heb hoe dan ook besloten om verbruik lokaal op te slaan en niet langer via PVOutput.

closed 2019-10-30 by ph0real

dennissiemensma, 2019-10-31:
Prima, bedankt voor je terugkoppeling. Mocht je nog ergens anders tegenaan lopen, hoor ik het graag.