dsmrreader/dsmr-reader #755: Track weather API hammered DNS server

mrvanes, 2019-10-24:
### Wat gebruik je?
* kersverse DSMR pull

Sinds gisteravond 17.00 triggert de Track weather plugin ~1000 api.buienradar.nl DNS requests per 10 minuten, wat opviel in m'n Pi-hole dashboard. Disablen van de Track Weather plugin stopt de requests.
[image]

tiki1980, 2019-10-24:
Ik heb dit ook gezien idd, per seconde ongeveer één uitvraag.

[image]

mrvanes, 2019-10-24:
Het lijkt erop dat het API request mislukt, maar dat de plugin nogal agressief door blijft vragen. Dat heb ik verder niet onderzocht.

vincedh, 2019-10-24:
Hier hetzelfde probleem inderdaad... Voorlopig de module maar uitgezet.

dennissiemensma, 2019-10-24:
Bedankt voor jullie melding. Ik zie hier nu hetzelfde: ` Failed to read Buienradar API `.

Het is uiteraard niet de bedoeling dat die het zovaak opnieuw probeert. Zoiets was al eerder gemeld in #399 en ik dacht dat ik er toen ook iets aan had gedaan. Ik onderzoek eerst even de oorzaak.

dennissiemensma, 2019-10-24:
De API zelf lijkt kapot of veranderd: https://api.buienradar.nl/

Er komt wel een HTTP 200 OK terug, maar geen inhoud. Ik weet niet of er een sotring is, of dat RTL de API de nek om heeft gedraaid.

dennissiemensma, 2019-10-24:
Ik zie in https://www.buienradar.nl/overbuienradar/gratis-weerdata geen melding van wijzigingen, maar wel een referentie naar https://data.buienradar.nl/2.0/feed/json met soortgelijke gegevens.

Wellicht is dat een oplossing.

dennissiemensma, 2019-10-24:
Ik ben er ook achter waarom DSMR-reader het elke seconde opnieuw probeert.

Er zie [een check in](https://github.com/dennissiemensma/dsmr-reader/blob/v2.8.0/dsmr_weather/services.py#L48) om bij onbereikbare API 5 minuten te wachten. Hetzelfde [geldt voor onjuiste data](https://github.com/dennissiemensma/dsmr-reader/blob/v2.8.0/dsmr_weather/services.py#L74). Helaas klapt die er uit op een derde scenario: geen data. Dus ik zal het herschrijven naar een nog betere check.

dennissiemensma, 2019-10-24:
Ik heb zowel de foutafhandeling verbeterd als de nieuwe API van buienradar ingebouwd. Gaat mee met volgende release.

dennissiemensma, 2019-10-24:
Even dit issue pinnen zodat helder is dat het opgelost wordt.

dennissiemensma, 2019-10-25:
Uitgebracht in [v2.9.0](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.9.0).

closed 2019-10-25 by dennissiemensma