dsmrreader/dsmr-reader #761: Home Assistant automatische integratie

Giel538, 2019-10-26:
Is er vanuit gebruikers of ontwikkelaar(s) behoefte om een home assistant integratie te maken?

In Home Assistant moeten mensen nu alle sensoren (een sensor is de naam voor een weer te geven eenheid, bijvoorbeeld het vermogen, gas verbruik, spanning enz.) apart met de hand configureren. Dat wil zeggen het mqtt topic opgeven, naam, eenheid, icoon enz. 

Er is in Home Assistant een mogelijkheid om de sensoren automatisch te laten integreren. Dat houdt in dat vanuit DSMR reader de sensoren al aangemaakt worden en in Home Assistant kant en klaar zijn voor gebruik in je overzicht. Dit wordt gedaan via eenmalige mqtt berichten waarin je Home Assistant vertelt wat de sensor is, in welk mqtt topic de waarde aankomt, de eenheid enz. 

Mocht hier behoefte aan zijn dan biedt ik mijn hulp aan met het maken van deze berichten om de sensors te configureren.

dofl, 2019-10-27:
Zou erg handig zijn. Het was best een werk om via MQTT alles in Home Assistant te krijgen.

dennissiemensma, 2019-10-27:
Bedankt voor je aanbod. Hiervoor heb ik nog wat meer achtergrondinformatie nodig. 

Gaat dit puur via MQTT-berichten of is er een soort template die je ergens kan instellen in HA? En gaat dat dan op voor alle velden en formaten (JSON, losse topics) of hoe zie je dat voor je?

Giel538, 2019-10-27:
Jep. Een MQTT bericht ziet er dan zo uit:

payload:

{
  "unit_of_measurement" : "kW",
  "device_class" : "power",
  "value_template" : "{{ value_json }}",
  "icon" : "mdi:mdi:numeric",
  "state_topic" : "dsmr/reading/electricity_returned_1",
  "name" : "Stand opgewekt laag",
  "unique_id" : "dsmrreadermqtt_electricity_returned_1",
  "device" : {
    "identifiers" : [ "dsmrreadermqtt" ],
    "name" : "DSMR Reader",
    "sw_version" : " 2.7",
    "manufacturer" : "Dennis"
  }
}

Deze payload stuur je naar het topic: homeassistant/sensor/electricity_returned_1/config

Voor elke waarde die je automatisch in home assistant wil krijgen moet je dit dus een keer versturen. Het is dus eenmalig een hoop typewerk, ook al zal je in je code al werken met variabelen en dan is het een kwesite van variabelen op de juiste plek neerzetten. 

[image]

[image]

Op de MQTT integratie pagina in home assistant wordt het weergegeven als een apparaat met al je bijbehorende sensoren.
[image]

https://www.home-assistant.io/docs/mqtt/discovery/

dennissiemensma, 2019-10-27:
Ik ben zelf niet bekend met HA, maar zou ik als alternatief een knip-en-plak config kunnen opnemen in de DSMR-reader docs, uitgaande van de standaard MQTT-topics in de applicatie? Of kan er alleen via MQTT een sensor toegevoegd worden in HA?

Erik050572, 2019-10-29:
Zouden jullie misschien ook willen kijken of een koppeliing met een Fibaro HC2 mogelijk is.
Deze werkt helaas niet met MQTT

Giel538, 2019-10-29:
> 
> 
> Ik ben zelf niet bekend met HA, maar zou ik als alternatief een knip-en-plak config kunnen opnemen in de DSMR-reader docs, uitgaande van de standaard MQTT-topics in de applicatie? Of kan er alleen via MQTT een sensor toegevoegd worden in HA?

Het kan via MQTT of het kan door een "custom integration" te maken. Maar dan zit je in python alle sensoren te programmeren. En ik weet niet of dit dan werkt met MQTT of alleen met API calls.

Knip en plak zit er eigenlijk niet in. Het kan wel grotendeels maar je moet dan nog steeds zelf de "payload" zoals hierboven  in een topic publishen. En dat voor elke waarde die je wilt in home assistant. Al de payloads en topics kan je natuurlijk documenteren. 

Er schiet me nog wel iets te binnen maar dat moet ik zelf eerst onderzoeken. Misschien dat ik een automation kan maken die je eenmalig draait en dat daarin alle mqtt berichten verstuurd worden. 

Wordt vervolgt

dennissiemensma, 2019-10-29:
@Erik050572 daarvoor is wat meer achtergrondinformatie nodig. Als er een API is, kan er in principe een plugin voor DSMR-reader gemaakt worden die als doorgeefluik functioneert. Je kunt er het beste een apart issue voor maken hier, gezien het niet over HA gaat.

dennissiemensma, 2019-10-29:
@Giel538  ik denk dat er wel iets voor te maken is, mits het eenmalige MQTT-berichten zijn. De huidige implementatie van MQTT in DSMR-reader werkt als een uitgaande queue, dus daar zijn wel dingen op klaar te zetten, mits het 1 formaat is en niet te ingewikkeld.

Giel538, 2019-11-03:
Het leek wel te lukken tot ik Home Assistant opnieuw starte. Toen waren de sensoren verdewenen. 
Na elke restart moeten ze opnieuw aangemaakt worden....

Ik weet niet precies hoe dit werkt. Ik weet alleen dat als je het goed instelt Home Assistant een wake up message stuurt via MQTT. Dit zou een trigger kunnen zijn om de mqtt berichten met sensor info weer opnieuw te sturen. Ik ben hier ook niet mee bekend.

Ik laat het voor nu even los. Ik had gehoopt dat het makkelijker zou gaan.

depl0y, 2019-11-10:
Toevallig ben ik met een component bezig voor Home Assistant die automatisch sensors vanuit MQTT toevoegd (voor DSMR Reader). Enige limitatie (omdat ik vrij lui ben) die er vanuit dit custom component is, is dat het alleen werkt met Split topic en de default mappings.

[image]

Wanneer ik alle topics in mijn custom component heb gestopt kan ik hem eventueel wel bij Home Assistant aanbieden, zodat het er standaard in komt, dat heeft echter als nadeel dat ik dat ding ook echt moet gaan onderhouden. 😃 Voordeel is, dat iedereen die MQTT en Split Topic gebruikt, dan de volgende configuratie aan HA kan toevoegen en het werkt verder zonder configuraties:

```
sensor:
  platform: dsmr_reader
```

dennissiemensma, 2019-11-10:
Bedankt voor je aanbod. Volgens mij kloppen ze bijna allemaal:

- `dsmr/reading/extra_device_delivered` => [m3](https://github.com/dennissiemensma/dsmr-reader/blob/v2.11.0/dsmr_datalogger/models/reading.py#L82)

depl0y, 2019-11-10:
@dennissiemensma Ik twijfel vooral over de kW en de kWh  bij electricity currently delivered etc. Ik zal die m3 even goedzetten voor extra_device_delivered.

Ik ben nu aan het testen en code aan het opruimen, maar ik hoop eigenlijk dat ik hem misschien deze week al kan aanbieden bij Home Assistant. Dit is versie 1 waar de mapping gewoon vast staat, maar ik gok dat ik ook wel custom mapping kan aanbieden in de toekomst (als er vraag naar is).

Giel538, 2019-11-10:
Top @depl0y !

Ik denk dat custom mapping niet nodig is, maar dan spreek ik puur voor mij zelf. Het idee moet juist zijn dat je met zo min mogelijk werk een integratie kan doen. 

Het actuele verbruik is in W of kW.  
Als je 1000W een uur lang hebt gebruikt is het actuele vermogen ergens in dat uur 1000W en na een uur heb je 1000W/h ofwel 1kWh verbruikt. 

Ga je de friendly name en icons ook automatisch aan laten maken?

dennissiemensma, 2019-11-10:
@depl0y je kunt [hier de omschrijving van de meeste velden](https://github.com/dennissiemensma/dsmr-reader/blob/v2.11.0/dsmr_datalogger/models/reading.py#L16) vinden.

depl0y, 2019-11-10:
@Giel538 Friendly names in ieder geval, al is dat niets anders dan de naam van de variabele? Of je moet met wat voorstellen komen. Icons eigenlijk hetzelfde, ik wil wel even kijken of ik bijpassende iconen kan vinden (ze moeten [hier](http://materialdesignicons.com/) vandaan komen)

@dennissiemensma Ik ga de lijst even bij langs, thanks.

Ik ben op dit moment v1 van het component door de tests van Home Assistant aan het laten lopen. Source code van het component kan je [hier](https://github.com/depl0y/home-assistant/tree/dsmr-reader/homeassistant/components/dsmr_reader) al vinden. Met een beetje geluk kan ik vandaag of morgen een PR maken.

Giel538, 2019-11-10:
[image]

Het is inderdaad de naam van de variabele. Maar wat ik bij jou zie is het nu bijvoorbeeld: reading.phase_voltage_l1. Je kan er ook bijvoorbeeld ook van maken: Voltage L1.

Voor meterstanden gebruik ik: mdi:counter. Voor actueel vermogen: mdi:flash (maar die pakt die denk ik standaard). voor gas: mdi:fire en de spanning mdi:alpha-v-box. De laatste heb ik een V omdat spanning uitgedrukt wordt in Volt. Ik kon er verder even niets bij verzinnen :)

depl0y, 2019-11-11:
Ok, er van uit gaande dat je dit in HA ook nog kan customizen, heb ik dit als default values. @Giel538 en @dennissiemensma mochten jullie suggesties, op- en aanmerkingen hebben, dan hoor ik het graag :)

[image]

Giel538, 2019-11-11:
Ziet er goed uit! 

die 2 gas topics die je aangeeft hebben dezelfde waarde volgens mij. Zit er dus dubbel in.

Wat je nog zou kunnen overwegen is om de waardes van actueel verbruik om te rekenen naar watt ipv kilo watt. Zoals in je screenshot zie je nu bij usage L1 0.007 staan. Ik reken ze zelf om naar 70 W in dit geval. Ik heb die vraag hier ook eerder voorbij zien komen om dat al in dsmr reader zelf te doen maar dit was niet of lastig mogelijk. 
Je dsmr versie kan je omzetten naar 5.0 ipv 50. Ik denk dat dat in alle gevallen delen door 10 is. 
Je elektricity tariff zou ik om zetten naar low and high. ipv 1 en 2. 
Power usage L1, L2 en L3 en power return L1, L2 en L3 zijn eigenlijk ook "current" waardes.

Waar je mij persoonlijk blij mee zou maken en misschien anderen ook is dat je kan aangeven of je een 1-fase of 3-fase meter hebt. En dat je in het geval van een 1-fase meter niet alle L2 en L3 sensoren krijgt.

Dan krijg je zo iets:

- platform: dsmr_reader
   no_of_phases: 1 (of 3)

Vervolgens kan je in je eigen code het als volgt opnemen:

NO_OF_PHASES = "no_of_phases"
DEFAULT_NO_OF_PHASES  = "3"

Dus als je hem niet invult maakt die standaard alles aan.  Ik weet niet helemaal zeker of deze code zo werkt maar als ik goed naar een andere integratie kijk in home assistant werkt het zo.

dennissiemensma, 2019-11-11:
Wellicht is het gebruik van de fases automatisch te detecteren door een check op `null` of `None`. Als ze niet ondersteund worden door je meter, defaulten ze naar die waarden. Anders zijn ze 0 of hoger.

depl0y, 2019-11-11:
Het toevoegen van de fases die je niet gebruikt is niet zo'n groot probleem en bij versie 1 zal ik ze er ook gewoon alle 3 in zetten. Het is inderdaad een optie om het als configuratie toe te voegen.

Verder ben ik bij de definities zogenaamde transformatie functies aan het toevoegen, zodat elke waarde omgezet kan worden naar iets anders. De waardes omzetten van kW naar W, daar heb ik niet echt een mening over moet ik zeggen, volgens mij is dat meer een soort gewenning. Ik zou daarin graag consistent blijven, of ALLES naar W, of alles in kW houden 😃 

Versie 1.0 heb ik alvast een PR van gemaakt voor HA, aangezien het goedkeuren soms tijden duurt, leek dat me al handig. Verder updates kunnen daarna gewoon toegevoegd worden.

dennissiemensma, 2019-11-11:
Mee eens. In een latere versie kan altijd nog een optie toegevoegd worden om optioneel een multiplier of iets toe te passen. Als iedereen daar een PR op kan doen, is dat ook tzt zo doorgevoerd vermoed ik.

depl0y, 2019-11-12:
Inderdaad, ik heb de PR die ik open heb staan nog even aangevuld met een voorbeeld voor de 'transforms' zoals ik ze noem. Je kan dat hier bekijken: https://github.com/home-assistant/home-assistant/pull/28701/files#diff-3e50bb6168138a1d068c14a8d936fbb9

Je definieert een methode die je aan de `transform` property van een definitie hangt. Een definitie is op dit moment niets meer dan een dictionary, maar mocht het nodig zijn, kan ik daar altijd nog classes van maken om het wat op te leuken.

Bijvoorbeeld het DSMR versie nummer wordt zo getransformed:

```
def dsmr_transform(value):
    """Transform DSMR version value to right format."""
    return float(value) / 10
```

En dat is zo opgenomen in de definitie:

```
"dsmr/meter-stats/dsmr_version": {
        "name": "DSMR version",
        "icon": "mdi:alert-circle",
        "transform": dsmr_transform,
},
```

Ik zou zelfs zo ver kunnen gaan dat je transforms zou kunnen definieren in de HA configuratie, maar dan kan je net zo goed een template sensor aanmaken (die bijvoorbeeld kW omrekent naar W). Voor mijn gevoel valt dit ook een beetje buiten de scope van deze plugin.

dennissiemensma, 2019-11-12:
Die transformers lijken me een prima aanvulling. Ik zou het verder lekker simpel houden eerst qua scope.

depl0y, 2019-11-29:
Kleine update, het eerste PR is goegekeurd en gemerged in dev. Wat zou moeten betekenen dat in 1 van de volgende releases, de integratie beschikbaar moet komen.

dennissiemensma, 2019-11-29:
Top!

megapearl, 2019-11-30:
Goed bezig! hier zoek ik ook al een tijdje naar, ik heb nu alle sensoren in homeassistant met de hand gemaakt, werkt natuurlijk ook prima, maar een homeassistant component is natuurlijk veel mooier en makkelijker.

depl0y, 2019-12-12:
En live: https://www.home-assistant.io/integrations/dsmr_reader/

Updaten naar versie 0.103.0 en de volgende sensor toevoegen:

```
sensor:
  - platform: dsmr_reader
```

2xdehelft, 2019-12-12:
Ik heb deze direct toegepast vandaag, werkt perfect! Dank hiervoor!

Misschien heel ver gedacht, maar zou de DSMR_reader niet als addon gemaakt kunnen worden voor Homeassistant, en dan direct met Ingress ondersteuning zodat de webpagina ook bruikbaar is vanuit Homeassistant?

adebree, 2019-12-14:
@depl0y top gedaan, dank! :) 

Wat nog fijn zou zijn is als je de yaml van je full example (je laatste screenshot hierboven) kan delen. Ik denk dat die info een mooi vertrek punt is voor mensen om de DSMRReader data in hun HA te gaan gebruiken. Het zijn best een hoop entities namelijk :D

ronniebee, 2019-12-14:
Ik ben bezig dsmr live te krijgen en loop bij parser aan tegen de volgende melding:

Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd       
De statusinformatie wordt gelezen... Klaar
**_E: Kan pakket dmsr-parser niet vinden_**

Wat dien ik te doen? (ben een newbee en geen programeur helaas).

Verder hen ik ook de dsmr-reader in mij yaml gezet. maak krijg ook in mijn notifcaties:

_The following components and platforms could not be set up:
sensor.dsmr
Please check your config._

wie kan mij helpen? wil het graag werkend krijgen..

depl0y, 2019-12-14:
@ronniebee , volgens mij heb je de configuratie niet helemaal goed overgenomen. Er is een DSMR component en een DSMR-Reader component (waar we het hier over hebben).

@adebree Bedoel je ook de configuratie in de interface? Want meer dan de sensor toevoegen is niet nodig om HA te koppelen aan DSMR Reader (via MQTT), in de interface wil je waarschijnlijk niet ALLE sensoren toevoegen, maar ik zal de configuratie van het tabblad hier ook even neerzetten.

adebree, 2019-12-14:
@depl0y yes, daarom zou een full example wel handig zijn zodat je dat als startpunt kan gebruiken om entities weer uit te snoeien. Wat ik uit luiheid net gedaan heb is de `custom:auto-entities` card gebruikt met een wildcard op `sensor.dsmr_*` waardoor ze allemaal met values te zien zijn. Van daaruit ben ik gaan cherry picken wat ik wilde hebben.

```
      - type: custom:auto-entities
        card:
          type: entities
          show_header_toggle: false
          title: DSMR
        filter:
          include:
            - entity_id: "sensor.dsmr_*"         
        sort:
          method: entity_id 
```

https://github.com/thomasloven/lovelace-auto-entities en dan via `HACS` geinstalleerd.

ronniebee, 2019-12-14:
> @ronniebee , volgens mij heb je de configuratie niet helemaal goed overgenomen. Er is een DSMR component en een DSMR-Reader component (waar we het hier over hebben).
> 

in mijn yaml staat:

sensor:
  - platform: dsmr-reader
  
  - platform: dsmr
    port: /dev/ttyUSB0
    dsmr_version: 2.2

    group:
    meter_readings:
        name: Meter readings
        entities:
         - sensor.power_consumption_low
         - sensor.power_consumption_normal
         - sensor.power_production_low
         - sensor.power_production_normal
         - sensor.gas_consumption

maar mijn cofig geeft aan:

Platform error: sensor - Integration 'dsmr-reader' not found.

depl0y, 2019-12-14:
@adebree Bij deze:

```
entities:
  - entity: sensor.dsmr_reading_electricity_delivered_1
  - entity: sensor.dsmr_reading_electricity_returned_1
  - entity: sensor.dsmr_reading_electricity_delivered_2
  - entity: sensor.dsmr_reading_electricity_returned_2
  - entity: sensor.dsmr_reading_electricity_currently_delivered
  - entity: sensor.dsmr_reading_electricity_currently_returned
  - entity: sensor.dsmr_reading_phase_currently_delivered_l1
  - entity: sensor.dsmr_reading_phase_currently_delivered_l2
  - entity: sensor.dsmr_reading_phase_currently_delivered_l3
  - entity: sensor.dsmr_reading_phase_currently_returned_l1
  - entity: sensor.dsmr_reading_phase_currently_returned_l2
  - entity: sensor.dsmr_reading_phase_currently_returned_l3
  - entity: sensor.dsmr_reading_extra_device_delivered
  - entity: sensor.dsmr_reading_phase_voltage_l1
  - entity: sensor.dsmr_reading_phase_voltage_l2
  - entity: sensor.dsmr_reading_phase_voltage_l3
show_header_toggle: false
title: reading
type: entities
```

```
entities:
  - entity: sensor.dsmr_consumption_gas_delivered
  - entity: sensor.dsmr_consumption_gas_currently_delivered
  - entity: sensor.dsmr_consumption_gas_read_at
show_header_toggle: false
title: gas
type: entities
```

```
entities:
  - entity: sensor.dsmr_day_consumption_electricity1
  - entity: sensor.dsmr_day_consumption_electricity2
  - entity: sensor.dsmr_day_consumption_electricity1_returned
  - entity: sensor.dsmr_day_consumption_electricity2_returned
  - entity: sensor.dsmr_day_consumption_electricity_merged
  - entity: sensor.dsmr_day_consumption_electricity_returned_merged
  - entity: sensor.dsmr_day_consumption_electricity1_cost
  - entity: sensor.dsmr_day_consumption_electricity2_cost
  - entity: sensor.dsmr_day_consumption_electricity_cost_merged
  - entity: sensor.dsmr_day_consumption_gas
  - entity: sensor.dsmr_day_consumption_gas_cost
  - entity: sensor.dsmr_day_consumption_total_cost
show_header_toggle: false
title: day
type: entities
```

```
entities:
  - entity: sensor.dsmr_meter_stats_dsmr_version
  - entity: sensor.dsmr_meter_stats_electricity_tariff
  - entity: sensor.dsmr_meter_stats_power_failure_count
  - entity: sensor.dsmr_meter_stats_long_power_failure_count
  - entity: sensor.dsmr_meter_stats_voltage_sag_count_l1
  - entity: sensor.dsmr_meter_stats_voltage_sag_count_l2
  - entity: sensor.dsmr_meter_stats_voltage_sag_count_l3
  - entity: sensor.dsmr_meter_stats_voltage_swell_count_l1
  - entity: sensor.dsmr_meter_stats_voltage_swell_count_l2
  - entity: sensor.dsmr_meter_stats_voltage_swell_count_l3
  - entity: sensor.dsmr_meter_stats_rejected_telegrams
show_header_toggle: false
title: meter-stats
type: entities
```

depl0y, 2019-12-14:
@ronniebee ik heb geen idee wat jij aan het doen bent, maar dit is niet het issue hiervoor, lijkt me. Volgens mij heb je 2 dingen door elkaar lopen. Dit issue was voor het ondersteunen van **DSMR Reader** in Home Assistant en niet het **DSMR** component in Home Assistant.

ronniebee, 2019-12-14:
> @ronniebee ik heb geen idee wat jij aan het doen bent, maar dit is niet het issue hiervoor, lijkt me. Volgens mij heb je 2 dingen door elkaar lopen. Dit issue was voor het ondersteunen van **DSMR Reader** in Home Assistant en niet het **DSMR** component in Home Assistant.

Excuus hiervoor, ik ben een newbee en die veel doet door trail en error (kost zeer veel tijd helaas). Ik heb je reader werkend gekregen! thx, nu de waarden nog.. maar dat zal ik hier niet meer vragen maar op een ander forum.

depl0y, 2019-12-15:
> Excuus hiervoor, ik ben een newbee en die veel doet door trail en error (kost zeer veel tijd helaas). Ik heb je reader werkend gekregen! thx, nu de waarden nog.. maar dat zal ik hier niet meer [...]

No problem. Wat nodig is om de waardes van DSMR Reader in HA te krijgen:

- DSMR Reader (je zit al op de juiste github repo)
- MQTT
- DSMR Reader instellen dat hij data 'split topic' naar MQTT stuurt
- HA DSMR Reader sensor toevoegen aan je HA configuratie

Giel538, 2019-12-15:
@ronniebee 

Zorg ook dat dit stukje uit je home assistent configuratie is:

platform: dsmr
port: /dev/ttyUSB0
dsmr_version: 2.2

Je kan je seriële poort namelijk maar 1x gebruiken. En als hij bij home assistent in gebruik is kan dame reader hem niet gebruiken.

ronniebee, 2019-12-15:
Dank voor de hulp! erg blij mee

Wat ik he gedaan:
-Parser geïnstalleerd 
-de handleiding gevolgd van ualex73/dsmr-reader-docker
  (docker run en compse)
-yaml aangepast
( 4 uur mee aan het klooien geweest, moet mijzelf alles aanleren van docker, portainer, ssh, ubuntu ha en hacs etc etc.. . maar stapje bij stapje). De dsmr wil ik erg graag hebben draaien dus bijt me hierin vast :-)
 
maar nu...
>
> No problem. Wat nodig is om de waardes van DSMR Reader in HA te krijgen:
> 
> * DSMR Reader (je zit al op de juiste github repo)
check!
> * MQTT
check!
> * DSMR Reader instellen dat hij data 'split topic' naar MQTT stuurt
??
> * HA DSMR Reader sensor toevoegen aan je HA configuratie
check
sensor:
  - platform: dsmr_reader 
  - platform: dsmr
    port: /dev/ttyUSB0
    dsmr_version: 2.2

Dank, ik kom steeds verder. komt een newbee vraag .. Hoe stel ik dsmr reader in voor data split. Geen idee nl....

En of of dsmr wel goed werkt vraag ik me af.. Ik heb getracht dit proces te stoppen maar hier reageert ssh op met:

Named volume "<dir>/db:/var/lib/postgresql/data:rw" is used in service "db-dsmr" but no declaration was found in the volumes section.  

In portainer zie ik dsmr nl wel draaien en kan ik hem wel stoppen..

Ik heb de stappen gevolgd van de HA site en parser geinstalleerd. Zie nu dus wel allemaal enoteiten maar geen waarden.  Ik weet niet of dit het juiste platform is voor dit soort vragen dus excuus op voorhand. 

groet uit een winderig maar droog  Zwolle..

Giel538, 2019-12-15:
Dsmr reader kan je in je browser openen. In je docker geef je tijdens het installeren van dsmr reader aan via welke port. Dit is normaal port 80 maar dit kan je met de optie -p xxxx:80 aanpassen naar elk gewenste port. Met het lokale op adres en port nummer kom je dan in dame reader. Bekijk eerst hier is of er data uit je meter binnenkomt

ronniebee, 2019-12-15:
> Dsmr reader kan je in je browser openen. In je docker geef je tijdens het installeren van dsmr reader aan via welke port. Dit is normaal port 80 maar dit kan je met de optie -p xxxx:80 aanpassen naar elk gewenste port. Met het lokale ip adres en port nummer kom je dan in dsmr reader (dus bij. [Http://192.168.0.120:80](Http://192.168.0.120:80)). Bekijk eerst hier is of er data uit je meter binnenkomt

Helaas. ik werk met poort 8888 (vlgs docker alex) dus met 192.168.x.xxx:8888  
resultaat:

Deze site is niet bereikbaar192.168.1.xx heeft de verbinding geweigerd.

Giel538, 2019-12-15:
Ok. Als je portainer hebt geïnstalleerd dan kan je die openen in je browser met het op adres en port 9000. Als je ingelogd bent moet je naar tab containers. Daar staat dsmr en dsmrdb. Als je dsmr aanklikt kan je ook een log openen. Misschien staat daar wat nuttigs in.

ronniebee, 2019-12-15:
Ik mis dsmr db.. iets gaat daar niet goed... 

de log van dsmr geeft aan:

...........................................................
2019-12-15 11:07:01
Start DSMR Reader - Mode=SERVER
...........................................................
2019-12-15 11:08:04
Start DSMR Reader - Mode=SERVER

...........................................................

Giel538, 2019-12-15:
Heb je ze geïnstalleerd met docker-compose? Ik kan mijn compose wel posten eventueel. 

Wat kan helpen is als je docker-compose commando uitvoert is om er sudo voor te zetten. Dus: sudo docker-compose up -d

Dan run je het met admin rechten

ronniebee, 2019-12-15:
Hoi Giel, dank voor je hulp! zeer gewaardeerd... Mij vrouw vraagt al waneer ik een keer klaar ermee ben.. als het werkt :-)

Bijgaand mijn compose.yml

GNU nano 2.9.3 /home/ronniexxx/docker/docker-compose.yml
version: "3.6"
services:
db-dsmr:
image: postgres
container_name: db-dsmr
volumes:
-

/db:/var/lib/postgresql/data
restart: unless-stopped
environment:
- POSTGRES_USER=dsmrreader
- POSTGRES_PASSWORD=dsmrreader
- POSTGRES_DB=dsmrreader
networks:
- net-dsmr
dsmr:
container_name: dsmr
image: ualex73/dsmr-reader-docker
container_name: dsmr
restart: unless-stopped
- POSTGRES_PASSWORD=dsmrreader
- POSTGRES_DB=dsmrreader
networks:
- net-dsmr
dsmr:
container_name: dsmr
image: ualex73/dsmr-reader-docker
container_name: dsmr
restart: unless-stopped
volumes:
- ./dsmr/backups:/home/dsmr/app/backups
- /dev:/dev
environment:
- DB_HOST=db-dsmr
- DSMR_USER=admin
- DSMR_EMAIL=root@localhost
- DSMR_PASSWORD=admin
- VIRTUAL_HOST=localhost
ports:
- 8888:80
devices:
- /dev/ttyUSB0:/dev/ttyUSB0
networks:
- net-dsmr
networks:
net-dsmr:
driver: bridge

Giel538, 2019-12-15:
`version: '3.6'

services:
  dsmrdb:
    image: postgres:alpine
    container_name: dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /data/dsmr/dsmrdb:/var/lib/postgresql/data
    environment:
      - TZ=Europe/Amsterdam
      - PG_TZ=Europe/Amsterdam
      - POSTGRES_USER=dsmrreader
      - POSTGRES_PASSWORD=dsmrreader
      - POSTGRES_DB=dsmrreader

  dsmr:
    image: xirixiz/dsmr-reader-docker:latest
    container_name: dsmr
    depends_on:
      - dsmrdb
    links:
      - dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /data/dsmr/backups:/home/dsmr/app/backups
      - /data/dsmr/settings.py:/dsmr/dsmrreader/settings.py
    environment:
      - TZ=Europe/Amsterdam
      - VIRTUAL_HOST=localhost
#      - DSMRREADER_BACKEND_SLEEP=10
#      - DSMRREADER_DATALOGGER_SLEEP=10
    ports:
      - 7777:80
      - 7779:443
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
`

Giel538, 2019-12-15:
Probeer die. Ik zie bij jou namelijk niet de installatie van dsmr-db

ronniebee, 2019-12-15:
helaasssss. je mag van me wel even teamviewen ...

Stephan296, 2019-12-15:
@ronniebee Je bent 2 dingen door elkaar aan het halen. Je bent met docker bezig en met een sensor waarbij je de slimme meter rechtstreeks op HA aansluit.. Dit kan niet.

ronniebee, 2019-12-15:
> @ronniebee Je bent 2 dingen door elkaar aan het halen. Je bent met docker bezig en met een sensor waarbij je de slimme meter rechtstreeks op HA aansluit.. Dit kan niet.

ik ben een newbee... dus de parser killen?

Stephan296, 2019-12-15:
@dennissiemensma ik krijg geen verbinding met mqtt. Hoeveel karakters mag het wachtwoord hebben? (Ben hier vaker tegenaan gelopen namelijk

ronniebee, 2019-12-15:
> Probeer die. Ik zie bij jou namelijk niet de installatie van dsmr-db

moet ik deze stappen nog doorlopen? ik ga via docker dus vraag me af of dit nodig is..

https://dsmr-reader.readthedocs.io/nl/v2/installation/quick.html

Giel538, 2019-12-15:
Nee dat hoeft niet. Dat gebeurt in docker.

Maar even terug. Op wat voor systeem draai je? Zit rocker op hetzelfde systeem als waar je p1 kabel op zit? Draait Matt en home assistent hier ook op?

Wat gaat er mis als je de code van mij probeert uit te voeren met het commando: sudo docker-compose up -d ?

ronniebee, 2019-12-15:
Hoi Giel, dit krijg ik:

ronniexxx@ronald:~$ sudo docker-compose up -d ?
[sudo] password for ronniebee: 
ERROR: 
        Can't find a suitable configuration file in this directory or any
        parent. Are you in the right directory?

        Supported filenames: docker-compose.yml, docker-compose.yaml

Giel538, 2019-12-15:
Aha, 

Haal deze regel uit mijn configuratie:

      - /data/dsmr/settings.py:/dsmr/dsmrreader/settings.py

en doe dan nogmaals sudo docker-compose up -d 

Kijk of de melding dan weg is

ronniebee, 2019-12-15:
yess stapje verder..
en nu?

[image]

Giel538, 2019-12-15:
geef het een paar minuten en wacht of er data binnenkomt. Bij statistieken moet je je meterstanden gaan zien. Zie je die niet dan denk ik dat je misschien niet naar de goede usb port linkt in de docker-compose file:

    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0

Dat zou dan misschien ttyUSB1 ofzo kunnen zijn

ronniebee, 2019-12-15:
ttyusb0.

helaas nog geen data. nog andere instellingen die ik kan wijzigen?

dennissiemensma, 2019-12-15:
> @dennissiemensma ik krijg geen verbinding met mqtt. Hoeveel karakters mag het wachtwoord hebben? (Ben hier vaker tegenaan gelopen namelijk

@Stephan296 vanuit DSMR-reader 256 tekens. Wellicht dat de MQTT-library er minder toe staat, maar dat kan ik me bijna niet voorstellen.

ronniebee, 2019-12-15:
dsmr db geeft aan in portainer:

PostgreSQL Database directory appears to contain a database; Skipping initialization

[34 more replies]

closed 2019-12-16 by dennissiemensma