dsmrreader/dsmr-reader #763: Foutmelding Buienradar API

michielvisser, 2019-10-27:
### Wat gebruik je?
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: 2.9
* Hardware *(RaspberryPi 3, 4 of iets anders)*:  CentOS server (data komt binnen via rest API)
* DSMR-protocol slimme meter *(v2, v4 of v5)*: v4

Het originele issue was al gesloten, vandaar dat ik een nieuwe open. Vandaag de update gedaan naar versie 2.9 in verband met de buienradar API. Deze geeft echter nog steeds een foutmelding: Oct. 27, 2019: 

Failed to read Buienradar API: HTTPSConnectionPool(host='data.buienradar.nl', port=443): Max retries exceeded with url: /2.0/feed/json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f27e0e8dda0>: Failed to establish a new connection: [Errno -2] Name or service not known',))

Geen idee of dat de API op dit moment gewoon problemen heeft of dat er wat anders aan de hand is. Overigens netjes dat er nu een foutmelding verschijnt, ik was al aan het zoeken geslagen in versie 2.8 waarom dit ineens niet meer werkte (ik draai een behoorlijk customized setup in vergelijking tot een standaard RPi).

Tijdelijk opgelost met workaround door in mijn dashboards de temperatuurtracking uit te zetten.

dennissiemensma, 2019-10-27:
Bedankt voor je melding. Krijg je nog steeds fouten?

De URL die (in v2.9) gebruikt wordt is: https://data.buienradar.nl/2.0/feed/json. Je kunt even kijken of die voor jou bereikbaar is. Hier lijkt die gewoon te werken.

michielvisser, 2019-10-28:
Dank je wel, Dennis. Het issue bleek inderdaad in de resolving the zitten, geen idee wat daar precies gebeurd is. Het is daarmee opgelost.

closed 2019-10-28 by michielvisser