dsmrreader/dsmr-reader #767: Test & coverage: pytest unrecognized arguments:

MarbBram, 2019-10-29:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.9`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3 B+`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

*Beschrijf hier je verhaal...*

The easiest way to run tests is to use the in-memory tests:

source ~/.virtualenvs/dsmrreader/bin/activate
./tools/quick-test.sh

Resultaat:
--- Testing with SQLite (4 processes)...
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --cov --cov-report=html --cov-report=term                                    --ds=dsmrreader.config.test.sqlite -n
  inifile: None
  rootdir: /home/dsmr/dsmr-reader

real    0m2,192s
user    0m1,627s
sys     0m0,544s

--- Running Pylama for code audit...
OK

dennissiemensma, 2019-10-29:
Bedankt voor je melding. Ik neem aan dat je dit **niet op je raspberrypi uitvoert** he, maar op een aparte ontwikkelomgeving?

Werkt dit wel als je het direct uitvoert?
```
pytest --ds=dsmrreader.config.test.sqlite --cov --cov-report=html
```

MarbBram, 2019-10-30:
Ik voer dit uit op mijn RP, is dat niet de bedoeling?
    
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --ds=dsmrreader.config.test.sqlite
  inifile: None
  rootdir: /home/dsmr/dsmr-reader

dennissiemensma, 2019-10-31:
Ik denk dat het goed is dat je je intenties aangeeft. Ik weet niet of je een softwareontwikkelaar bent, maar normaal is het not-done om te ontwikkelen op 'productie'. Waarbij productie in dit geval de Pi is die direct aan je slimme meter hangt. 
Als je wilt ontwikkelen kan ik je aanraden om er simpelweg een ontwikkelomgeving voor op te zetten op je PC (mits je Linux gebruikt). Ik heb afdoende tools gemaakt om de input van de slimme meter te simuleren. Daarnaast zul je dev-tools moeten installeren op de ontwikkelomgeving. Dat is de reden dat je foutmeldingen krijgt.

closed 2019-11-08 by dennissiemensma