dsmrreader/dsmr-reader #769: Enhance DSMR for ORBIS codes used by Smarty smartmeters in Luxembourg

JeanMiK, 2019-10-30:
The smartmeters being introduced in Luxembourg (called 'Smarty') are derived from the Dutch DSMR4/5 Standard.

They use the DSMR5 physical P1 port and transmission Speed and similar telegrams, with the following differences:
1) Still using the DSMR4 10 seconds intervals between reading publications
2) Readers are typically 3-phase readers, but for the moment, the per-phase data is not yet incldued in the telegram, but can only be viewed on the reader's display; this is supposed to get changed
3) Communication on the P1 port is encrypted with a personal key changing from meter to meter and that users have to request from the metering company (Luxmetering)
4) In general, the meters support only 1 tariff, and don't split the readings by tariff. Only a global reading is sent (1-0:1.8.0/1-0:2.8.0 instead of 1-0:1.8.1/1-0:2.8.1 respectively 1-0:1.8.2/1-0:2.8.2)

In order to be able to use the dsmr-reader program with this smartmeter, one needs to work around the differences 3 and 4.

For 3, there exists proxy software doing the decryption and forearding the decrypted telegrams on a serial port that dsmr-reader can use, so there is no immediate need to change the dsmr-reader (even though I might make a pull request to integrate a decryption option directly into dsmr-reader).

For 4, the easiest is to map 1-0:1.8.0/1-0:2.8.0 onto 1-0:1.8.1/1-0:2.8.1 and set  1-0:1.8.2/1-0:2.8.2 to 0. 

I will enter a pull request for a proposed implementation.

dennissiemensma, 2019-10-31:
Thanks for your input. There is another similar issue: #750. 

I will need more information regarding the DSMR-specs used in Luxembourg before I can decide how to resolve this issue. If there happens to be differences among telegrams in Netherlands, Belgium and Luxembourg, I might end up by implementing a dynamic mapping config, similar to MQTT-configuration in DSMR-reader.

Do you happen to have a Luxembourg telegram for me? You can strip any smart meter serials numbers from it.

dennissiemensma, 2019-10-31:
As for the proxy software, I think it should be something outside of this project to reduce complexity. Depending on how complex it actually is.

JeanMiK, 2019-10-31:
Here is a sample of my own meter:

```
/Lux5\xxxxxxxxx_D

1-3:0.2.8(42)
0-0:1.0.0(191031142239W)
0-0:42.0.0(xxxxxxxxxxxxxxxxxxxxxxxx)
1-0:1.8.0(005675.956*kWh)
1-0:2.8.0(000000.002*kWh)
1-0:3.8.0(000120.721*kvarh)
1-0:4.8.0(000697.336*kvarh)
1-0:1.7.0(03.074*kW)
1-0:2.7.0(00.000*kW)
1-0:3.7.0(00.000*kvar)
1-0:4.7.0(00.000*kvar)
0-0:17.0.0(27.600*kVA)
0-0:96.3.10(1)
0-0:96.13.0()
0-0:96.13.2()
0-0:96.13.3()
0-0:96.13.4()
0-0:96.13.5()
!D65F
```

Here is the specification of the Luxembourgish smart meters:
https://smarty.creos.net/wp-content/uploads/P1PortSpecification.pdf

The list of Orbis codes is on pages 7/8, but for the moment, only the codes marked as 'Default list' really get delivered on all meters (some people are luckier than me and get more info, like per phase current or power).

As far as current / voltages / power is concerned, the Orbis codes are the same than in the Dutch specifications. But for the delievered / returned Energy, the different countries seem to have taken different point of views (1.7.1 for tariff 1 / 1.7.2 for tariff 2 in the Netherlands, 1.7.0 for day-time, 1.7.1 for night-time in Belgium, 1.7.0 as single tariff in Luxembourg [no idea what the meter uses if one subscribes to a 2-tariff model]).

My approach would be to define all the 'interesting' Orbis codes in dsmr_datalogger/dsmr.py, and then to map them in dsmr_datalogger/services.py, either based on a country Setting or by an empiric, pragmatic mapping like in my proposed pull request.

The decryption is not all that complex, there is even a python version of it which I am using right now:
https://github.com/mweimerskirch/smarty_dsmr_proxy

(but I am also working a bit on a C++ version that would act at the same time also as an Modbus/TCP Server /  MQTT Publisher)

dennissiemensma, 2019-10-31:
Thanks for all the informatie. Do you have some more info about:
```
1-0:3.8.0(000120.721*kvarh)
1-0:4.8.0(000697.336*kvarh)
1-0:3.7.0(00.000*kvar)
1-0:4.7.0(00.000*kvar)
0-0:17.0.0(27.600*kVA)
```

I do not know what type of unit is it. The docs refer to it as Q+/-.

dennissiemensma, 2019-10-31:
I've looked into this now, together with the Belgium specs and I'll implement a country-setting, as you've proposed above. 
There are too many differences and caveats among the implementations, so I'll create some kind of gateway that will map the codes read to fields. At least now I have a few samples to embed in the tests so I can make sure it will work well for everyone.

JeanMiK, 2019-10-31:
The „var“ is the  <https://en.wikipedia.org/wiki/Volt-ampere_reactive> volt-ampere reactive , the unit used to indicate the ‘reactive power’ that can exist in an AC electric system, but which, contrary to the ‘active’ power (expressed in W) is not billed to the customer.

 

In then similar was, varh is then a unit for the ‘reactive energy’. If you have a look at the linked Wikipedia entry, you see where the letter comes from: customary, the apparent power is given the letter S, the active power the letter P and the re-active power the letter Q. 3.8.0 is the summation of 3.7.0 (and in the same way 4.8.0 the summation of 4.7.0).

 

As they are not really relevant to customers, I doubt they will really appear in the telegrams, and don’t really deserve to be stored.

 

For the ‘apparent power limit’ 17.0.0, the unit is the Volt-Ampere VA (or here kVA).

 

Some of the data are shown on the meter itself in ‘expert mode’

 

Jean-Marie

dennissiemensma, 2019-10-31:
I see, then I'll ignore that varh data as you suggest.

dennissiemensma, 2019-11-01:
The past days I've been working on integrating ndokter/dsmr_parser into the project. 

My legacy code for parsing telegrams was way too complex, fragile and problematic to extend. Even though dsmr_parser isnt perfect, it's way better and more dynamic. 

The main reason is [its dynamic definition of each protocal version](https://github.com/dennissiemensma/dsmr-reader/commit/80ce93e8613b417b5dd367ab1444fba5eb09fc04#diff-5ba69f998aaecb11b99f0be930790de6). This way I can add new dialects for Belgium and Luxembourg's meters and link them to some admin-setting in the interface.

I'll try to finish it the next days or week so you can have a try testing it at home.

braham2019, 2019-11-02:
great news, thanks !

dennissiemensma, 2019-11-02:
@JeanMiK I think it's ready. Can you check this test with your telegram to see if I've missed anything?

* https://github.com/dennissiemensma/dsmr-reader/commit/cb0ff9281142d51be60b51f302ab5dc98a6e0c35#diff-84ae7df03924650847f48d5a654109acR73

Most values are defaults to zero/none, but there are a few that need to be checked. I've mapped the single tariff to DSMR tariff 2, which translates to 'high tariff' in the interface.

dennissiemensma, 2019-11-02:
I found the Smarty docs quite confusing by the way. They write about DSMR v5 and even include a DSMR v5 telegram on the last page, but it's nothing like your telegram. You seem to run a custom version of v4.2, but I'll just have to asume that similar meters communicate the same format.

JeanMiK, 2019-11-02:
As I said, it seems to be kind of a mix, the DSMR v5 physical interface, but the DSMR v4 polling frequency, and their own list of standard / enhanced Orbis codes.

 

From what I understand, Luxembourgish smart-meters will always be using a sub-set of the Orbis codes from the Luxembourgish documentation, but only the values marked as ‘standard’ are guaranteed to be present in the telegram. The data in my telegram seems to correspond to this ‘mimimal’ sub-set, all other meters should deliver the same or a super-set-

 

I know for sure that other people *do* receive currently some of the ‘optional’ fields (e.g. some of the guys having written the decryption software), and I have been told by Luxmetering that they consider making more data available to everybody, but it is not their priority (for the moment, they are doing the physical roll-out of the meters in the houses all through Luxembourg, but the backbone is not yet ready everywhere, and they still have not connected my upgraded gas meter to the electricity meter either).

 

Jean-Marie

JeanMiK, 2019-11-02:
Seems to be working on my side (after installing PyCRC and updating the datalogger configuration).

 

A bit strange to put all the data into tariff 2 if there is only a single tariff, but ok, just had to do a few updates on the PostrgreSql database to  update my history.

 

In a few weeks, I will get a second smart meter, as I am going to get a PV installation. [In Luxembourg, the grid operator CREOS imposes in such cases a separate (also bidirectional) meter for the energy production (typically, everybody is injecting all the generated electricity into the grid, and then possibly buying it back from the grid)]. 

 

I will then be able to compare the telegrams for both meters and see if there are any differences (e.g. per phase tracking activated on the second meter [where it would be less useful]).

 

Ideally, I would like store the ‘delivered’ electricity from the first meter, and the ‘returned’ one from the second one, but I will check this out when I get the system.

 

*	 <https://github.com/dennissiemensma/dsmr-reader/commit/cb0ff9281142d51be60b51f302ab5dc98a6e0c35#diff-84ae7df03924650847f48d5a654109acR73> cb0ff92#diff-84ae7df03924650847f48d5a654109acR73

dennissiemensma, 2019-11-02:
Thanks for the verification. Should I choose a different tariff? Since there is no tariff 0.

dennissiemensma, 2019-11-02:
Also, in the frontend settings you can opt to merge the display of tariffs, so it should be indifferent anyway, right?

dennissiemensma, 2019-11-03:
> A bit strange to put all the data into tariff 2 if there is only a single tariff

I guess you are right, so I'll change it to 1 then. Thanks for the heads up!

dennissiemensma, 2019-11-03:
It will be included in the upcoming release. Most likely somewhere next week.

closed 2019-11-03 by dennissiemensma