dsmrreader/dsmr-reader #771: Dashboard: energiekost elektriciteit fout berekend?

dimitrizetzsche, 2019-10-31:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.9`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 2`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Het ziet er naar uit dat de energiekost voor elektriciteit op het dashboard fout wordt berekend:
[image]

Volgend energiecontract is actief — hier zijn de totalen wel OK:
[image]

Op het dashboard zou de prijs van mijn elektriciteit 4,15*0,7776€=3,23€ moeten zijn. De gas is OK: 11,38*0,02847=0,32€

dennissiemensma, 2019-10-31:
Bedankt voor je melding.

Ik zie dat je een enkel tarief gebruikt. Voor de zekerheid: Je hebt dezelfde prijs ingevuld voor hoog- als laagtarief he?

dennissiemensma, 2019-10-31:
Nevermind, ik zie nu pas je tweede screenshot. De meeste recente wijziging m.b.t. prijzen zijn afrondingen, maar dat zou hooguit een cent verschil moeten maken. Ik zal kijken of ik het kan reproduceren.

dennissiemensma, 2019-10-31:
Bij nader inzien, volgens mij klopt de prijs, maar wellicht hebben we een andere kijk op dezelfde gegevens? De ingevoerde prijs is 7 cent en niet 77 cent. Zit daar een afwijking in?

Ik kom ook uit op 32 cent:
[image]

Teruglevering van stroom wordt overigens momenteel niet verrekend in de prijs.

dimitrizetzsche, 2019-10-31:
Je hebt gelijk. Ik heb om een of andere reden (meerdere keren) verkeerd gerekend :-). Toevallig zijn bij vermenigvuldigingen hetzelfde :-).

No issue dus.
--

closed 2019-10-31 by dimitrizetzsche

dennissiemensma, 2019-10-31:
Kan gebeuren, bedankt voor je terugkoppeling.