dsmrreader/dsmr-reader #772: Sinds een maand draaien mijn statistieken en trends niet meer.

cyberjunky, 2019-11-02:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

Ik gebruik docker images en kan de log file dsmr_backend.log niet vinden?

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: v2.9
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Docker op Ubuntu
* DSMR-protocol slimme meter *(v4 of v5)*: v4 denk ik

Ik kom er achter dat mijn laatste statistieken van 4 october zijn.
nieuwe telegrammen komen binnen, maar de stats worden niet bijgewerkt.
ik heb de dsmrdb docker image al gerestart en daarna de dmsr docker.
hoe kan ik zien of er wel data gelogged is in afgelopen tijd, of dat alleen een process niet goed draait welke de stats berekend?

Thanks!

dennissiemensma, 2019-11-02:
Bedankt voor je melding. Ik kan je helaas niet inhoudelijk helpen bij Docker images, omdat ik niet weet waar de logfile opgeslagen wordt en hoe je er bij kan.

Wat wel opvallend is, is dat iemand anders in #764 last heeft van exact hetzelfde probleem, met dezelfde versie, ook in Docker en ook rond 4 oktober.

Wil je eens kijken of je wat resultaten hebt uit de stappen in #764?

dennissiemensma, 2019-11-03:
Ik sluit deze voor nu, gezien het andere ticket over hetzelfde gaat. Ik raad je aan die te volgen. Ik heb een mogelijke oorzaak gevonden.

closed 2019-11-03 by dennissiemensma