dsmrreader/dsmr-reader #774: CPU gebruik wat aan de hoge kant

jorkzijlstra, 2019-11-03:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.9`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Rock PI 4B`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Na het introduceren van het monitoring van de host machine viel het ons op dat de dsmr software redelijk wat cpu gebruikt: https://github.com/xirixiz/dsmr-reader-docker/issues/81

[image]

[image]

Zowel de `python3 -u /dsmr/manage.py dsmr_datalogger` als de ` python3 -u /dsmr/manage.py dsmr_backend` lijken dit te doen.

jorkzijlstra, 2019-11-03:
Ik zag al https://github.com/dennissiemensma/dsmr-reader/issues/600 maar ik vind dat meer symptoom bestrijden ipv een prolbeem oplossing.

dennissiemensma, 2019-11-03:
Bedankt voor je melding. Het is wat lastig debuggen wat het exact veroorzaakt. 

Ik heb eerder vanavond nog wel een mogelijk issue in de dsmr_backend gefixt, wanneer iemand DSMR-v5 heeft in combinatie met groepering per minuut. Het systeem ging elke (halve) seconde controleren of de meting al gegroepeerd kon worden, per meting. Dus uiteindelijk deed die dat aan het einde van elke minuut bijna 60x per (halve) seconde. Ik verwacht dat dat ook al wat oplost.

Wat betreft je opmerking met #600. Ik kan me daar deels in vinden en ik snap je punt, maar ik ben het niet helemaal met je eens. De architectuur van het systeem is dat (met name dsmr_backend) zo gemaakt is dat die zovaak mogelijk aangeroepen wordt. Dit kan via een cron of langlopend proces. 
Ik heb voor dat laatste gekozen, met als groot nadeel dat het oneindig veel CPU kan slurpen, mits er geen limieten op zitten. Daarom zit er een korte sleep op van 0.5 seconde, maar die is in veel gevallen al te kort. Voor DSMR-v4 of bij gelimiteerde doorstroom van telegrammen, is 5 seconden sleep ook prima, en dat scheelt 10x de load. Daarnaast vraag ik ook het OS om hulp, door `nice` te gebruiken. Ik zie dat dus niet helemaal als symptoon bestrijding, maar meer als bij-effect van mijn keuze qua architectuur.

Overigens [ben ik al wel bezig met een alternatief](https://github.com/dennissiemensma/dsmr-reader/blob/v2.9.0/dsmr_backend/models/schedule.py#L17). In plaats van de backend zovaak mogelijk aanroepen en weer te sleepen, weet ik voor de meeste dingen eigenlijk al wel wanneer ik ze weer wil doen (groeperen, backup maken, data opschonen, etc) en dus kan ik ze plannen. Een hope code (en queries) wordt dan pas uitgevoerd wanneer er echt wat te doen is, ipv telkens ter plekken constateren dat er (nog) niets te doen is. Ik verwacht daar een grote performance-winst door, maar het is veel werk om alle bestaande code en checks aan te passen.

Wat betreft de datalogger, die is het lastigste te debuggen. Ik heb zelf geen v5-meter en ik vraag me ook af waar de CPU in zit. Mogelijk is dat puur het constant pollen van de seriele poort. Je kunt wellicht zelf wat experimenteren door [in de datalogger loop](https://github.com/dennissiemensma/dsmr-reader/blob/v2.9.0/dsmr_datalogger/services.py#L73) zelf ergens een sleep van 0.1 seconde te zetten, alleen vraag ik me af of dat wat uitmaakt, gezien er [tussen de metingen door ook al een sleep zit](https://github.com/dennissiemensma/dsmr-reader/blob/v2.9.0/dsmr_datalogger/management/commands/dsmr_datalogger.py#L18).

Vandaar ook de mogelijkheden in #600. Als iemand geen reden heeft om korte sleeps te gebruiken, kan dat ook al performance schelen, maar ik zet ze standaard heel laag.

Heel verhaal, hopelijk geeft het je wat meer inzicht in hoe dingen werken en waarom ze wel of niet traag zijn.

dennissiemensma, 2019-11-03:
Ter vergelijking, een rPi 4 op een DSMR-4 meter. Meestal is het alleen Postgresql die af en toe een paar % CPU pakt, maar voor de rest niets.

[image]

Ik denk dat er een ontzettend groot verschil zit in de hoeveelheid telegrammen per seconde en het effect op de load. Ik gebruik overigens de standaard (lage) sleep-waarden.

jorkzijlstra, 2019-11-04:
Hoi Dennis,

Duidelijk verhaal.

Ik weet ook dat het bijzonder lastig is om dit te debuggen, zeker als je zelf niet de mogelijkheden hebt. Wel bijzonder dat het met een dsmr v4 geen issue lijkt te zijn.

De postgresDb lijkt bij mij geheel stabiel:
[image]

Ik zal nog eens kijken of ik misschien wat kan vinden. Het leek wel alsof beide processen telkens tegelijk de cpu pakte of beide juist niet, maar geen idee of dit met elkaar in relatie staat.

Zelf ben ik wel fan van applicatie metrics en kan misschien eens kijken of het mogelijk is om wat Prometheus metrics te implementeren waar je dan een Grafana dashboard op kan bouwen. 
Het grootste voordeel is dan dat je over de tijd heen inzichtelijk krijgt of je applicatie minder gaat performen of juist beter.

dennissiemensma, 2019-11-04:
De volgende release later deze week (wellicht morgen) kan dus al wat verbeteringen hebben. Ik hoor dan graag of je daarmee al een verbetering ziet in de performance.

dennissiemensma, 2019-11-08:
Naar verwachting is morgen de release van v2.11, waar al een aantal processen zijn omgezet naar de 'nieuwere' manier. Ik hoor dan graag of dat de performance verbeterd heeft.

jorkzijlstra, 2019-11-08:
Ik moest wachten op de release van de docker container en heb net 15 min geleden v2.10 gedeployed.

[image]

[image]

Ook deze release lijkt dus al een klein verschil te maken.

Ik laat het weten waneer ik v2.11 heb gereleased.

dennissiemensma, 2019-11-09:
Uitgebracht in [v2.11](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.11.0)

jorkzijlstra, 2019-11-11:
Ik heb vandaag om 20:00 de v2.11 gedeployed.

[image]

Zoals je van zien zijn er 2 drops. De 1e grote drop is v2.10 en 2e kleinere van vandaag van versie 2.11. Het is je gelukt om van ongeveer 30% naar 23% cpu te gaan. Daarnaast is het postgres database gebruik ook van 11.7% naar 8.5% gegaan.

Goed bezig, dit is een zeer grote verbetering.

dennissiemensma, 2019-11-11:
Bedankt voor het delen en goed om te horen dat het ook daadwerkelijk iets verbetert.

Voor de volgende release probeer ik weer wat om te zetten naar de nieuwe manier. Er zijn nu nog 4 dingen die op de oude manier gaan:
- Backups check
- Dropbox sync
- Notificaties
- Groeperen van metingen

Met name de laatste is denk ik nog vrij intensief, dus die zal ik dan tzt als eerste doen.

jorkzijlstra, 2020-01-07:
Ik ben vannacht van 2.11.3 naar 2.13.0 gegaan en zie dat het standaard cpu gebruik weer ietsjes omhoog is gegaan voor de applicatie zelf. De database is hetzelfde gebleven.

[image]

dennissiemensma, 2020-01-07:
Bedankt voor je update. Ik heb dit voor nu even op een laag pitje staan, omdat verdere rework niet zo eenvoudig bleek. Ik ga er in DSMR-reader v3.x weer mee verder.

Je kunt ook nog overwegen om de sleep van de backend wat hoger te zetten voor een lagere load als je er echt last van hebt.

jorkzijlstra, 2020-01-07:
Hoi Dennis,

ik dacht doe alleen een updateje met deze nieuwe release. Volgens de release notes had ik ook niet verwacht dat je er aan gewerkt had, maar op zich mooi dat je er eventueel later nog naar wil kijken.

Al lees ik net de release notes en misschien dat de email backup de verhoging in cpu load heeft veroorzaakt. Wanneer er weer een nieuwe docker versie uit is zal ik weer een updateje doen. Al zal deze nieuwe docker versie wat werk vereisen gezien er wat settings aangepast zijn. Ik verwacht niet dat deze komende week uitkomt.

In ieder geval bedankt voor het verdere werk

jorkzijlstra, 2020-01-19:
Updateje voor V3.0.0

De 80% cpu piek is een fout gegane deployment waarbij de usb p1 poort niet gebind kon worden en het process dus geen data kon uitlezen. Al ken ik de software natuurlijk niet, maar heb het vermoeden dat er iets van een retry loop in zit die geen rekening houd met de ingestelde interval / geen backoff strategie heeft. Bij een redeployment was dit verholpen en is dus iets voor mij zelf om niet de deur uit te rennen zonder te checken dat de upgrade ook gelukt ie. :)

Het cpu gebruik ten op zichte van v2.13.0 is significant omhoog gegaan.
Bij het even snel bekijken lijkt het in `python3 -u /dsmr/manage.py dsmr_datalogger` te zitten.

Het inkomend netwerk verkeer ook ook van 12Kb/s naar 20Kb/sec gegaan. Al is dit niet veel maar had dit niet verwacht. Ik dacht dat er naast een update check niet zoveel vanaf het inerwebs gechecked werd.

v2.13.0 -> v3.0.0

| Type         | Values           | %  |
| ------------- |:-----------------:| ----:|
| CPU      | 25% -> 35% | x 1.4 |
| Memory      | 238Mb -> 218Mb | x 0.9 |
| Incoming network      | 12 kb/s -> 20Kb/sec | x 1.6 |

[image]

dennissiemensma, 2020-01-19:
Bedankt voor je update. Het kan zijn dat de datalogger inderdaad aggresief is qua retry. Ik kan wel een keertje kijken of ik die async kan maken, zodat die niet hangt op input. 
Wellicht kun je de datalogger een keertje kort uitzetten om te zien hoeveel CPU die doorgaans pakt.

Wat betreft netwerking, er is volgens mij niets toegevoegd en alles is uit te zetten. Uit mn hoofd:
- Update check
- PVoutput export
- Mindergas export
- Dropbox
- Pushover notificaties
- Backup per e-mail
- MQTT
- Handmatige plugins die telegrammen doorsturen naar andere installatie

jorkzijlstra, 2020-01-19:
Hierbij de datalogger eventjes uitgezet:

De hogere pieken zin de herstarts.

[image]

Wat betreft settings, ik heb deze nog niet aangepast tussen al deze versies

dennissiemensma, 2020-01-20:
Bedankt voor je test. Ik had niet verwacht dat de datalogger zo'n groot deel van de CPU in beslag nam! Zoals gezegd zal ik tzt kijken of ik een asynchrone variant kan maken, die niet telkens wacht op seriele input.

jorkzijlstra, 2020-02-06:
Weer een updateje voor V3.2. 

Type | Values | %
-- | -- | --
CPU | 32% -> 35% | x 1.09
Memory | 225Mb -> 221Mb | x 0.99
Incoming network | 22 kb/s -> 16Kb/sec | x 0.72

[image]
[image]

dennissiemensma, 2020-02-07:
Bedankt voor de update!

dennissiemensma, 2020-02-10:
Ik denk dat ik er nu ook achter ben waarom de datalogger zo zwaar is. Voor elke uitlezing wordt de seriele poort opnieuw geopend. Dit had ik gefixt in de remote datalogger, maar nog niet in de ingebouwde.

Via #847 ga ik ervoor zorgen dat ze beiden dezelfde code gebruiken. Mogelijk scheelt dat ook al aanzienlijk CPU-gebruik.

jorkzijlstra, 2020-02-14:
Hoi Dennis, gisteren heb ik tijd gehad om release 3.3.0 te deploymen (rond 19:50) en hier dus weer een updateje.

[image]

Zoals je kan zien maak deze release niet oveel uit in load.

dennissiemensma, 2020-02-15:
Bedankt voor je update. Dat is jammer want ik had een grote verbetering verwacht. Ik neem aan dat je de ingebouwde datalogger gebruikt?

GieltjE, 2020-02-15:
3.3.0 heeft wel een aardige positieve impact gehad op de load :)

dennissiemensma, 2020-02-16:
@GieltjE bedankt voor de update. Draai je native of via Docker? En heb je een schatting van hoeveel de load beter is geworden?

jorkzijlstra, 2020-02-16:
Nog even om terug te komen op je vraag. Ik gebruik idd de ingebouwde datalogger.

GieltjE, 2020-02-16:
@dennissiemensma draai hem native op Gentoo, hangt toch achter een reverse apache proxy met digest auth.
Load is ongeveer tussen de .15 en .30 gezakt, wat best aanzienlijk is.

jorkzijlstra, 2020-02-23:
v.3.4.0 updateje

Het CPU gebruik fluctueerd is sinds v3.4.0 een heel stuk meer dus i.p.v. per minuut een punt in de grafiek heb ik deze nu op per 30 minuten waardoor de average beter te zien is.

Gisteren net voor lunch geupdate en je ziet sinds vannacht een hele kleine verbetering van +- 34% naar 33%. De 2 pieken heb ik niet kunnen verklaren. Ik dacht dat de 1e iets met de v3.4.0 update zelf te maken had maar voor de 2e heb ik geen verklaring. Al is deze nu ook weer weg. 

[image]

dennissiemensma, 2020-02-23:
Bedankt voor de update. Zou je de datalogger weer eens een minuutje of wat willen uitzetten om te kijken of die verhoudingsgewijs nog steeds het meerendeel in beslag neemt qua CPU usage?

jorkzijlstra, 2020-02-26:
Hoi Dennis,

ik heb eindelijk even tijd gehad. Geen idee waarom de memory zo omhoog schoot.
[image]

dennissiemensma, 2020-02-26:
Dank voor het uitzoeken! In dat geval weet ik niet zo goed hoe ik het verder kan verlagen qua usage. De applicatie zelf verbruikt dus amper wat, puur de datalogger.

dennissiemensma, 2020-02-27:
In de volgende versie (waarschijnlijk morgen) is de datalogger weer deels zoals in v3.2. Deze heropent de verbinding weer na elke uitlezing. Ik verwacht verder geen grote verschillen.

dajappie, 2020-03-09:
Het CPU-gebruik is bij mij volgens RPi-monitor ook in stappen eind september en daarna half december flink omhoog geschoten, wat samenvalt met nieuwe versies van Postgres/DSMRreader. Draai nu DSMRReader 3.6 met Postgres 9.6.17 op een Rpi 3b. Omdat volgens top het vnl. Postgres is die veel gebruikt eens pg_activity geinstalleerd om te zien wat binnen Postgres de belasting veroorzaakt en dat is m.n. onderstaande query. Deze lijkt voor elke meeting uitgevoerd te worden. Misschien dat deze met wat herschrijven/indexering/caching te verbeteren is?

`SELECT DATE_TRUNC('hour', "dsmr_datalogger_dsmrreading"."timestamp" AT TIME ZONE 'UTC') AS "item_hour" FROM "dsmr_datalogger_dsmrreading" WHERE ("dsmr_datalogger_dsmrreading"."processed" = true AND "dsmr_datalogger_dsmrreading"."timestamp" < '2019-09-23T11:14:07.131791+00:00'::timestamptz) GROUP BY DATE_TRUNC('hour', "dsmr_datalogger_dsmrreading"."timestamp" AT TIME ZONE 'UTC') HAVING COUNT("dsmr_datalogger_dsmrreading"."id") > 2 ORDER BY "item_hour" ASC LIMIT 24`

dennissiemensma, 2020-03-09:
Bedankt voor je melding. Die query wordt gebruikt voor het berekenen van welke gegevens wegkunnen met retentie. Momenteel wordt die nog bij elke run van de backend uitgevoerd, maar dat kan wel wat gefaseerder (wanneer de retentie geen achterstand meer heeft), zoals andere taken die ik eerder al had omgezet. Ik zal deze ook meenemen. Is een relatief kleine moeite.

dajappie, 2020-03-09:
> Bedankt voor je melding. Die query wordt gebruikt voor het berekenen van welke gegevens wegkunnen met retentie. Momenteel wordt die nog bij elke run van de backend uitgevoerd, maar dat kan wel wat [...]

Top met de volgende release even kijken of het scheelt!

jorkzijlstra, 2020-03-10:
Geen idee op deze query wordt uitgevoerd op mijn systeem, gezien ik de retentie op None heb staan. Inmiddels staan er dus 18 miljoen records in de tabel.

De bovenstaan query duurt inmiddels meer dan 30 seconden en in mijn geval worden altijd dezelfde result opgehaald.

dennissiemensma, 2020-03-10:
Die query geeft vanzelfsprekend hetzelfde resultaat terug omdat er een harde datumtijd in staat. In de praktijk verschuift die per dag. Als je retentie uit hebt staan, wordt de query niet uitgevoerd.

dennissiemensma, 2020-03-19:
Verbetering uitgebracht in [v3.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.7.0)

Phyxion, 2020-03-24:
Het CPU gebruik zat hier constant rond de 10% in totaal (python + postgres). Maar de sleep tijden wat aanpassen heeft het aardig gereduceerd, ik heb alle tijden maar * 5 gedaan en kan niet zeggen dat ik enig verschil merk in het gebruik. Misschien een idee om de standaard waardes hiervan wat aan te passen? 0.5s lijkt mij sowieso wel erg weinig, een slimme meter geeft maar één keer per 10 seconden een telegram.

dennissiemensma, 2020-03-24:
@Phyxion er zijn momenteel twee varianten in Nederland:
- Meters met DSMR v4 die elke 10 seconde een telegram sturen
- Meters met DSMR v5 die elke seconde een telegram sturen (en eveneens de gasstand meerdere keren per uur bijwerken)

Om die reden laat ik de keuze voor de hoogte van de sleep(s) over aan de gebruikers. Voel je vrij om ze zo hoog te zetten als je zelf wilt. Het totale dagverbruik dat gemeten wordt zal er inderdaad amper door afwijken.

Phyxion, 2020-03-24:
Klopt, ik heb zelf ook een v5 maar ik gebruik alleen de laatste telegram uit 10 seconden. Het verschil in waardes lijkt mij niet echt meetbaar omdat de meter altijd een totaal bijhoudt, hooguit heb je op de volle minuten een miniscule afwijking in de grafieken zitten over wat er tussen dat interval in zit. Maar als je zo'n extreme nauwkeurigheid nodig hebt dan kan je beter voor een andere oplossing dan de P1 poort gaan denk ik 😉

jorkzijlstra, 2020-05-05:
Hoi Dennis,

Door mijn werk ben ik er achter gekomen dat ik denk ik de verkeede metrics heb gebruikt om het cpu gebruik te meten. 

Het lijkt dat ik het CPU gebruik "binnen de docker container" heb gemeten en niet het cpu gebruik van de "host machine"

Die van de host machine ziet er uit zoals andere eerder al beschreven, net boven de 10%.
[image]

terwijl die binnen de container rond de 35% gebruitk van max ???
[image]

Dit lijkt dus veel minder ene issue te zijn dan dat ik jouw had voorgesteld.

closed 2020-05-05 by jorkzijlstra

dennissiemensma, 2020-05-05:
Bedankt voor de opheldering. Desnietemin heb ik wel een deel kunnen verbeteren qua performance, dus het is zeker nuttig geweest!

Tralapo, 2020-10-06:
Ik haak even aan op deze issue als dat ok is. Sinds april gebruik DSMR Reader op mijn Raspberry Pi 3B.
Sinds een week of 3 ben ik geswitched naar een Docker installatie (die van xirixiz) en draait er naast DSMR en PiHole, ook Home Assistant en de Unifi Controller op de Pi.

Sinds die switch heb ik het probleem dat mijn Pi een aantal keer per dag volledig lijkt vast te lopen. Zojuist bleek na een reboot dat DSMR niet was opgestart, maar zag ik tegelijkertijd dat de load en temperatuur flink waren gedaald.

Daarom een testje gedaan, zie onderstaande grafiekjes. De bovenste is de 15m load (de 1m en 5m log ik niet, sorry), de onderste de processor temperatuur.
- Ergens tussen 12:20-12:30 reboot, waarna DSMR niet bleek te runnen
- Rond 12:40 DSMR weer opgestart, rond 12:55 weer uitgezet
- Rond 13:05 DSMR weer gestart, tot einde grafiek.
Het is wel duidelijk te zien dat DSMR een flinke invloed heeft, zoals eerder in dit issue ook besproken.
Schaal is hier tussen de 0.5 en 1.0 trouwens.

[image]

Máár, nu komt het: die paar keer per dag dat de Pi lijkt te crashen, loopt de load soms wel op tot nabij de 50.0, zie hier de 15m load grafiek van de afgelopen paar dagen:

[image]

Zou dit DSMR kunnen zijn? En zo ja, zijn er manieren waarop ik kan controleren of monitoren wat dit veroorzaakt?

GieltjE, 2020-10-06:
Weet niet of je dit gecombineerd soepel op iets lichts als een rpi kan draaien, hoe staat het met ram usage en swap?

Tralapo, 2020-10-06:
Op zich wel. Het duurt een minuut of 2-3 voor alles up is na een reboot, maar qua performance merk je er weinig van.
Ram gaat wel vaak richting 700 MB used, swap is rond de 50 mb.

Onderstaand is het effect van het backend proces sleep verhogen naar 5 seconden en de datalogger proces sleep verhogen naar 2 seconden (vanaf 17:40). Zo kijk ik 'm eerst maar even aan denk ik.

[image]

dennissiemensma, 2020-10-06:
@Tralapo de eerste stap is denk ik om `atop` installeren, zodat je na een crash kan terugkijken naar de standen van `top`, vlak voor de crash.

Voor de rest is het zeer aannemelijk dat DSMR de oorzaak is, maar dat hangt nauw samen met de hoeveelheid data die je opslaat (des te langer queries duren, des te meer load er kan ontstaan) en de sleep time van de backend/datalogger (vooral bij DSMR-v5-meters), die de hoeveelheid/frequentie van queries bepalen. 

Voor meer dingen die je kunt checken, zie alle comments in #1139.

jorkzijlstra, 2020-10-06:
Tegenwoordig zal een herstart van de docker container eventueel een nieuwe versie installeren. Hierdoor heb je dus 2 a 3 minuten downtime bij een herstart.

Kwa CPU gebruik is het de laatste versies goed naar beneden gegaan.
[image]

Hoe draai je je DB? Is dit not de oude installatie of een nieuwe? 
Ik had in het begin het issue dat er een aantal indexen niet goed waren aangemaakt. Zie #688 
Dit issue kon ik dan ook terugzien in het `atop`

Gezien je toch docker gebruikt. Hierbij ook en docker-compose die ik zelf draai om een complete monitoring stack te draaien op 
 je pi. 
[monitoring.zip](https://github.com/dennissiemensma/dsmr-reader/files/5336262/monitoring.zip)

Op poort 3000 wordt er een grafana gehost met al een aantal dashboards. Zelf kijk ik eigenlijk alleen maar naar `Docker and system monitoring`

Tralapo, 2020-10-20:
Welke stap precies welk effect had weet ik niet, maar:
- Heb m'n 8GB SD kaart vervangen door een 16GB kaart
- Heb m'n adapter vervangen door een 3A exemplaar
- Heb backend sleep op 10 seconden gezet
- Heb datalogger sleep op 2 seconden gezet

Sindsdien is m'n load vrij stabiel tussen de 0.2 - 0.4. Hij komt eigenlijk bij normaal gebruik niet meer boven de 1.0, laat staan de 50.0 van eerder.

jorkzijlstra, 2020-10-20:
Het zal een combi zijn van alles. Met een nieuwere snellere SD-kaart performed alles beter terwijl je met de sleeps minder aan het doen bent.

Ik draai bijvoorbeeld op een Rock Pi 4B met een M.2 SSD.
- Backend sleep 1 seconde
- Datalogger sleep op default
- Retentie op oneindig

en alles is gewoon responsive maar wel met een cpu load van ~35%. Voor mij is dit niet erg gezien er niks anders op het systeem draait.