dsmrreader/dsmr-reader #776: Loopt vast

peterjohan52, 2019-11-07:
Nov. 6, 2019: Failed to read Buienradar API: HTTPSConnectionPool(host='data.buienradar.nl', port=443): Max retries exceeded with url: /2.0/feed/json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x73807f30>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

en dan tientallen keren achter elkaar.

DSMR reader versie 2.10
Raspberry pi 3

dennissiemensma, 2019-11-07:
Bedankt voor je melding. Ik zal voor de volgende release fixen dat er minder dezelfde fouten getoond worden.

Mochten het er nu honderden bij je zijn, dan kan ik wel een query voor je maken waarbij je ze in 1x als gelezen markeert.

dennissiemensma, 2019-11-07:
Ik heb er een fix voor gemaakt die meegaat met de volgende release. Waarschijnlijk komende weekend of volgende week.

closed 2019-11-07 by dennissiemensma

dennissiemensma, 2019-11-09:
Uitgebracht in [v2.11](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.11.0)