dsmrreader/dsmr-reader #778: Geen Gas

peterjohan52, 2019-11-08:
Mijn gasmeter staat op kan 2. De eerste gasmeter wilde niet koppelentijdens de gas en elektrometer vervanging. Toen heeft de monteur een tweede geprobeerd en die werkte. Maar nu is het meter 2. Hoe krijg ik die in DSMR.

1-3:0.2.8(50)
0-0:1.0.0(191108145519W)
0-0:96.1.1(4530303433303037333238373631333139)
1-0:1.8.1(001043.936*kWh)
1-0:1.8.2(000870.706*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.189*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00008)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(190115011853W)(0000000389*s)
1-0:32.32.0(00005)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(235.4*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.190*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0()
0-1:24.2.1(700101010000W)(00000000)
0-2:24.1.0(003)
0-2:96.1.0(4730303339303031383433313135303138)
0-2:24.2.1(191108145505W)(00241.773*m3)
!6DEB
/ISK5\2M550E-1012

dennissiemensma, 2019-11-08:
Bedankt voor je melding. Ik moet als eerste weten welke versie van DSMR-reader je gebruikt.

peterjohan52, 2019-11-08:
Laatste versie 2.10.

dennissiemensma, 2019-11-08:
Ik zie al wat het is, maar dat is een interessante uitdaging. 

Je slimme meter communiceert alsof er twee gasmeters aangesloten zijn.
```
0-1:24.2.1(700101010000W)(00000000)
0-2:24.2.1(191108145505W)(00241.773*m3)
```

De ene geeft een ongeldige datumtijd en meterstand terug. De andere is goed maar wordt niet gebruikt.

Ik zal kijken of ik daar iets op kan verzinnen.

peterjohan52, 2019-11-08:
OK, succes.

closed 2019-11-08 by peterjohan52

dennissiemensma, 2019-11-08:
Ik heb er een workaround voor kunnen maken, die ook lijkt te werken voor alle andere situaties. Mocht echter blijken dat het in de praktijk niet overal werkt, dan draai ik het weer terug.

Toevallig stond voor dit weekend een release gepland, dus het gaat daarin me.

dennissiemensma, 2019-11-09:
Uitgebracht in [v2.11](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.11.0)

peterjohan52, 2019-11-09:
Het werkt weer prima. Bedankt.

dennissiemensma, 2023-01-23:
Van weekend een gerelateerd issue gerapporteerd: #1794
Alleen gaat dat over twee "goede" gasmeters (al staat er eentje op nul), dus de fix van het huidige issue krijgt een iets andere vorm.