dsmrreader/dsmr-reader #779: 502 Bad Gateway

hjkoster400d, 2019-11-09:
Sinds een tijdje geeft de Raspberry Pi 3 een 502 error.
Afgelopen week heb ik de update gedaan naar de laatste versie (probleem was daarvoor ook al wel).

Raspberry leek wel te kampen met een power issue, maar dat is nu met een nieuwe adapter opgelost.

./manage.py levert het onderstaande resultaat op:
[image]

Er staat wel dat er email settings missen, maar die hebben er nooit ingestaan.... Enig idee wat het zou kunnen zijn?

Alvast bedankt.

M.v.g.
Rik Koster

dennissiemensma, 2019-11-09:
Bedankt voor je melding. Het is een stuk code genaamd EmailSettings die er mist, niet de instellingen zelf, want die zijn optioneel.

Vermoedelijk is er iets met de installatie, want het pad klopt verder wel. Kun je eens dit proberen:
```
sudo su - dsmr
git reset --hard
./post-deploy.sh
```

dennissiemensma, 2019-11-09:
En als dat niet werkt, kun je dit nog proberen, de interne cache van Python-bestanden weggooien:

```
sudo su - dsmr
find . -type f -iname '*.pyc' -delete
./reload.sh
```

hjkoster400d, 2019-11-09:
Heey Dennis,

Allereerst; Heel erg bedankt voor je snelle hulp!

Beide oplossingen heb ik geprobeerd (weliswaar is de omgekeerde volgorde, dus eerst je laatste oplossing en daarna de eerste).

Nu is de 502 weg, maar geeft de webpagina een Internal Server Error.

Hieronder de laatste regels van de console na het uitvoeren van de ./reload.sh

[image]

Alvast bedankt (nogmaals).
M.v.g.
Rik

hjkoster400d, 2019-11-09:
Aanvullend de dsmr.log file geeft deze melding, wellicht dat dat er iets mee te maken heeft?

[image]

dennissiemensma, 2019-11-09:
Ik denk dat er meerdere dingen aan de hand zijn. Je kunt het beste nog eens updaten en dan alles herstarten.

```
sudo su - dsmr
./deploy.sh
CTRL+D

sudo supervisorctl restart all
```

Dat lost het niet van de telegrammen op, maar het is een begin. Die logs zijn ook van 2 november, dus dat is al een tijdje terug.

hjkoster400d, 2019-11-09:
Top! 
Ik ga het even proberen en laat het z.s.m. weten.

Bedankt voor de geweldig goede service trouwens.

hjkoster400d, 2019-11-09:
Zojuist even deploy opnieuw gedaan en alles gerestart. De processen leken allemaal netjes op te starten maar een aantal blijven herstarten.... Webserver draait wel op poort 81, dat zal het probleem toch niet zijn lijkt mij?

[image]

[image]

dennissiemensma, 2019-11-09:
Er mist software, maar een deploy zou dat automatisch moeten installeren. Je kunt het handmatig proberen:

```
sudo su - dsmr
pip3 install -r dsmrreader/provisioning/requirements/base.txt
CTRL+D

sudo supervisorctl restart all
```

hjkoster400d, 2019-11-09:
Alle software is 'satisfied' zegt ie, dus alles lijkt geïnstalleerd te zijn. Op de zelfde Raspberry staat ook de homeassistant installatie, kan dat elkaar wellicht in de weg zitten?

dennissiemensma, 2019-11-09:
Dat zou niet uit mogen maken. Wat krijg je hier te zien:

```
sudo su - dsmr
pip freeze | grep PyCRC
```

hjkoster400d, 2019-11-09:
[image]

dennissiemensma, 2019-11-09:
En krijg je hier een fout op?

```
sudo su - dsmr
python3

from PyCRC.CRC16 import CRC16
```

Bijvoorbeeld `ImportError: cannot import name 'CRC16'`.

dennissiemensma, 2019-11-09:
En hier krijg je dus sowieso een fout op?
```
sudo su - dsmr
./manage.py dsmr_frontend_clear_cache
```

hjkoster400d, 2019-11-09:
Klopt inderdaad, beiden geven de CRC16 import error.

hjkoster400d, 2019-11-09:
Kan dit een oplossingsrichting zijn?

https://www.google.com/url?sa=t&source=web&rct=j&url=https://stackoverflow.com/a/37228009&ved=2ahUKEwiZ0tG8-d3lAhUSa1AKHWptAUcQ4-4CMAB6BAgFEAg&usg=AOvVaw2mEJEFqqwL0-t2GjN_rfwH&cshid=1573330757442

dennissiemensma, 2019-11-10:
In principe gaat alles via Python3 en pip3, dus ik verwacht daar geen issues. Ik heb geen idee wat de oorzaak is, want aan de ene kant geeft pip aan dat de package geinstalleerd is en aan de andere kant geeft Python aan dat die de package niet kent. 

Wat je nog kunt doen is de virtualenv met alle packages verwijderen en opnieuw installeren.
```
sudo supervisorctl stop all

sudo su - dsmr
deactivate
rm -rf /home/dsmr/.virtualenvs
mkdir /home/dsmr/.virtualenvs
virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3
pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

CTRL+D
sudo su - dsmr
# Als het goed is kreeg je geen fout bij het vorige command.

# Kijk of je nog steeds fouten krijgt:
./manage.py dsmr_frontend_clear_cache

# Zo nee, alles starten
CTRL+D

sudo supervisorctl stop all
```

hjkoster400d, 2019-11-12:
Heb de stappen uitgevoerd, maar hij geeft nu de volgende melding

[image]

Enig idee?

jeroenpeters1986, 2019-11-12:
Heb je het volledig uitgevoerd zoals Dennis het aangaf? 
Je zou verwachten dat Django goed geinstalleerd moet zijn, heb je nog meldingen gekregen toen je de `pip3 install -r ...` commando's uitvoerde?

Zo niet, wil je dan eens kijken wat `pip3 freeze` voor je aangeeft?

hjkoster400d, 2019-11-12:
Ja heb de stappen exact opgevolgd, en zojuist voor de zekerheid nogmaals gedaan maar krijg dezelfde foutmelding. 
De pip -r commando's geven aan dat de packages uit de cache worden gebruikt, verder geen bijzondere meldingen. 

Het pip3 freeze command geeft deze melding:
[image]

Nogmaals erg bedankt voor de hulp die jullie bieden, dat waardeer ik enorm!

jeroenpeters1986, 2019-11-12:
Ik weet niet of ik dan wat kan toevoegen nog. Als je geen foutmeldingen krijgt en alles geinstalleerd lijkt te worden, dan is het heel erg raar dat je met `pip3 freeze` geen packages te zien krijgt.

Toch nog voor de zekerheid, je hebt er geen `sudo` voor de commando's met `pip3 install -r`, en je weet ook zeker dat je `pip3` gedaan hebt ipv `pip`?

dennissiemensma, 2019-11-12:
Ik zie dat ik nog 1 command was vergeten. Na het aanmaken van de virtualenv moet die nog geactiveerd worden. Kun je dit proberen:

```
sudo su - dsmr
source /home/dsmr/.virtualenvs/dsmrreader/bin/activate

pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

# Controleer nu of pip3 freeze meer teruggeeft.
pip3 freeze

CTRL+D
sudo su - dsmr
# Als het goed is kreeg je geen fout bij het vorige command.

# Kijk of je nog steeds fouten krijgt:
./manage.py dsmr_frontend_clear_cache

# Zo nee, alles starten
CTRL+D

sudo supervisorctl start all
```

hjkoster400d, 2019-11-12:
Helemaal top! De activatie was inderdaad nodig. freeze gaf netjes een hele lijst terug en clear_cache gaf geen fouten meer. Reader loopt weer! 

Heel erg bedankt Dennis en Jeroen!

hjkoster400d, 2019-11-13:
Dag,

Wat mij opviel was dat na deze actie de grafieken op de hoofdpagina niet meer werken en de datum verkeerd staat. 
Daarnaast geeft de Status pagina een Server Error:

[image]

Actuele verbruik e.d. werken prima. Archief lijkt ook te werken, behalve dat deze denkt dat het 7 nonvember is...

hjkoster400d, 2019-11-13:
Heb zojuist ./manage gestart en heeft 2.11.1 geïnstalleerd. Nadien werkte alles weer naar behoren.
Hij heeft nog wel 55000+ telegrammen die niet verwerkt zijn.... Dat verwerkingsprocess heeft natuurlijk een tijdje uit gestaan. Worden die met terugwerkende kracht nog verwerkt trouwens?

Nogmaals bedankt!

dennissiemensma, 2019-11-13:
Je kunt het beste de meest recente versie installeren met `./deploy.sh`. Die voert eventuele database migraties ook uit.

Als alle processen verder draaien, haalt die vanzelf de achterstand in.

closed 2019-11-13 by dennissiemensma

hjkoster400d, 2019-11-13:
Ik bedoelde inderdaad ./deploy. Draait weer als een zonnetje hoor! Achterstand is intussen weggewerkt! Bedankt voor je hulp