dsmrreader/dsmr-reader #780: backup direct comprimeren

maarten-k, 2019-11-09:
Het viel me op de dat de backup eerst naar een normaal sql bestand word geschreven om vervolgens te worden gecomprimeerd. Het zou een stuk effectiever zijn om dit direct te doen zodat er minder I/O is.

Iets in de trend van:

`
with gzip.open(backup_file + ".gz", 'wb') as bu_file, subprocess.Popen(
        (command, env={'PGPASSWORD': settings.DATABASES['default']['PASSWORD']}, stdout=subprocess.PIPE) as process:
    shutil.copyfileobj(process.stdout, bu_file)`

Is dit een goed idee?

dennissiemensma, 2019-11-10:
Bedankt voor je suggestie. Dat lijkt me prima.

In een eerdere versie was het mogelijk om zelf te kiezen om wel of niet te comprimeren. Die keuze heb ik inmiddels weggehaald, dus kan het inderdaad mooi samengevoegd worden.

maarten-k, 2019-11-10:
Misschien is het een idee de dump te comprimeren met zstd. Dit gaat een stuk sneller als gzip met ongeveer(of iets kleiner) resultaat dan gzip. Nadeel is dat er geen default python module is die dit ondersteunt(misschien is er wel een module in pip die ook ondersteund). Terugvallen naar een systeem wijde zstd installatie is dat er een extra dependency bij komt die niet in het upgrade framework past wat er nu is.

dennissiemensma, 2019-11-10:
Dank voor de tip. Ik snap de toegevoegde waarde, echter houd ik het liever bij standaard software, gezien er al een hoop afhankelijkheden zijn voor DSMR-reader.

Daarnaast ga ik over een paar maanden voor versie 3.x het backuppen uberhaupt versimpelen. Waarschijnlijk gaat het dan ofwel buiten DSMR-reader om of DSMR-reader maakt alleen nog backup's van dag- en uur-totalen. Snelheid is dan minder een issue omdat het waarschijnlijk niet meer blokkerend is voor het dsmr_backend proces.

dennissiemensma, 2019-11-11:
@maarten-k ik krijg het niet aan de praat. En als ik het op een andere manier doe:

```
    with gzip.open(backup_file, 'w') as gzip_handle:
        gzip_handle.write(backup_process.stdout)
```

Dan is het in mijn ontwikkelomgeving 3x (!) trager dan op de bestaande manier. Enig idee?

maarten-k, 2019-11-11:
Ik heb niet zo snel een pasklaar antwoord. Maar misschien helpt het als je het bestand als binary weg schrijft ('wb' ipv 'w') (misschien gebeurt dit al impliciet )

dennissiemensma, 2019-11-11:
Dat levert helaas geen verschil op. Wellicht is het gebruiken van een PIPE in Python wat meer overhead dan wanneer native in de shell?

Mocht je niets anders te binnen schieten dan is dat niet erg hoor. Zoals al eerder aangegeven gaan de backups toch op de schop over een paar maanden.

dennissiemensma, 2019-11-13:
Ik laat dit nu voor wat het is. Toch bedankt voor je suggestie. : ]

maarten-k, 2019-11-14:
Python gebruik standaard compression level 9, waar de comand line tool standaard 6 gebruikt.  Dit kan heel goed het snelheids verschil van een factor 3 verklaren

```
    with gzip.open(backup_file, 'wb',6) as gzip_handle:
        gzip_handle.write(backup_process.stdout)
```

dennissiemensma, 2019-11-14:
Bedankt voor je aanvulling. Goed dat je het zegt, want ik had dat niet door.

Ik heb het even getest en hoewel het nog steeds iets trager is, is het niet meer zo'n extreem verschil als eerst.

Het gaat mee in de volgende release.

dennissiemensma, 2019-12-17:
Dit zit in de v2.12 release die net uitgebracht is.

dennissiemensma, 2019-12-18:
Er zijn wat mogelijke issues door deze feature. Nu het comprimeren in het geheugen plaatsvindt, kan het zijn dat grote databases voor out-of-memory fouten zorgen.

Daarom denk ik dat ik dit voor nu terug draai. De oude variant kostte wellicht meer I/O, maar was wel veel betrouwbaarder.

dennissiemensma, 2019-12-19:
Done

closed 2019-12-19 by dennissiemensma

maarten-k, 2019-12-19:
Blijkbaar neem subprocess.run schrijft de output naar het geheugen ipv direct door te sturen naar de pipe. 

Met subprocess.Popen  gaat het wel soepel: geheugen is minimaal (getest met 3GB bestand, memory was een paar MB) en snelheid was vergelijkbaar met een bash pipe. Ik heb het nog niet voor elkaar gekregen om de tests draaiende te krijgen, maar hier staat het idee:
https://github.com/maarten-k/dsmr-reader/tree/streamcompression

dennissiemensma, 2019-12-21:
Bedankt voor je input. Ik laat het voorlopig voor wat het is. Ik vind verbeteringen altijd goed, maar ik blijf in dit geval bij hetgene wat al jaren goed heeft gewerkt. Wellicht dat het ooit bij een complete rewrite opnieuw bekeken wordt.