dsmrreader/dsmr-reader #781: backup failed (postgres 12) pg_dump version mismatch

RezzZ, 2019-11-11:
**Wat gebruik je?**
* DSMR-reader versie: `v2.10`
* Hardware: `i7 server, docker: xirixiz/dsmr-reader-docker, postgres 12 alpine`
* DSMR-protocol slimme meter: `v5`

Ik heb mijn dsmr gemigreerd naar een iets krachtigere server en ben daarbij ook overgestapt van postgres 11 naar v12 (alpine). Draait prima alleen de backups falen:
```
2019-11-11 10:06:38,670 DEBG 'dsmr_backend' stderr output:
[2019-11-11 10:06:38,670] CRITICAL  - Unexpected exit code (1) for backup: b'could not find a "pg_dump" to execute\npg_dump: server version: 12.0; pg_dump version: 11.5\npg_dump: aborting because of server version mismatch\n'
2019-11-11 10:06:38,682 DEBG 'dsmr_backend' stderr output:
[2019-11-11 10:06:38,681] ERROR    b'could not find a "pg_dump" to execute\npg_dump: server version: 12.0; pg_dump version: 11.5\npg_dump: aborting because of server version mismatch\n'
```
ik denk dat er een locale pg_dump wordt gebruikt? Kan deze worden geupdate naar v12?

dennissiemensma, 2019-11-11:
Bedankt voor je melding. DSMR-reader zelf installeert altijd dezelfde client-versie als die van de Postgres-server, want het is dan op hetzelfde OS. Er wordt geen specifieke versie gedefinieerd, gewoon de versie uit de distro.

Er zijn Docker-images van dit project, maar ik beheer ik zelf niet. Ik kan me goed voorstellen dat ze niet met elkaar compatible zijn. Je kunt dan het beste een ticket bij het project `xirixiz/dsmr-reader-docker` aanmaken, maar ik vraag me af of dit gemakkelijk op te lossen is. Het gaat alleen werken als de Postgres 12 client overweg kan met oudere server-versies. Anders zou xirixiz weer meerdere versies moeten beheren.

closed 2019-11-11 by dennissiemensma

RezzZ, 2019-11-12:
ahh ok, dus docker specifiek issue. bedankt! heb ondertussen ook een post bij xirixiz gevonden:https://github.com/xirixiz/dsmr-reader-docker/issues/75