dsmrreader/dsmr-reader #783: gunicorn 20.0.0 dependency breaks use of docker Alpine Linux

marcelvanveen, 2019-11-13:
The dependency on gunicorn 20.0.0 (committed in https://github.com/dennissiemensma/dsmr-reader/commit/eb90995bfa23f19fc23c4ac6ee4edca7e693a3ba) breaks the web frontend on Alpine Linux docker containers (both xirixiz and ualex73):

```
/bin/nice -n 15 gunicorn dsmrreader.wsgi --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_reader.socket --pid /var/tmp/gunicorn--dsmr_reader.pid
Traceback (most recent call last):
  File "/usr/local/bin/gunicorn", line 5, in <module>
    from gunicorn.app.wsgiapp import run
  File "/usr/local/lib/python3.8/site-packages/gunicorn/app/wsgiapp.py", line 9, in <module>
    from gunicorn.app.base import Application
  File "/usr/local/lib/python3.8/site-packages/gunicorn/app/base.py", line 12, in <module>
    from gunicorn.arbiter import Arbiter
  File "/usr/local/lib/python3.8/site-packages/gunicorn/arbiter.py", line 16, in <module>
    from gunicorn import sock, systemd, util
  File "/usr/local/lib/python3.8/site-packages/gunicorn/sock.py", line 14, in <module>
    from gunicorn.socketfromfd import fromfd
  File "/usr/local/lib/python3.8/site-packages/gunicorn/socketfromfd.py", line 26, in <module>
    raise OSError('libc not found')
OSError: libc not found
```

More details are available in see https://github.com/benoitc/gunicorn/issues/2160. 

Is it possible to downgrade the required gunicorn package back to 19.9.0? As a workaround, the fix mentioned in  https://github.com/benoitc/gunicorn/issues/2160#issuecomment-552496543 works.

xirixiz, 2019-11-13:
Thanks Marcel! Alpine Edge can install 19.9.0, Alpine 3.10 can only install 19.7.1-r2.

dennissiemensma, 2019-11-13:
Bedankt voor de melding. Welke versie is uiteindelijk gewenst? 19.9 of 19.7?

dennissiemensma, 2019-11-13:
Ik heb de laagste gepakt. Staat nu in nieuwe release [v2.11.2](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.11.2)

closed 2019-11-13 by dennissiemensma

xirixiz, 2019-11-14:
Works like a charm again! Thanks!

dennissiemensma, 2019-12-31:
@marcelvanveen @xirixiz weten jullie toevallig of dit inmiddels wel werkt? Issue 2160 in Gunicorn lijkt gefixt namelijk.

Is het voor jullie eenvoudig om even te testen of `gunicorn==20.0.4` daadwerkelijk werkt met Alpine?