dsmrreader/dsmr-reader #784: Unpin requirements patches

dennissiemensma, 2019-11-13:
Momenteel staan alle PIP-requirements voor het project vast op de patch-versies. In principe mag dat veranderd worden naar de minors, gezien we erop mogen vertrouwen dat er geen backwards-incompatible releases gedaan worden. Mocht dit toch in de toekomst niet goed gaan, dan kan het altijd weer terug. 

Voordeel is dat ik niet voor elke patch weer de requirements hoef bij te werken. Daarnaast zal ik ook nieuwere minors toestaan, gezien die ook compatible moeten zijn.

dennissiemensma, 2019-11-13:
Bij nader inzien is het wellicht wat veel om ook de minors automatisch te doen. Ik pin wel gewoon de minors, op een nkele uitzondering na (urllib3).

dennissiemensma, 2019-11-13:
Doorgevoerd. Lijkt me beter zo.

closed 2019-11-13 by dennissiemensma

mjanssens, 2019-12-17:
Hoi, ik snap de insteek die je hebt gekozen. oa om niet bijvoorbeeld voor de security releases van django een extra release van DSMRreader te hoeven uitgeven.
Ik probeer te begrijpen wat dan voortaan de workflow is om voor de eindgebruiker up to date te blijven. Heb ik het goed als ik het volgende omschrijf?
Normaal zouden enkel de requirements updaten bij een nieuwe release van dsmrreader. Na deze release kan dsmrreader gelijk blijven, maar de requirements kunnen een patch upgrade krijgen door het draaien van ./deploy.sh

Wellicht is het een idee aangezien de releasenummers van de requirements nu niet meer exact bepaald worden door dsmrreader om de huidige requirements gemakkelijk zichtbaar te maken.
Ik vermoed dat de meeste mensen dit niet interesseert, maar ik denk dat er toch behoefte aan kan zijn, net zoals je bijvoorbeeld de onderliggende debian up to date wilt houden en dan wilt weten of er nog patches uit te rollen zijn. Het is maar een ideetje. Ben benieuwd wat jij ervan denkt.

dennissiemensma, 2019-12-17:
Ik had er op die manier nog niet eens over nagedacht. Wellicht dat ik het dan tzt weer terugdraai. Het ging mij vooral om het telkens moeten bijwerken van de requirements, want de releases komen doorgaans niet eerder vanwege een requirement update.

mjanssens, 2019-12-17:
Misschien even laten bezinken en kijken wat de reacties van anderen zijn op deze wijziging.

mjanssens, 2019-12-18:
En meteen de proef op de som ;-)
Vandaag was er weer een django security patch gereleased.
Draaien van ./deploy.sh en we zitten op 2.2.9, verder niets veranderd en alles werkt nog.

dennissiemensma, 2019-12-18:
Het punt is alleen wel dat dat vereist dat je op de hoogte moet zijn dat er een update is. Dat vind ik wel het grootste nadeel.