dsmrreader/dsmr-reader #786: Dataverwerking loopt achter: <20 dagen>

MarcoLooy, 2019-11-19:
Help!

Nadat Liander mijn slimme meter heeft vervangen, heb ik de melding "**Dataverwerking loopt achter: ...**". Dit is nu 20 dagen geleden.

Dit komt waarschijnlijk omdat na installatie het nog eens drie dagen duurde om de nieuwe meter te configureren. Tot die tijd hadden de telegrammen **een verkeerde timestamp**; 2000-01-01.

Inmiddels heb ik deze records verwijderd uit de **dsmr_datalogger_dsmrreading** tabel.

Echter nu vraag ik me af hoe ik de **berekening van de dagtotalen opnieuw kan aftrappen**, zodat de dataverwerking weer bij is?

Marco.

dennissiemensma, 2019-11-19:
Het hangt er vanaf welke versie van DSMR-reader je draait. Welke versie staat er rechtsbovenin?

MarcoLooy, 2019-11-20:
Dennis,

Geweldig, na een update naar v2.11.2 zijn alle de dagtotalen weer op orde, en is de dataverwerking weer bij.

Bedankt!

Marco.

closed 2019-11-20 by MarcoLooy