dsmrreader/dsmr-reader #787: Geen informatie bij 'All time highs & lows'

Xironics, 2019-11-20:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.11.2`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Debian VM
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Op de statistics pagina zie ik geen 'ALL TIME HIGHS & LOWS' informatie meer.

In de database is er wel 1 row maar hierin bevind zich geen data.

dsmrreader=# select * from dsmr_stats_electricitystatistics;
 id | highest_usage_l1_timestamp | highest_usage_l2_timestamp | highest_usage_l3_timestamp | highest_return_l1_timestamp | highest_return_l2_timestamp | highest_return_l3_timestamp | highest_usage_l1_value | highest_usage_l2_value | highest_usage_l3_value | highest_return_l1_value | highest_return_l2_value | highest_return_l3_value | lowest_usage_l1_timestamp | lowest_usage_l1_value | lowest_usage_l2_timestamp | lowest_usage_l2_value | lowest_usage_l3_timestamp | lowest_usage_l3_value
----+----------------------------+----------------------------+----------------------------+-----------------------------+-----------------------------+-----------------------------+------------------------+------------------------+------------------------+-------------------------+-------------------------+-------------------------+---------------------------+-----------------------+---------------------------+-----------------------+---------------------------+-----------------------
  1 |                            |                            |                            |                             |                             |                             |                        |                        |                        |                         |                         |                         |                           |                       |                           |                       |                           |
(1 row)

Na een 'truncate table dsmr_stats_electricitystatistics;' wordt er wel een nieuwe rij gemaakt maar nog steeds niet gevuld met data.

dennissiemensma, 2019-11-21:
Bedankt voor je vraag. Heb je in de datalogger instellingen toevallig het bijhouden van groepen/fasen uit staan?

Xironics, 2019-11-21:
Ik kon deze setting niet (meer) vinden. Ik heb het bij eerdere versies wel eens zien staan. 
Kun je aangeven waar dit staat?

dennissiemensma, 2019-11-21:
Het kan zijn dat die dan al uitgefaseerd is en altijd aan staat. Ik kijk van het weekend even voor je hoe het werkt qua alle time stats. Ik heb nu geen toegang tot m'n werkplek.

Xironics, 2019-11-21:
Top, alvast bedankt!

GieltjE, 2019-11-22:
Misschien is het een idee om ook de min/max van voltage te gaan loggen en weergeven?

dennissiemensma, 2019-11-24:
@Xironics ik kan niet zo 1-2-3 een reden bedenken waarom het niet zo werken, dus je kunt het beste debuggen. Ik heb een branch voor je gemaakt waarin expliciete logging zit wat de applicatie exact doet op dat punt. Een voorbeeld hieronder:

```
[2019-11-24 12:11:57,035] DEBUG    High/low stats: Checking all time stats
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Comparing highest_usage_l1 all time (0.468) to now (1.292)
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Updating highest_usage_l1 to 1.292
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Comparing highest_usage_l2 all time (0.105) to now (0.372)
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Updating highest_usage_l2 to 0.372
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Comparing highest_usage_l3 all time (1.440) to now (0.119)
[2019-11-24 12:11:57,037] DEBUG    High/low stats: No value for phase_currently_returned_l1
[2019-11-24 12:11:57,037] DEBUG    High/low stats: No value for phase_currently_returned_l2
[2019-11-24 12:11:57,037] DEBUG    High/low stats: No value for phase_currently_returned_l3
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Comparing lowest_usage_l1 all time (0.006) to now (1.292)
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Comparing lowest_usage_l2 all time (0.019) to now (0.372)
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Comparing lowest_usage_l3 all time (0.001) to now (0.119)
[2019-11-24 12:11:57,037] DEBUG    High/low stats: Saving stats
```

Hiervoor zul je het volgende moeten doen:
* Debug-logging aanzetten: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

* Branch wisselen (is gebaseerd op de laatste release, die je al draait):
```
sudo su - dsmr
git fetch
git checkout -b debug/787-all-time-stats origin/debug/787-all-time-stats
./deploy.sh
```

Afhankelijk hoe je telegrammen inschiet (datalogger of API), vind je de logging als het goed in een van deze:
* `/var/log/supervisor/dsmr_datalogger.log` (Logs over de datalogger)
* `/var/log/supervisor/dsmr_webinterface.log` (Logs over de webinterface)

Ik hoor graag of je ze kan vinden en of er iets opvallends in staat.

dennissiemensma, 2019-11-24:
@GieltjE je kunt daar het beste een apart verzoek/issue voor maken.

Xironics, 2019-11-25:
Bedankt voor je (re)actie!

Ik voer de data in d.m.v. de API v2 - datalogger/dsmrreading, de debug informatie is daarbij als volgt:

`[2019-11-25` 09:01:08,810] DEBUG    High/low stats: Checking all time stats
[2019-11-25 09:01:08,812] DEBUG    High/low stats: No value for phase_currently_delivered_l1
[2019-11-25 09:01:08,813] DEBUG    High/low stats: No value for phase_currently_delivered_l2
[2019-11-25 09:01:08,813] DEBUG    High/low stats: No value for phase_currently_delivered_l3
[2019-11-25 09:01:08,813] DEBUG    High/low stats: No value for phase_currently_returned_l1
[2019-11-25 09:01:08,813] DEBUG    High/low stats: No value for phase_currently_returned_l2
[2019-11-25 09:01:08,813] DEBUG    High/low stats: No value for phase_currently_returned_l3
[2019-11-25 09:01:08,814] DEBUG    High/low stats: No value for phase_currently_delivered_l1
[2019-11-25 09:01:08,814] DEBUG    High/low stats: No value for phase_currently_delivered_l2
[2019-11-25 09:01:08,814] DEBUG    High/low stats: No value for phase_currently_delivered_l3`

In mijn JSON POST stuur ik de volgende waardes:
`{
  "electricity_currently_delivered": 0.43,
  "electricity_currently_returned": 0,
  "electricity_delivered_1": 3558.19,
  "electricity_delivered_2": 3446.38,
  "electricity_returned_1": 0.02,
  "electricity_returned_2": 0,
  "timestamp": "2019-11-25T09:02:35+01:00",
  "extra_device_delivered": 1321.61,
  "extra_device_timestamp": "2019-11-25T09:00:00+01:00"
}`

waarbij ik de volgende response krijg:

`{
  "id": 5206299,
  "timestamp": "2019-11-25T09:02:35+01:00",
  "electricity_delivered_1": "3558.190",
  "electricity_returned_1": "0.020",
  "electricity_delivered_2": "3446.380",
  "electricity_returned_2": "0.000",
  "electricity_currently_delivered": "0.430",
  "electricity_currently_returned": "0.000",
  "phase_currently_delivered_l1": null,
  "phase_currently_delivered_l2": null,
  "phase_currently_delivered_l3": null,
  "extra_device_timestamp": "2019-11-25T09:00:00+01:00",
  "extra_device_delivered": "1321.610",
  "phase_currently_returned_l1": null,
  "phase_currently_returned_l2": null,
  "phase_currently_returned_l3": null,
  "phase_voltage_l1": null,
  "phase_voltage_l2": null,
  "phase_voltage_l3": null
}`

Ik heb een Sagemcom-XS210 meter (enkele fase), hieronder een voorbeeld van een telegram.
Moet ik de waarde van 'electricity_currently_delivered' ook als 'phase_currently_delivered_l1' meesturen in de POST?

`/Ene5\XS210 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(XXXXXXXXXXX)
0-0:96.1.1(XXXXXXXXXXX)
1-0:1.8.1(001288.176*kWh)
1-0:1.8.2(001183.790*kWh)
1-0:2.8.1(000000.016*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.187*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00008)
0-0:96.7.9(00005)
1-0:99.97.0(4)(0-0:96.7.19)(180525124734S)(0000001524*s)(180329122502S)(0000000243*s)(180322065703W)(0000008822*s)(171214052229W)(0000000250*s)
1-0:32.32.0(00006)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(234.0*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.187*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303533303033363739383734383137)
0-1:24.2.1(181011135500S)(00385.086*m3)
!XXXX`

Alvast bedankt voor je reactie!

dennissiemensma, 2019-11-25:
Bedankt voor het debuggen. Ik denk dat je dan inderdaad het beste eerst kunt proberen om simpelweg `phase_currently_delivered_l1` mee te geven.

Xironics, 2019-11-29:
> Bedankt voor het debuggen. Ik denk dat je dan inderdaad het beste eerst kunt proberen om simpelweg `phase_currently_delivered_l1` mee te geven.

Ik had vandaag even tijd om de ESP opnieuw te flashen met de nieuwe code.
Nu ik deze variabel wel meegeef worden de statistieken ook weer juist weergegeven:

```
[2019-11-29 11:45:28,482] DEBUG    High/low stats: Checking all time stats
[2019-11-29 11:45:28,484] DEBUG    High/low stats: Comparing highest_usage_l1 all time (0.380) to now (0.370)
[2019-11-29 11:45:28,484] DEBUG    High/low stats: No value for phase_currently_delivered_l2
[2019-11-29 11:45:28,485] DEBUG    High/low stats: No value for phase_currently_delivered_l3
[2019-11-29 11:45:28,485] DEBUG    High/low stats: No value for phase_currently_returned_l1
[2019-11-29 11:45:28,485] DEBUG    High/low stats: No value for phase_currently_returned_l2
[2019-11-29 11:45:28,486] DEBUG    High/low stats: No value for phase_currently_returned_l3
[2019-11-29 11:45:28,486] DEBUG    High/low stats: Comparing lowest_usage_l1 all time (0.220) to now (0.370)
[2019-11-29 11:45:28,486] DEBUG    High/low stats: No value for phase_currently_delivered_l2
[2019-11-29 11:45:28,487] DEBUG    High/low stats: No value for phase_currently_delivered_l3
```

Hartelijk dank voor je hulp!

dennissiemensma, 2019-11-29:
Top, bedankt voor je terugkoppeling!

Je kunt dan weer terug naar `master`:
```
sudo su - dsmr
git checkout master
./deploy.sh
```

closed 2019-11-29 by dennissiemensma