dsmrreader/dsmr-reader #788: All time highs & low reset

glsf91, 2019-11-24:
Is het mogelijk om de all time Highs en lows te resetten zodat ze opnieuw bepaald worden vanaf nu?

dennissiemensma, 2019-11-25:
```
sudo su - postgres
psql dsmrreader

# Let op: lange query
update dsmr_stats_electricitystatistics set highest_usage_l1_timestamp = NULL,highest_usage_l2_timestamp = NULL,highest_usage_l3_timestamp = NULL,highest_return_l1_timestamp = NULL,highest_return_l2_timestamp = NULL,highest_return_l3_timestamp = NULL,highest_usage_l1_value = NULL,highest_usage_l2_value = NULL,highest_usage_l3_value = NULL,highest_return_l1_value = NULL,highest_return_l2_value = NULL,highest_return_l3_value = NULL,lowest_usage_l1_timestamp = NULL,lowest_usage_l1_value = NULL,lowest_usage_l2_timestamp = NULL,lowest_usage_l2_value = NULL,lowest_usage_l3_timestamp = NULL,lowest_usage_l3_value = NULL;
```

Nu zou alles gereset moeten zijn.

closed 2019-11-25 by dennissiemensma

glsf91, 2019-11-25:
Thanks.
May add this to the documentation (FAQ database)?

dennissiemensma, 2019-11-25:
Ik vind het een goed idee, maar ik laat het even voor wat het is. Ik voeg dingen doorgaans aan de FAQ toe als er veel of meerdere vragen over zijn. :]