dsmrreader/dsmr-reader #789: DSMR very chatty with Postgresql

balk77, 2019-11-26:
**Wat gebruik je?**
* DSMR-reader versie: `v2.11.2`
* Hardware: `ualex73/dsmr-reader-docker` on Ubuntu
* DSMR-protocol slimme meter *(v4 of v5)*: `v4`

*Beschrijf hier je verhaal...*
I noticed that DSMR reader is consuming a significant portion of my CPU time, and more specifically the DSMR related activity of Postgresql is high. So I activated logging on postgresql and found that it is firing many queries: 1000 in one minute. Is this normal behavior?

Log:
https://www.dropbox.com/s/il4i3hx9s50kczq/dsmrquery.log?dl=0

My database is few years old; do I have some old defaults?

I am running a docker instance like this:
`docker run -d --restart always --name dsmr -p 7080:80  -e DSMR_USER=admin  -e DSMR_PASSWORD=xxxx  -e DSMR_EMAIL=root@localhost -e DB_USER=dsmrreader -e DB_PASS=xxxx  -e DB_NAME=dsmrreader -e DB_HOST=192.168.0.9  -v /dev:/dev  --device /dev/ttyUSB-P1:/dev/ttyUSB-P1 ualex73/dsmr-reader-docker`

What can I do to tame it down?

dennissiemensma, 2019-11-27:
Since you're using a DSMR-v4 smart meter, you can opt to lower the interval of the backend-process.

See https://dsmr-reader.readthedocs.io/en/latest/settings.html#dsmrreader-backend-sleep
You could try setting `DSMRREADER_BACKEND_SLEEP` to `5` seconds or even higher. It should reduce the number of runs and queries significantly.

balk77, 2019-11-27:
Dennis, thanks for your response. I have set this now to 5 and that reduces the number of queries to 300 per minute. I think that is still a lot but it definitely helps! It seems to reduce the CPU load by 10%points.

What happens if I increase the number to 10 instead of 5? What is the purpose of this setting? (and of all those queries?)

Since I am using a docker setup I was unable to run the `post-deploy.sh` script without disabling the check for virtual environment in it. Simply commenting those lines let the script run. This may be more a question for @ualex73 : is there a way to pass on these kind of settings as environmental values to docker? How would that work with `post-deploy.sh`?

dennissiemensma, 2019-11-28:
You can try even higher values, but executing (select) queries is not a bad thing what so ever. 

The backend process is constantly doing lots of stuff and sleeps now and then, but every setup requires a different sleep duration.

I might move the sleep settings to the webinterface anyway, making it more convenient to change them without editing files.

balk77, 2019-11-28:
I have moved to https://github.com/xirixiz/dsmr-reader-docker which allows me to specify `DSMRREADER_BACKEND_SLEEP` from the `docker-compose.yaml` file 👍 
Thanks @xirixiz

ualex73, 2019-11-28:
I am not home at this moment, but I will add this option early next week to
the docker.

dennissiemensma, 2019-11-29:
Ik denk dat ik ze sowieso wel verplaats naar de webinterface hoor. Het is wat gek dat de docker-maintainers voor elke setting weer een aparte envvar moeten maken. Daar was initieel geen rekening mee gehouden vanuit mijn zijde.

pyrocumulus, 2019-12-10:
Would be nice to make it possible to influence this through the webinterface. 

Just to be clear: is there a negative side to increasing the sleep duration? Will I miss data by increasing that setting to 5 or 10 even; or will it still just collect all the data but wait longer before sending it to the database?

dennissiemensma, 2019-12-10:
Increasing the sleep for the backend will just make it wait a bit and not affect data. Increasing the sleep for the datalogger may cause in skipping telegrams.

pyrocumulus, 2019-12-11:
That's what I thought but I just wanted to make sure :) Thanks!

closed 2019-12-16 by dennissiemensma