dsmrreader/dsmr-reader #790: In- en uitzoomen 'gesture' op grafiek werkt niet meer

SaturnusDJ, 2019-11-27:
**Wat gebruik je?**
Safari op iOS 12.4. Niet gewijzigd tussen laatst werkende DSMR Reader versie voor dit probleem.

**DSMR-reader versie**
2.12.2, maar voor mij begon dit probleem bij 2.9.0 al.

Voorheen kon je met twee vingers de dashboard grafieken in- en uitzoomen. Dit werkte een stuk gemakkelijker dan de balk onder de grafieken. Het werkt echter niet meer sinds rond versie 2.9.0.

dennissiemensma, 2019-11-28:
Bedankt voor je melding. Helaas kan ik weinig voor je doen. Voor zover ik weet is er niets aan veranderd, sowieso niet bewust.

Ik heb even voor je gegraven in de geschiedenis. De code voor de dashboard-grafieken is bijna [een jaar geleden](https://github.com/dennissiemensma/dsmr-reader/tree/master/dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/dashboard) voor het laatst gewijzigd. En de laatste upgrade van eCharts (de grafieken die ik gebruik) was op [v2.3](https://github.com/dennissiemensma/dsmr-reader/compare/v2.2.0..v2.3.0#diff-bf4f42c0b61d258cd51d1aaf8da00e8b).

Het kan wel aan eCharts liggen, maar dat is een externe library waar ik niets aan kan veranderen. Wellicht dat het zichzelf oplost met een toekomstige update. Ik zal tzt kijken voor een update.

dennissiemensma, 2019-11-28:
Ik heb net een update gedaan voor de volgende release. Ik weet niet of dat het oplost, maar je kunt het tzt proberen.

SaturnusDJ, 2019-11-28:
Zal het laten weten!

SaturnusDJ, 2019-12-12:
Geen verbetering in v2.11.3.

dennissiemensma, 2019-12-12:
v2.11.3 was een patch met alleen een security update voor Django. De wijziging zit in de volgende release van v2.12.

dennissiemensma, 2019-12-17:
@SaturnusDJ er is een nieuwe versie uit met de bijgewerkte echarts. Je kunt updaten en kijken of dat iets voor je verbetert.

dennissiemensma, 2019-12-17:
Als ik het zelf even probeer op mijn telefoon, dan werkt het wel goed in Chrome, maar niet in Firefox (kan alleen uitzoomen). Safari en Edge kan ik niet testen.

dennissiemensma, 2019-12-18:
Voor nu sluit ik deze. Ik kan sowieso inhoudelijk niets veranderen aan hoe de grafieken reageren op gestures.

closed 2019-12-18 by dennissiemensma

SaturnusDJ, 2019-12-27:
Werkt weer. 👍