dsmrreader/dsmr-reader #791: Parsing van Fluvius meter werkt niet voor I1/I2/I3

djm300, 2019-12-04:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.xxx`
Versie 2.11.2
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi ...`
Debian x64 
* DSMR-protocol slimme meter *(v4 of v5)*: `v...`
Fluvius
*Beschrijf hier je verhaal...*
Ik gebruik DSMR om de meters (elek & gas) te lezen en te exporteren naar MQTT. Ik heb de meter op "Fluvius" mode gezet maar merk dat er bepaalde velden niet geparset worden. Via "View DSMR reading" zie ik hoe een telegram geparset wordt:

De velden "Phase currently delivered" & "Phase currently returned" zijn "-"alhoewel het ruwe telegram als volgt is:

/FLU5\253769484_A

0-0:96.1.4(<even weggelaten>)
0-0:96.1.1(<even weggelaten>)
0-0:1.0.0(191204101703W)
1-0:1.8.1(000662.563*kWh)
1-0:1.8.2(000549.603*kWh)
1-0:2.8.1(000058.119*kWh)
1-0:2.8.2(000049.869*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(02.960*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(230.6*V)
1-0:52.7.0(231.5*V)
1-0:72.7.0(230.4*V)
1-0:31.7.0(005*A)
1-0:51.7.0(004*A)
1-0:71.7.0(004*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(<even weggelaten>)
0-1:24.4.0(1)
0-1:24.2.3(191204101617W)(00000.136*m3)
!B8C7

dennissiemensma, 2019-12-04:
Bedankt voor je melding. Ik begrijp niet helemaal wat je bedoelt. Als ik je telegram parse, kom ik hier op uit:

```
{   '_state': <django.db.models.base.ModelState object at 0x7fbb568e1dd8>,
    'electricity_currently_delivered': Decimal('2.960'),
    'electricity_currently_returned': Decimal('0.000'),
    'electricity_delivered_1': Decimal('662.563'),
    'electricity_delivered_2': Decimal('549.603'),
    'electricity_returned_1': Decimal('58.119'),
    'electricity_returned_2': Decimal('49.869'),
    'extra_device_delivered': Decimal('0.136'),
    'extra_device_timestamp': datetime.datetime(2019, 12, 4, 9, 16, 17, tzinfo=<UTC>),
    'phase_currently_delivered_l1': None,
    'phase_currently_delivered_l2': None,
    'phase_currently_delivered_l3': None,
    'phase_currently_returned_l1': None,
    'phase_currently_returned_l2': None,
    'phase_currently_returned_l3': None,
    'phase_voltage_l1': Decimal('230.6'),
    'phase_voltage_l2': Decimal('231.5'),
    'phase_voltage_l3': Decimal('230.4'),
    'timestamp': datetime.datetime(2019, 12, 4, 9, 17, 3, tzinfo=<UTC>)}
```

In het telegram (en dus hierboven) zie ik geen individuele informatie over de fasen, afgezien van de voltages.

Ik denk dat je alleen de beschikking hebt over `electricity_currently_delivered`. Waarschijnlijk komt dit doordat Fluvius-meters DSMR slechts deels implementeren of ontsluiten via de P1-poort.

Normaal [ziet fase-informatie er zo uit](https://github.com/dennissiemensma/dsmr-reader/blob/v2.11.2/dsmr_datalogger/tests/datalogger/test_landisgyr350_dsmr42.py#L44).

M-VdM, 2019-12-07:
Hier vindt u alle technische info:  [Fluvius > Handleidingen Digitale Meter](https://www.fluvius.be/nl/thema/meters-en-meterstanden/handleidingen-digitale-meter)

dennissiemensma, 2019-12-08:
@M-VdM bedankt voor de info. Ik heb echter in #750 en #769 al geconstateert dat de 'buitenlandse' slimme meters niet alle gegevens teruggeven of dat er verschil zit tussen meters. Tot dat verandert kan ik denk ik weinig voor jullie doen.

closed 2019-12-08 by dennissiemensma

frankforpresident, 2020-01-05:
@djm300,

Zou jij mij kunnen vertellen waar jij je kabel gekocht hebt?