dsmrreader/dsmr-reader #793: Alle meldingen in 1x sluiten

wouzzie, 2019-12-06:
Hoi Dennis,

Allereerst bedankt voor je fantastische project! 

Gisteren heeft door een storing mijn internet er halve dag uitgelegen. Voor DSMR-reader geen probleem natuurlijk, ware het niet dat dit ook betekent dat de buienradar API niet bereikt kon worden. Sinds een paar updates geleden toont dit natuurlijk netjes als notificatie. Dit levert nu bij mij de situatie op dat er een lijst is ontstaan van 25+ notifiticaties die ik handmatig moet accepteren. 

[image]

Zou er een mogelijkheid  zijn om een "accept all" achtige mogelijkheid in te bouwen voor de notificaties? 

Alvast bedankt en keep up the good work!

dennissiemensma, 2019-12-07:
Bedankt voor je suggestie. Dat lijkt me een goede toevoeging. Ik denk dat ik sowieso alle notificaties verplaats naar een aparte pagina en in het dashboard alleen een melding geeft dat er notificaties zijn. Bulk sluiten is daar dan een nuttige feature.

dennissiemensma, 2020-01-05:
Voor `v2.14` heb ik de notificaties verplaatst naar een aparte pagina, waar je ze straks ook in bulk kunt sluiten. Screenshots:

[image]

[image]

dennissiemensma, 2020-01-05:
Doorgevoerd in develop branch.

closed 2020-01-05 by dennissiemensma