dsmrreader/dsmr-reader #794: Django security releases issued: 2.2.8

mjanssens, 2019-12-07:
https://www.djangoproject.com/weblog/2019/dec/02/security-releases/

> This is a backwards-incompatible change, and the Django security team is aware that some users of Django were depending on the ability to allow editing of inlines in the admin form of an otherwise view-only parent model.

Ik weet niet of dit van toepassing is op dsmr-reader.

dennissiemensma, 2019-12-08:
Bedankt voor je melding. Ik had hem toevallig van de week al klaarstaan voor de volgende release via #792. 

Maar ik zal nu even een patch uitbrengen die alleen dit fixt.

dennissiemensma, 2019-12-08:
Uitgebracht in [v2.11.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.11.3)

closed 2019-12-08 by dennissiemensma