dsmrreader/dsmr-reader #795: Database wat groter opeens.

ph0real, 2019-12-09:
Hoi Dennis,

Onlangs (ik vermoed vrijdag op zaterdag) is de postgresql database gaan groeien; zodanig dat heel de /dev/root partitie van de raspberry werd opgegeten. De database is ongeveer 2 gig momenteel. Ik ben begonnen met meten ergens in september. Ik vermoed dat de applicatie als een malle records is gaan aanmaken door een probleem; echter kan ik t niet zo goed pinpointen.

Ik heb met je excellente instructie de database verplaatst naar een ruimere locatie. Daarna nog getracht met vacuumdb de database te verkleinen; maar hij blijft rond de 2gig. Weet je een logfile waar ik kan beginnen om t probleem te troubleshooten?

jeroenpeters1986, 2019-12-09:
Hoi @ph0real, je schrijf dat het opeens wat groter is geworden. Maar weet je dat ook zeker? 

Fijn in ieder geval dat het gelukt is om het te verplaatsen, want dat geeft ruimte, ook om te werken. Als je partitie bijna vol is kan het ook zijn dat het compacten/opschonen niet meer lukt en dat het daar door is gekomen dat er niets meer af ging. Weet je overigens of je het compacten/opschonen had ingeschakeld? Zo niet, dan is het een goed idee om dat te doen, dat kan met deze instructies:https://dsmr-reader.readthedocs.io/en/latest/admin/datalogger.html#data-cleanup-retention

dennissiemensma, 2019-12-09:
@ph0real aanvullend op bovenstaande, mits je een DSMR-v5-meter hebt, kun je er ook nog voor kiezen om je datalogger minder telegrammen te laten loggen. Bijvoorbeeld elke 5 of 10 seconden. Zie [DSMRREADER_DATALOGGER_SLEEP](https://dsmr-reader.readthedocs.io/en/latest/settings.html#dsmrreader-datalogger-sleep).

Wat betreft de database-grootte, je kunt hiermee zien wat veel data kost. Ik vermoed telegrammen en consumptie:
```
sudo su - postgres
psql dsmrreader

SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 20;
````

ph0real, 2019-12-10:
> Hoi @ph0real, je schrijf dat het opeens wat groter is geworden. Maar weet je dat ook zeker?
>
> Fijn in ieder geval dat het gelukt is om het te verplaatsen, want dat geeft ruimte, ook om te werken. Als [...]

Bedankt voor de tip Jeroen. Ik ga zeker ff kijken naar deze settings om te zien of ik daarmee wat ruimte kan winnen. Verder ben ik nog niet toegekomen aan een root cause analysis om te zien wat er nu werkelijk is misgegaan. Ik heb de database verplaatst als workaround, en enkel geconstateerd dat de postgresql locatie de grootste ruimtevreter was op de partitie op dat moment. Wordt dus vervolgd.

ph0real, 2019-12-10:
> @ph0real aanvullend op bovenstaande, mits je een DSMR-v5-meter hebt, kun je er ook nog voor kiezen om je datalogger minder telegrammen te laten loggen. Bijvoorbeeld elke 5 of 10 seconden. Zie [...]

Ga ik ook checken, dank je!

ph0real, 2019-12-13:
Er is in ieder geval een aanzienlijk grotere backup gemaakt maandag. Zie onder voor de gebruikelijke hoeveelheden. Ik heb een data retentie ingesteld, en ben bezig de database te vacuumen op basis van jullie documentatie. Ook de telegram tips neem k mee. Wat mij betreft kunnen we dit sluiten.

drwxr-xr-x 3 dsmr dsmr 4,0K sep  4 21:06 archive
-rw-r--r-- 1 dsmr dsmr    0 dec 13 09:00 dsmrreader-postgresql-backup-Friday.sql
-rw-r--r-- 1 dsmr dsmr  90M dec  6 02:05 dsmrreader-postgresql-backup-Friday.sql.gz
-rw-r--r-- 1 dsmr dsmr 783M dec  9 23:10 dsmrreader-postgresql-backup-Monday.sql
-rw-r--r-- 1 dsmr dsmr  86M dec  2 02:04 dsmrreader-postgresql-backup-Monday.sql.gz
-rw-r--r-- 1 dsmr dsmr  91M dec  7 02:05 dsmrreader-postgresql-backup-Saturday.sql.gz
-rw-r--r-- 1 dsmr dsmr  92M dec  8 17:00 dsmrreader-postgresql-backup-Sunday.sql.gz
-rw-r--r-- 1 dsmr dsmr  89M dec  5 02:05 dsmrreader-postgresql-backup-Thursday.sql.gz
-rw-r--r-- 1 dsmr dsmr  87M dec  3 02:04 dsmrreader-postgresql-backup-Tuesday.sql.gz
-rw-r--r-- 1 dsmr dsmr  88M dec  4 02:05 dsmrreader-postgresql-backup-Wednesday.sql.gz

closed 2019-12-13 by ph0real

ph0real, 2019-12-13:
DB is van 2 gb naar 198mb gegaan net. 🥇