dsmrreader/dsmr-reader #796: Highest return L1- verkeerd geregistreerd

pm-b, 2019-12-10:
* DSMR-reader versie:  #`v2.11.2`
* Hardware: `RaspberryPi 3b`
* DSMR-protocol slimme meter: `v5.0`

Als ik in statistics kijk zie ik bij de all time highs regelmatig een waarde staan voor `Highest return L1-`, terwijl ik helemaal niets kan terugleveren. Onder status -> capabilities zie ik *No electricity return recorded*, dus dat klopt wel gewoon. 

[image]

dennissiemensma, 2019-12-10:
Bedankt voor je melding. Ik heb dit vaker gezien en ook bij mijn meter. Maar ik weet niet precies waar het aan ligt. 

Mijn vermoeden is dat de slimme meter dit daadwerkelijk teruggeeft in de fase-informatie, maar ik kan het niet makkelijk reproduceren.

Aangezien je meting van gisteren was, kun je kijken of je hem kan vinden en daar hetzelfde ziet:
```
sudo su - postgres
psql dsmrreader

select timestamp, phase_currently_returned_l1 from dsmr_datalogger_dsmrreading where phase_currently_returned_l1 > 0 order by timestamp desc limit 10;
```

dennissiemensma, 2019-12-16:
@pm-b ben je in de gelegenheid geweest om bovenstaande te checken?

pm-b, 2019-12-17:
@dennissiemensma Ja, ze staan er daadwerkelijk nog in. 
   
           timestamp        | phase_currently_returned_l1 
    ------------------------+-----------------------------
     2019-12-09 20:19:40+01 |                       1.258
     2019-12-09 20:19:38+01 |                       1.258
     2019-12-09 20:19:36+01 |                       1.258
     2019-12-09 20:19:34+01 |                       1.258
     2019-12-09 20:19:32+01 |                       1.258

dennissiemensma, 2019-12-17:
Ik denk dat de meter dat dan ook gemeten/doorgegeven heeft, maar dat is wat lastig te checken gezien de telegrammen zelf nieet bewaard worden, alleen het geparste. Het is mogelijk dat er een fout in de parser zit, maar dat lijkt me zeer onwaarschijnlijk, gezien er meerdere automatische tests voor zijn.

Je zou nog telegram logging kunnen inschakelen, maar daarmee slijt je SD-kaartje weer wat sneller en het is onbekend of en wanneer het weer voorkomt. Dus die keuze laat ik aan jou.

pm-b, 2019-12-25:
Als ik deze datum opzoek in het archief staat er nu ook een grafiek bij voor returned, maar daarin staat het wel op 0. 
[image]

dennissiemensma, 2019-12-25:
Dat komt omdat er teminste 1 meting is met teruglevering, daar wordt dat op gebaseerd. Je kunt eventueel die metingen op nul zetten. 

Of je kunt dit doen, is wellicht wat makkelijker: https://dsmr-reader.readthedocs.io/en/latest/settings.html#dsmrreader-disabled-capabilities
```
DSMRREADER_DISABLED_CAPABILITIES = ['electricity_returned']
```

pm-b, 2019-12-25:
Mijn vorige opmerking was vooral omdat ik me afvraag waarom ik in het archief van 9 december om 20.00u geen return zie, terwijl bij de all time highs nog steeds te zien is dat die er dag om 20:19u een return was van 1.258 watt. 

Ik zal sowieso de returns op disabled zetten.

dennissiemensma, 2019-12-25:
Dat komt doordat het ene gegeven afhankelijk is van de actieve teruglevering (W). En de andere is n.a.v. van of er daadwerkelijk iets teruggeleverd is, de meterstand (Wh).

dennissiemensma, 2019-12-31:
Mocht je nog meer info nodig hebben, reageer gerust later.

closed 2019-12-31 by dennissiemensma